OctoFarm/OctoFarm #88: Printers connected but not showing info

byurick, 2020-08-08:
Was using a previous version of octofarm when I started having this issue so I upgraded it, no change. Then I did a fresh install and still having the same issues. Both printers are on octoprint 1.4.2 I can see the cameras and they say they're both online and connected but no filename/temperatures or the "current operations" tabs shows any info.

NotExpectedYet, 2020-08-08:
This is a duplicate of #85 as I haven't updated OctoFarm to work with 1.4.1+

byurick, 2020-08-08:
Thanks, I did see #85  but wasn't sure if it was the same.

NotExpectedYet, 2020-08-08:
No worries, I should have it updated tomorrow.

bolsoncerrado, 2020-08-10:
Came here with similar issues. However you reffer to it as #85 having socket issues, however I'm running both 1.41 and 1.42 in some printers but my webocket icons are all green :?

[image]

NotExpectedYet, 2020-08-10:
> 
> 
> Came here with similar issues. However you reffer to it as #85 having socket issues, however I'm running both 1.41 and 1.42 in some printers but my webocket icons are all green :?
> 
> [image]

This should be fixed on the latest RC, 

Would you mind trying it out at all? 

OctoFarm to save the users time makes a guess at the user you use. Going forward for 1.4.2 you will have to use either the user key or create an application key specifically for OctoFarm. My only other alternative was triggering the app registration plugin which would require you to open the octoprint web interface for each. I think that is why people are seeing different results though, anyone like me using the global key that will no longer work after upgrading octoprint to 1.4.2.

bolsoncerrado, 2020-08-10:
Yes, do I just git clone the latest Rc and that's all or do I need to enable the dev too?

My only other pending task is how to damn launch it upon boot up on macos, i cant figure it out 👍

NotExpectedYet, 2020-08-11:
> 
> 
> Yes, do I just git clone the latest Rc and that's all or do I need to enable the dev too?
> 
> I'm using the global key, not a specific one btw.
> 
> My only other pending task is how to damn launch it upon boot up on macos, i cant figure it out 👍

Yes that's it :) I'd probably setup a seperate instance but that's totally up to you. I've squashed a lot of bugs I just want to make sure no more exist before releasing properly. 

Have you checked the wiki? the latest version uses pm2 and can be made to run on startup for all the major OS'

byurick, 2020-08-11:
So far so good with the latest RC, seems to have cleared up all the connection issues I was seeing.

NotExpectedYet, 2020-08-12:
> 
> 
> So far so good with the latest RC, seems to have cleared up all the connection issues I was seeing.

Epic thanks for the report!

Crowlord, 2020-08-12:
Latset dev version loves my Kossel and works perfectly. 

All three prusas it hates. It won't even see them as online if i have a key entered (either the base api which shouldn't work anyway and the app key which i generated via user)

All on oprint 1.4.2

NotExpectedYet, 2020-08-12:
@Crowlord no that issue again. 

That's a crap regression and can only think it's the file scanning changes...

That's annoying. Could you open another ticket and put it down to specifically when a Prusa is attached and grab me some logs if you get chance? Also a decent description of what the UI is doing if anything.

Crowlord, 2020-08-12:
I will bundle you up some logs etc

NotExpectedYet, 2020-08-12:
@Crowlord legend thank you

NotExpectedYet, 2020-08-16:
Master updated closing this issue, should be resolved.

closed 2020-08-16 by NotExpectedYet