OctoFarm/OctoFarm #89: Prusa printers not communicating when using API or Application Keys. (Dev)

Crowlord, 2020-08-12:
Four printers connected. 3 Prusas 1 anycubic. 

Anycubic connects to websocket fine and communicates state perfectfly. Prusas show as offline (will show as online with no api is api is removed (for purposes of log I will say API but mean new APP key)

[image]
[image]

Logs attached. 
[OctoFarm-Server.log](https://github.com/NotExpectedYet/OctoFarm/files/5064895/OctoFarm-Server.log)
[OctoFarm-Scripts.log](https://github.com/NotExpectedYet/OctoFarm/files/5064903/OctoFarm-Scripts.log)
[OctoFarm-InformationCleaning.log](https://github.com/NotExpectedYet/OctoFarm/files/5064910/OctoFarm-InformationCleaning.log)
[OctoFarm-HistoryCollection.log](https://github.com/NotExpectedYet/OctoFarm/files/5064916/OctoFarm-HistoryCollection.log)
[OctoFarm-State7 (1).log](https://github.com/NotExpectedYet/OctoFarm/files/5064919/OctoFarm-State7.1.log)

API log not posted as contains API!
Filament log not relevant?

NotExpectedYet, 2020-08-12:
That's weird those instances are completely refusing connection to the users api. They're erroring with an  ETIMEDOUT which I have never seen before and I'll have to investigate further. 

I did notice this though: `timeout: 64000` What's your timeout settings set at? I don't think this is any bearing but I just wanna make sure there isn't a bug in that part of the code too.

Crowlord, 2020-08-12:
currently 1 / 5/ 10 / 60 / 60 but I have been doing a lot of tweaking of them. 

I had a major clearout of sd cards after the last one so they shouldnt be particularly file heavy this time especially Jim as I wiped his by accident fairly recently!

NotExpectedYet, 2020-08-12:
@Crowlord that's not bad at all then. Yeah it's the first offline connection which is why it's larger than I originally thought. It's not even been able to get the Auth to even continue on to trying to grab the files. It's basically responding saying the I'm not there. 

I gather octoprint wise there's no real difference between your setups? As with it been different printers is leading me to believe that something is changing in octoprint settings. Weird thing is the line of code it's failing on isn't any different between the two versions. That though would only be correct by completely ignoring this same setup works on earlier versions.

Crowlord, 2020-08-12:
I suspected it may be a plugin but Rosie (192.168.0.140) running in safe mode still had no success. 

Working unit is on a Pi3B+ IP 142
Prusa Mini (ian) is on a Pi3B+ IP 10
both other units are on Pi4 2gbs. IPs 140 and 144

The mini is wired ethernet all the others are wireless and all on official psus. 

Plugins are virtually the same on all of them as i like my instances neat 👍

NotExpectedYet, 2020-08-12:
Could you get me some octoprint logs from when it made the connection please?

Crowlord, 2020-08-12:
[octoprint 142.log](https://github.com/NotExpectedYet/OctoFarm/files/5065131/octoprint.142.log)
Working machine todays log. 

[octoprint 140.log](https://github.com/NotExpectedYet/OctoFarm/files/5065146/octoprint.140.log)
This one is a non working and i did see a 403 error. linked to tornado api

[octoprint 140.log](https://github.com/NotExpectedYet/OctoFarm/files/5065162/octoprint.140.log)
same again my server is on 94 so it does seem to be that. 

[octoprint 144.log](https://github.com/NotExpectedYet/OctoFarm/files/5065175/octoprint.144.log)

NotExpectedYet, 2020-08-13:
I'm wondering if the Prusa printer is changing something in OctoPrint as working one was this:

`2020-08-12 09:33:08,913 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2020-08-12 09:33:18,864 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 179111}
2020-08-12 09:36:00,943 - octoprint.server.util.flask - INFO - Passively logging in user crowlord from ::ffff:192.168.0.94
2020-08-12 09:36:00,949 - octoprint.access.users - INFO - Logged in user: crowlord
2020-08-12 09:36:02,084 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.0.94
2020-08-12 09:36:02,152 - octoprint.server.util.sockjs - INFO - User crowlord logged in on the socket from client ::ffff:192.168.0.94
2020-08-12 09:36:48,507 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.0.94
2020-08-12 09:36:48,828 - octoprint.server.util.flask - INFO - Passively logging in user crowlord from ::ffff:192.168.0.94`
Broken one was this:
`2020-08-12 10:21:12,531 - tornado.access - WARNING - 403 GET /api/access/users (::ffff:192.168.0.94) 10.94ms
2020-08-12 10:21:13,540 - tornado.access - WARNING - 403 GET /api/access/users (::ffff:192.168.0.94) 11.54ms
2020-08-12 10:21:39,246 - tornado.access - WARNING - 403 GET /api/access/users (::ffff:192.168.0.94) 9.46ms
2020-08-12 10:29:56,811 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2020-08-12 10:30:02,416 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 95406}
2020-08-12 10:44:56,813 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2020-08-12 10:45:02,464 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 96306}
2020-08-12 10:46:53,534 - octoprint.plugins.telegram - INFO - Sending a message: Printing at Z=4.0.\nBed 60.1/60.0, Extruder 219.2/220.0.\n11:11:05, 84%% done, 11:11:53 remaining.\nCompleted time 11:11:46. with_image=True with_gif=0 chatID= 321715893
2020-08-12 10:59:56,815 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2020-08-12 11:00:02,460 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 97206}`

I'm not actually connecting to that endpoint weirdly enough. Do you have a later OctoFarm state log that matches with those printer logs? 

Would you also be able to make a quick edit to a file to test? If so: 
`            users = await ClientAPI.getRetry(
                farmPrinters[i].printerURL,
                farmPrinters[i].apikey,
                'users',
            );`
That exists on line 766 of `runners/state.log`
`            users = await ClientAPI.getRetry(
                farmPrinters[i].printerURL,
                farmPrinters[i].apikey,
                'access/users',
            );`

****

NotExpectedYet, 2020-08-13:
What are the user setups like on each? 

Exactly the same?

NotExpectedYet, 2020-08-13:
Checking the logs further I think it might actually be me. It's getting a null value from a field somewhere trying to find it's length. Best bit is, I'm not checking for length where that crops up. If you could let me know your user setup on those problem devices I think it might help me figure out why.

Crowlord, 2020-08-13:
[octoprint-backup-20200813-094212 NOT WORKING.zip](https://github.com/NotExpectedYet/OctoFarm/files/5068018/octoprint-backup-20200813-094212.NOT.WORKING.zip)
[octoprint-backup-20200813-094351 WORKING.zip](https://github.com/NotExpectedYet/OctoFarm/files/5068029/octoprint-backup-20200813-094351.WORKING.zip)

Crowlord, 2020-08-13:
no change on that line adjustment. 

Heres the latest state log

[OctoFarm-State (1).log](https://github.com/NotExpectedYet/OctoFarm/files/5068098/OctoFarm-State.1.log)

NotExpectedYet, 2020-08-15:
Hello, 

It's something user permissions you'll have to correct. 

I looked at the users.yaml for both your provided backups and the working users.yaml has: 
`crowlord:
    active: true
    apikey: null
    groups:
    - admins
    - users
    password: ##############
    permissions: []
    roles:
    - user
    - admin
    settings:
        interface:
            language: _default
`
The none working one has: 
`
crowlord:
    active: true
    apikey: null
    groups:
    - admins
    - users
    password: ##############
    permissions: []
    roles:
    - user
    - admin
    settings:
        interface:
            language: _default
`

Crowlord, 2020-08-15:
I am baffled. I am going to try a manual edit of the yaml files but they should be totally identical. The second i add admin to any user it locks out! Even a new first run wizard does it.

NotExpectedYet, 2020-08-15:
Honestly couldn't tell you man. Wonder if its a bug during upgrade maybe? Have you tried a fresh flash? 

Its a weird one like.

Crowlord, 2020-08-15:
the two pi4s are both as fresh as a daisy. The oldest unaltered one is the working one. 

Users yaml looks exactly the same. I wondered if it was a per user API requirement but the working one has no per user key just the global and the app key.

The reason for the difference on the files i sent may be that I manually added extra permissions to troubleshoot.

NotExpectedYet, 2020-08-15:
Yeah its a weird one as I thought at first the working file shouldn't work. Wonder if that's the new configuration then and the old one kept the older one. 

The global key shouldn't work at all anymore from what I've been told. Albeit a bug but rolling with it. 

Yeah that could screw my theory up ahaha. The api is been denied by that 403 we saw on octoprints side so its certainly something with the api. Have you tried an application key?

Crowlord, 2020-08-15:
If i remove the App Key it connects online with no API message. 

If I add an API key it instantly goes offline
If I add an Application key it also instantly goes offline.

I am leaning towards an octoprint bug but that doesn't gel with the fact that it's only my 3 printers! They all all 1.4.2 on octopi 0.17. Even the python versions are the same. 2.7.16

NotExpectedYet, 2020-08-15:
That's what's confusing and would lead me to think its a settings issue but you've got clean install like you've said. Safe mode would be my next step if you haven't already tried that. 
I know you said they are the same across those devices but would at least rule it out. 
Have you got another device to try connecting over the api with?

Crowlord, 2020-08-15:
I will do some pi swapping and see what happens once my prints are done. 

I can also try a new octofarm instance on another machine. 

I also have a pi zero free so i can try some troubleshooting with a fresh install on that as well. 

I will report back. Food now!

NotExpectedYet, 2020-08-15:
Eat well man, do report back your findings. I'll keep open this ticket for now until we've confirmed something. I have only got a pi3 but I'll test it, need to do something for Gina anyway.

Crowlord, 2020-08-15:
Did a full update/upgrade on all 3. No difference. 

Swapping the working pi to an unworking printer is successful. 

I need to find an SD card for my pi zero to do testing on that.

At the moment it looks like its a software issue probably related to the pis as safe mode makes no difference. 

I am going to install a farm onto my main PC and see if i can get life from that to rule out my server PC.

Crowlord, 2020-08-15:
Great news. SOLVED

It seems to have been a corruption in the mongoDB database. Wiping a printer and readding it fixed it. To be sure i wiped the db and reinstalled the farm. Running well now. 

[image]

closed 2020-08-15 by Crowlord

NotExpectedYet, 2020-08-16:
> 
> 
> Great news. SOLVED
> 
> It seems to have been a corruption in the mongoDB database. Wiping a printer and re-adding it fixed it. To be sure i wiped the db and reinstalled the farm. Added all clients as new ones. Running well now.
> 
> [image]

I'm wondering how that occurred now aha. Glad it's sorted anyway.

commit dce458eb (2020-08-08): fixed connection issues with 1.4.2
Fixed #85
Fixed #89