OctoFarm/OctoFarm #90: Updated to 1.5.5.6: All Printers Offline?

bolsoncerrado, 2020-08-17:
Just upgraded new to 1.5.5.6 (ie rm -rf OctoFarm, git clone, npm install) and now all my printers are "offline" for OctoFarm...

Any clues?

Thanks!

NotExpectedYet, 2020-08-17:
Are you using octoprint 1.4.1+? 

You need to use the user api key or generate an application key for it.

bolsoncerrado, 2020-08-17:
Ah darn!

Any quick tutorial on that?

bolsoncerrado, 2020-08-17:
OK created user Octofarm on each printer, created API for user, assiged OPERATOR permisions and edited printers on Octofarm. Now Octofarm reports printers as Online, but API and WEBSOCKET still remain in RED.

Does octofarm require any specific permissions set on the printer? admin instead of operator?

NotExpectedYet, 2020-08-17:
You need to use Admin permissions for now yes. 

Those changes all came at once with the loss of the API when OctoPrint updated. I will have checks been put in place for future versions to work with any key over the coming weeks.

bolsoncerrado, 2020-08-17:
COol thanks!

bolsoncerrado, 2020-08-17:
OK definetly mondays must be the worst day to handle these kind of things!

Went ahead, added Admin usergroups to each octofarm user on my octoprint instances and now all printers show as SHUTDOWN on octofarm! WTF! ROTFL!!!

Sorry bud!! don't hate me!!

NotExpectedYet, 2020-08-17:
It's ok ahaha. This permissions thing couldn't have come at a worse time for me. 

Could you pass me the latest OctoFarm-State.log please? 

Doesn't seem as simple as I first thought ahaha

bolsoncerrado, 2020-08-17:
[OctoFarm-State.log](https://github.com/NotExpectedYet/OctoFarm/files/5082901/OctoFarm-State.log)

bolsoncerrado, 2020-08-17:
P.S. yes, i don't have all of them ON, just 3 or 4 for test purposes (printing actually, right now)

NotExpectedYet, 2020-08-17:
You've got the EHOST down error. 

Something on those OctoPrint boxes is still disconnecting the connection attempt to users. Maybe they need a restart to update the user permissions, I'm not sure as I'm new to OctoPrints permissions setup myself. 

Could you try completely wiping your printer on OctoFarm and re-adding it. It shouldn't matter that it's printing, and that's what another user did here to resolve the EHOST error. #89

bolsoncerrado, 2020-08-17:
Sure, lemme try that.

bolsoncerrado, 2020-08-17:
What a weirdo. Deleted 2. Readded both. One shows as awaiting, the other shows as half added :?? :?? I made sure both users octofarm are present and API are correctly copy/pasted.

[image]

NotExpectedYet, 2020-08-17:
> What a weirdo. Deleted 2. Readded both. One shows as awaiting, the other shows as half added :?? :?? I made sure both users octofarm are present and API are correctly copy/pasted.
> 
> [image]
> 
> Plus the second reads NO API WTF lol
> 
> Ps. Sent you my hangouts email just in case you wanna chat in realtime to fix this faster?

Sadly I can't I'm at work at the minute. 

I'm at a loss at the minute sadly as well. As far as I can tell it looks like it's still denying the connection. 

I'll run some more tests tonight starting with the old setup -> upgrading for you.

bolsoncerrado, 2020-08-17:
Is there a way to empty the printers db completely to start fresh? i mean not just deleting the rows manually thru the web GUI but resetting the db somehow to the "first time user" state...

NotExpectedYet, 2020-08-17:
> Is there a way to empty the printers db completely to start fresh? i mean not just deleting the rows manually thru the web GUI but resetting the db somehow to the "first time user" state...

Yeah for OctoFarm it's just a case of changing the URL in the database URI field of db.js. 

you'll have something like `https://localhost:27127/octofarm`
You can make a new database by `https://localhost:27127/octofarm2`

bolsoncerrado, 2020-08-17:
OK how do I stop Octofarm? npm stop did not work lol

NotExpectedYet, 2020-08-17:
`pm2 kill`

bolsoncerrado, 2020-08-17:
Well at least starting blank makes it partly coherent... Will await your input then:

[image]

NotExpectedYet, 2020-08-17:
> Well at least starting blank makes it partly coherent... Will await your input then:
> 
> [image]

Could you forward the latest state log again with the new setup? 

Sorry I can't advise more at the moment, I'll certainly get around to checking it out tonight.

bolsoncerrado, 2020-08-17:
[OctoFarm-State.log](https://github.com/NotExpectedYet/OctoFarm/files/5083241/OctoFarm-State.log)

NotExpectedYet, 2020-08-17:
From what the logs is telling me 1 of your printers is getting authenticated through the passive login, then the websocket is purposely failing it with EHOSTDOWN error. 

One of your other printers is just not getting past that at all and is been denied with the same error but at a different point. 

At the moment this is looking like OctoPrint is still denying the request (complete guess until I can confirm properly). 

I know some were printing, but do you have any to try rebooting and see if it's just that from when the permissions we're changed over? I'm wondering if OctoPrint needs something like that to enable them properly. 

It's seeming like the issue stems from a user been created with Observer privileges and then been set to Admin.

bolsoncerrado, 2020-08-17:
Sure, I'll turn on one of the non printing ones so I can reboot it upon creating the octofarm user and lyk.

NotExpectedYet, 2020-08-17:
Brilliant thank you very much!

bolsoncerrado, 2020-08-17:
So I did this:

1) Turned on offline printer+octoprint
2) Created octofarm + api user on octoprint
3) rebooted pi
4) once octoprint was back online, added printer to octofarm:

[image]

bolsoncerrado, 2020-08-17:
I can now add a 5th offline printer and then turn it on and add the api after adding it to octofarm but I have not much faith on that either lol

NotExpectedYet, 2020-08-17:
> I can now add a 5th offline printer and then turn it on and add the api after adding it to octofarm but I have not much faith on that either lol

Ones enough aha. Thanks for trying at least. 

I'll run off and do some testing in a dark corner tonight and see what madness I can produce. I've literally only tested this so far with the default admin user. Have you tried that? I totally understand if you don't want to give those permissions to octofarm btw, and like I said should work with any configuration. Due to having all the settings editing though I do require some lofty permissions at the moment as it will straight up fail the connection if one of those fails with how it was originally wrote. 

I will certainly get back to you as soon as I've tried re-producing your steps.

bolsoncerrado, 2020-08-17:
Some more weirdoo!!

https://www.dropbox.com/s/ybl3yfkt0w5pedk/ofwtf.mov?dl=0

NotExpectedYet, 2020-08-17:
> Some more weirdoo!!
> 
> https://www.dropbox.com/s/ybl3yfkt0w5pedk/ofwtf.mov?dl=0

That last one I will fix tonight :)

Gerry-007, 2020-08-17:
Hello there,

I do have the exact same problem.

I am running:

Version: 1.1.5.6.8
a vm with Ubuntu 18.04

Octoprint 1.4.2 and Application API
I am running four printer on the same octoprint (Raspberry Pi 3 B+)

looks like the API pass the info of the octoprint system (on line) but no info for the printer that result off-line
[OctoFarm-State16.log](https://github.com/NotExpectedYet/OctoFarm/files/5084608/OctoFarm-State16.log)

If I can help with my small knowledge...

Gerry-007, 2020-08-17:
Forget about the log I sent. 

That is the old installation
After going mad, this morning I reinstalled the lot, so the log I sent it refers to an old installation.

G

Gerry-007, 2020-08-17:
Here is the correct one.

G

[OctoFarm-State.log](https://github.com/NotExpectedYet/OctoFarm/files/5084700/OctoFarm-State.log)

Gerry-007, 2020-08-17:
Thank you for all your hard work!

Gerry-007, 2020-08-17:
You would expect that someone after 30 years working in IT, doing this for living, would not make mistake with typo an IP address.
The through is I am getting old!
I was trying to get the octopi on the same IP of the OctoFarm. Embarrassment!!!!

All working fine for me now.
Sending the first print on the prusa now.

I tested both API, the user and the Application and they both work fine.

Sorry to waist you time and again if you need any help......

Thank for the fab job.

G

NotExpectedYet, 2020-08-17:
> You would expect that someone after 30 years working in IT, doing this for living, would not make mistake with typo an IP address.
> The through is I am getting old!
> I was trying to get the octopi on [...]

No, thanks for getting back to me and saying it's fine! Those are the best kind of bugs ;) ahah

It's certainly not fresh installs as all mine auth fine. Also thanks for the kind words!

bolsoncerrado, 2020-08-17:
How do I test "the other" API?

I'm on MacOS fwiw... But I do plan to finally dedicate a miniOC to Octofarm, so it may run on Linux in the end... Or Windows, idk.

NotExpectedYet, 2020-08-17:
> 
> 
> How do I test "the other" API?
> 
> I'm on MacOS fwiw... But I do plan to finally dedicate a miniOC to Octofarm, so it may run on Linux in the end... Or Windows, idk.

The Application key as far as I'm aware uses the User permissions key. 

They are both in the user dropdown settings though if you want to try. 

How do you setup your users? as I gather they arn't setup as the default Admin account? I'll try and reproduce your setup shortly and see if I can get it working.

Gerry-007, 2020-08-17:
> How do I test "the other" API?

Octopi has three API possible

Global, User and Application
Please find the three snapshot of Octoprint below

If I am correct the Global key with the Octopi 1.4.2 is no longer supported.
[image]
[image]
[image]

> 
> I'm on MacOS fwiw... But I do plan to finally dedicate a miniOC to Octofarm, so it may run on Linux in the end... Or Windows, idk.

Gerry-007, 2020-08-17:
I believe that the Application key is a Plugin

NotExpectedYet, 2020-08-17:
> Octopi has three API possible
>
> Global, User and Application
> Please find the three snapshot of Octoprint below
>
> If I am correct the Global key with the Octopi 1.4.2 is no longer supported.
> [image] [...]

Yes only Application and User Keys will work, I think the Application key uses the Users permissions it was created under. Here it seems like the issue is something to do with the settings on the account maybe.

bolsoncerrado, 2020-08-17:
So there are only TWO APIs possible, the global one, now defunct for OctoFarm purposes, and the Access Control per User one, which is the one I moved to today, right?

Then I'm still out of the game. Hehe. 

How weird @Gerry-007 can get it fine. Wondering what's so different between OSs.

The ONLY thing I had to do is an nmp audit fix for 4 "problems" it was reporting upon installing this last version. 
Other than that, idk?

Gerry-007, 2020-08-17:
> So there are only TWO APIs possible, the global one, now defunct for OctoFarm purposes, and the Access Control per User one, which is the one I moved to today, right?
> 

[image]
This is my access control for Admin
I am using the same Admin and password on the OctoFarm

> Then I'm still out of the game. Hehe.
> 
> How weird @Gerry-007 can get it fine. Wondering what's so different between OSs.

I don't think is a question of OS
> 
> The ONLY thing I had to do is an nmp audit fix for 4 "problems" it was reporting upon installing this last version.

I had that too, but if you have the webclient up I think the problem is only of communication to the Octopi.

You can do it!!

> Other than that, idk?

NotExpectedYet, 2020-08-17:
Do you by any chance have two user accounts on your OctoPrint instance? 

Just to clarify that npm audit fix is on some dev packages that I use. It's just warning you because they're in my package.json but they'll be fixed for next pull. 

OS doesn't matter though. It's confusing, I've only seen 1 other person with this issue and it was bugging me with what his was. My clean install works perfectly fine so there must be some configurational difference I can't think what else would cause this.

bolsoncerrado, 2020-08-17:
> [image]
> This is my access control for Admin
> I am using the same Admin and password on the OctoFarm
>
>
> I don't think is a question of OS
>
>
> I had that too, but if you have the webclient up I think the [...]

Ah but then you're not using latest octoprint right???

bolsoncerrado, 2020-08-17:
> Do you by any chance have two user accounts on your OctoPrint instance?
>
> Just to clarify that npm audit fix is on some dev packages that I use. It's just warning you because they're in my [...]

Well as of today, octoprint admin is user 1 and Octofarm admin is user 2 on octoprint, yes. 

As on audit fix, do you mean it does not matter I ran it or it's best I shouldn't have run it?

Gerry-007, 2020-08-17:
> Ah but then you're not using latest octoprint right???

I do.

Version 1.4.2

G

bolsoncerrado, 2020-08-18:
Doh!

I was using an additional user for octofarm, never thought using one different than the main admin would make a difference. Lemme try that.

bolsoncerrado, 2020-08-18:
RIGHT ON!

Using Admin's API makes it work again!!

NotExpectedYet, 2020-08-18:
> RIGHT ON!
> 
> Using Admin's API makes it work again!!

I know why this is as well. OctoFarm is in a position where I would rather make adding a printer as streamlined as possible... so, not asking for user credentials and that, or making you accept the auth request on the web ui... for someone with 60+ printers this is just annoying. This worked perfectly fine to begin with until the Global API key lock out. Now I gather there was some confusion between the key OctoFarm was using and the User it was attempting to login with! 

I will try and make sure this is more streamlined in the future, but honestly thank you for sticking through all this and testing it out. I now can at the very least make recommendations to other users who may face this issue. 

I will close this for now as the original issue has been resolved, and I'll make a not to have a look at a more graceful way of grabbing the specific user without confusing it between two that exist.

bolsoncerrado, 2020-08-18:
Thank you for this piece of software man!!

NotExpectedYet, 2020-08-18:
> Thank you for this piece of software man!!

You are most welcome! Glad its finally working properly :)

closed 2020-08-18 by NotExpectedYet