OctoFarm/OctoFarm #94: Retrieve/push environment data from external source

TheLion, 2020-08-17:
Hi,

Is it possible to retrieve/push data for the environmental temperature and humidity from an external device to Octofarm? In my case an ESP32 with a BME180 module, which publishes the data to MQTT into my Domotica system.

NotExpectedYet, 2020-08-18:
Welcome to been my enviromental tester! I'm currently testing one myself. 

I've not given any instructions out because I'm unsure on whether the layout will stay as it is in the UI. It will also only store the last 9000 records I think as MongoDB isn't time series. 

Push will also be coming, I'm thinking of support influxdb and prometheus exports. 

Anyway, you need to send and authenticated POST request to: 
/input/roomData

Expected the following JSON format: 
`{temperature: value (°C), humidity:value (%), gas_resistance: value (KOhms), date: value}`
Null values are required when no data present.
[image]

TheLion, 2020-08-18:
Thanks!

Will take a look at this. 

Can you share an example what the full POST request looks like?

NotExpectedYet, 2020-08-18:
When I get home tonight I certainly will man no worries. 

What platform are you using to send it?

TheLion, 2020-08-18:
> 
> 
> When I get home tonight I certainly will man no worries.

Great! No hurry..

> What platform are you using to send it?

I'm not sure yet. Have to check what the best system for the job is. The ESP32 can do it I guess, but my Domotica system (Domoticz) can also do it.

NotExpectedYet, 2020-08-18:
Would it be beneficial for me to send you my javascript script? My Pi reads my gpio pins for the bme680 and then fires it off to OctoFarm.

TheLion, 2020-08-18:
Yes, that would be very beneficial. Thanks..

NotExpectedYet, 2020-08-18:
`
bme680.initialize().then(async () => {
    console.info('Sensor initialized');
    setInterval(async () => {
        let bme680Data = await bme680.getSensorData();
//      console.log(bme680Data)
        let temperature = bme680Data.data.temperature;
//      console.log(temperature)
        let pressure = bme680Data.data.pressure;
        let humidity = bme680Data.data.humidity;
        let gas_resistance = bme680Data.data.gas_resistance;
        let date = Date.now();
        let readOut = {
                temperature: temperature,
                pressure: pressure,
                humidity: humidity,
                gas_resistance: gas_resistance,
                date: date
        }
        fetch('http://192.168.1.5:4000/input/roomData', {
                 method: 'POST', // or 'PUT'
                headers: {
                'Content-Type': 'application/json',
                },
                body: JSON.stringify(readOut),
        }).then(e => {
                console.log(e)
        }).catch(e => {
                console.error(e)
        })

    }, 900000);
});

`

NotExpectedYet, 2020-08-20:
The latest dev push changes this a little: 
```
                            {temperature: value (°C),
                                                                                    humidity: value (%),
                                                                                    iaq: value (KOhms),
                                                                                    pressure: value (Pa)
                                                                                    date: value
                                                                                    }
```

TheLion, 2020-09-01:
I haven't gotten around to testing it :-(

bolsoncerrado, 2020-09-07:
Sorry to step in in a somewhat unrelated question: is there any way or place to tun off display of the EH or at least move it BELOW the weekly utilisation graphics? 👍

NotExpectedYet, 2020-09-07:
Appogise for that id pushed it by mistake. Ive been testing a fully edible dashboard which i was hoping to push last weekend but it will be hopefully Friday now. :)

bolsoncerrado, 2020-09-08:
Including the color of the bars? (hint! hint!) :P

NotExpectedYet, 2020-09-08:
> 
> 
> Including the color of the bars? (hint! hint!) :P

At the moment no, the colours represent the rest of the system colours ie red => Offline. Feel free to drop a feature request in at https://features.octofarm.net though and I'll see what I can do :P

TheLion, 2020-10-10:
> date: value

I finally got around to some testing.

Are all variables needed? Cause my current setup doesn't provide 

> 
> 
> The latest dev push changes this a little:
> 
> ```
> {
>   temperature: value (°C),
>   humidity: value (%),
>   iaq: value
>   pressure: value (Pa)
>   date: value
>                                                                                     }
> ```

I finally got around to some testing.

Are all variables mandatory? Cause my current setup doesn't provide pressure or iaq.
And is date mandatory? And if it is, in what format?

NotExpectedYet, 2020-10-11:
> I finally got around to some testing.
>
> Are all variables mandatory? Cause my current setup doesn't provide pressure or iaq.
> And is date mandatory? And if it is, in what format?

Hey man, sorry about my tardy reply. 

It's all mandatory at the minute. Requires null values if you can't provide the information. 

Date is the number of miliseconds since anuary 1, 1970 00:00:00 UTC. I'm not sure what coding platform your writing on but usually there is a method. 

Currently the null values are shown on the graph but I'll be tidying that up for the next release so it will work with what you provide.

NotExpectedYet, 2020-11-24:
Yo, the null values are still required but it will only display what you provide in the next release. It's on dev if you want to try.

TheLion, 2020-11-24:
I got the dev release running, will work on providing the info from my system.
Thanks!

NotExpectedYet, 2020-11-24:
Welcome :) 

I want to show the last values on the views too but I'm not sure where at the moment.

closed 2021-02-01 by github-actions[bot]