OctoFarm/OctoFarm #95: Unraid Docker is not porting data properly

Soukas, 2020-08-19:
Title may be incorrect from a CS standpoint.

I have an unraid server running, with the octofarm docker installed from the community app pool.

I have 4 octoprint servers running on raspberry pi's (2b-4b models), all 4 connect back to the octofarm.
All four allow me to load and print, and pass video feeds through.

Unfortunately, all 4 have Zeros for temp and set-points, even though the text color changes to green indicating that the printer is "hot". The "History" tab is also painfully empty, I have tried every way of loading a print to the printer via octofarm to get something on this list, I am willing to bet I am not doing it right on this one.

All core functions work, I can upload and print, I can cancel prints, and I can control motion and look at local files.

I will include a photo of the missing data, and can provide logs if needed.
[image]

NotExpectedYet, 2020-08-20:
Yeah could you get me the OctoFarm-State.log please. 

That's certainly weird as it's triggering green so the value is certainly there on the client... it's just not been shown. 

Could you also show me the output of F12 console log in the browser? 

This actually looks like OctoFarms doing tbh.

NotExpectedYet, 2020-08-20:
There's an settings page on Printers, could you screen shot the profile of one of your printers? I'll try and replicate that as I believe it's something specific to the profile after looking at the code. Probably something I haven't accounted for.

Soukas, 2020-08-20:
Thank you for the quick reply!

attached are copies of all logs. I just noticed my "History" log file is completely empty, so maybe the file is having permission based issues?
[OctoFarm-API.log](https://github.com/NotExpectedYet/OctoFarm/files/5104549/OctoFarm-API.log)
[OctoFarm-Scripts.log](https://github.com/NotExpectedYet/OctoFarm/files/5104550/OctoFarm-Scripts.log)
[OctoFarm-Server.log](https://github.com/NotExpectedYet/OctoFarm/files/5104551/OctoFarm-Server.log)
[OctoFarm-State.log](https://github.com/NotExpectedYet/OctoFarm/files/5104553/OctoFarm-State.log)

I also attached what I believe is the settings page you asked for. I blacked out the API to secure some private identity
[image]

Soukas, 2020-08-20:
As a follow up, it never shows the file selected. and it doesnt update the "printing" status when a job completes.

Hopefully that helps you sus out the bread down. In the photo attached, both jobs have completed.
[image]

Soukas, 2020-08-20:
I am running the Monolithic Ocotofarm build from here: https://hub.docker.com/r/octofarm/octofarm/
Attached is a photo with most of my d
[image]
ocker settings in unraid for reference.

NotExpectedYet, 2020-08-20:
[image]

This page I meant sorry. 

Everything thing seems fine in the logs. (I've deleted state for you as it contains your API key.) Does the client through up any errors? You need to press F12 in your browser to see the console log. 

Is your instance updated to the latest?

Soukas, 2020-08-20:
Got carried away there. Here is the output log.

Thank you again for all your help
[image]

Soukas, 2020-08-20:
Okay! I hadn't found that page yet, so thanks for pointing it out.

Here is my copy
[image]

Soukas, 2020-08-20:
Unraid lists the Instance as up to date (in photo) but I am running 1.5.5.5 not 1.1.5.6 listed online.

Is there any way to upgrade the instance in webapp? or should I find the maintainer for the docker?
[image]
[image]

NotExpectedYet, 2020-08-20:
On UnRaid go to your Docker page and scroll right down to the bottom. 

Press look for updates if you don't already see one next to your octofarm icon. Mine usually checks every few days.

Soukas, 2020-08-20:
Okay, Just pulled the update. Thank you for teaching me that.
I am running on 1.1.5.6 now, looks great btw.

Here is my current state of affairs, Awaiting websocket with errors in F12
[image]

Soukas, 2020-08-20:
I exported my printers and then removed ender 3 and ender 3 -2 (oldest printer and newst, oldest Pi and Octoprint build vs pulled this week on a pi 4b)

I then attempted to add them new, thinking maybe I was missing a step. The API key needed to be pasted in as an edit to get it to take (no extra spaces or anything, just pressed paste in the edit field, then save and it fixed it). Once it took, it sits at awaiting websocket.

I took screenshots of both parts. 1 where the ender 3 - 1 was rebuilt and ender 3 -2 still needed the API edit.
The second photo is both now waiting for a websocket.
[image]
[image]

I am going to regenerate all my api keys/move to application keys once this is over so no worries on hiding my keys now.

NotExpectedYet, 2020-08-21:
Are you using an Application Key / User Key as advised in the change log? 

Octoprint no longer allows websocket authentication with the Global API key.