OctoFarm/OctoFarm #96: File manager- Can't upload files 1.1.5.6

tdavis75, 2020-08-21:
If I upload a file, the dialog box shows up so to select the file, but it never uploads. After you reviewed the log files you noticed it was failing at the grabbing state. 

Octoprint version is 1.4.2. I only have one account which is the admin. I'm using a manually created API key at the user level.

NotExpectedYet, 2020-08-24:
1. Create Octofarm user in Operator and Admin Group

2. Login as Octofarm user, Generate an application key in the user profile.

3. Open access Controls and remove all other profiles.

4. Update Octofarm with new API key, verify connection works.

5. Log out of Octofarm user account, connection to octofarm will break.

6. Re-sync connection and never login to octoprint again

Another user did those steps and got it to work. I'll be re-factoring the connection stuff for the next version but worth a try currently.

tdavis75, 2020-08-24:
I tried doing what you listed above.  It did not work for me.  I ended up
deleting both printers in Octofarm.  Then I added just the one printer that
we've been working with (Ender-5) using the new profile and application
key.  Things seem to be working at the moment.  I'm going to do the same
thing on my other printer and add it back in and see if everything is
stable after that.

NotExpectedYet, 2020-08-28:
Weird like, I thought OctoFarm might be incorrectly keeping the old key but I can't see it

tdavis75, 2020-09-01:
I had started having problems again shortly after I mentioned what I did.
So I ended up using the API key that was tied specific to the user profile,
not the manually made up one or the one under the main settings. Everything
has been working fine since.  Add that to the weirdness.

[image: image.png]

NotExpectedYet, 2020-09-03:
At least it's working, but yeah both keys should work currently testing with that now.

I'm going to close this in favour of #99 as I believe these are all related to this and yours seems to be working correctly now. Please let me know if it isn't the case.

closed 2020-09-03 by NotExpectedYet

tdavis75, 2020-09-03:
I would agree. That is fine to close this one. Thank you!

commit 88a7d81d (2020-09-20): added new printer select menu, enabled on file manager
Fixed #96