OctoFarm/OctoFarm #103: Spool-Manger integration?

Doprintityourself, 2020-08-26:
As I killed my octopi instance during installation of filament manger and not being supported or udate any more!

Is there a plan to integrate or coopreate with the new spool manager from ollisgit?

https://github.com/OllisGit/OctoPrint-SpoolManager

Would really appreciate that
Thank you

NotExpectedYet, 2020-08-27:
No plans as of yet, I am watching the project though and currently there is no database support. I think the best bet for OctoFarm in this position is to take over control of the previous plugin and connect that to OctoFarms database. Thoughts welcome.

NotExpectedYet, 2020-09-11:
I am taking over maintaining the Filament Manager plugin to keep it alive, 

I'll be adding all the recent work people have tried to commit and getting it updated ready for the plugin store on OctoPrint

closed 2020-09-29 by NotExpectedYet