OctoFarm/OctoFarm #104: One printer keeps flashing offline/online

napter, 2020-08-31:
I added a few printers, all seem fine except for one - it keeps switching between online and offline like so:
[image]

Any idea what I can do to resolve it -or how I can troubleshoot it? The octoprint website and service seem fine.

NotExpectedYet, 2020-09-01:
Usual checks are is CORS activated and are you using the correct api key. 

The release notes have details about recent changes in OctoPrint if this version is 1.4.1+ and OctoFarm can now only connect over Application or User API Keys not the global one. 

Older versions should still work with the global key.

NotExpectedYet, 2020-09-19:
Hello this has been open for some time now with no response. I gather the original issue has been resolved so I'm going to close. If this isn't the case please feel free to re-open. Thanks!

closed 2020-09-19 by NotExpectedYet