OctoFarm/OctoFarm #105: Error when using  octofarm/octofarm:monolithic-latest  - "Cannot read property 'panelView' of undefined"

napter, 2020-08-31:
I am using the  octofarm/octofarm:monolithic-latest docker image. When I click on the panel tab it navigates to /mon/panel and I get this error:

TypeError: /app/views/panelView.ejs:4
    2| <% if(typeof printerCount != 'undefined' && printerCount > 0 ) {%>
    3| 
 >> 4|     <% if(clientSettings.panelView.currentOp){%> <%-
    5|     include('./partials/currentOperations.ejs') %> <% }%>
    6|     <div class="col-12 col-sm-12 col-md-12 col-lg-12 col-xl-12">
    7|       <div id="listView" class="row" data-jplist-group="printers">

Cannot read property 'panelView' of undefined
    at eval (/app/views/panelView.ejs:13:26)
    at panelView (/app/node_modules/ejs/lib/ejs.js:691:17)
    at tryHandleCache (/app/node_modules/ejs/lib/ejs.js:272:36)
    at View.exports.renderFile [as engine] (/app/node_modules/ejs/lib/ejs.js:489:10)
    at View.render (/app/node_modules/express/lib/view.js:135:8)
    at tryRender (/app/node_modules/express/lib/application.js:640:10)
    at Function.render (/app/node_modules/express/lib/application.js:592:3)
    at ServerResponse.render (/app/node_modules/express/lib/response.js:1012:7)
    at ServerResponse.res.render (/app/node_modules/express-ejs-layouts/lib/express-layouts.js:77:18)
    at /app/routes/index.js:169:9
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:93:5)

NotExpectedYet, 2020-09-01:
Can you get onto the System -> Client settings page? 

It seems like the default settings for that page arn't been picked up. If you go to that page, and press "Save" on the client settings it should let you access the panel again.

This shouldn't be happening so I'll take a look and see what's going on.

napter, 2020-09-01:
That fixed it.

NotExpectedYet, 2020-09-02:
Sweet thanks for the report back. I'll see if I can figure out why default settings arn't getting generated.

NotExpectedYet, 2020-09-20:
This should be resolved now in future branches, please let me know if this is not the case.

closed 2020-09-20 by NotExpectedYet