OctoFarm/OctoFarm #108: [BUG]: Cannot read property 'free' of undefined (fileManagerRunner.js)

bretterer, 2020-09-07:
One one of my printers, I can not do anything with the file manager.  I get the message

```
fileManagerRunner.js:256 Uncaught (in promise) TypeError: Cannot read property 'free' of undefined
    at Function.updatePrinterList (fileManagerRunner.js:256)
```

Only way I can make file manager work for my printer again is to fully remove the printer, and then re-add it. it then works fine for a little while.

I have 4 Ender 3 Pro's linked up to the system.  I am using the users application keys

[image]
[image]

I'm not sure which logs will be helpful here, so please let me know and I can add them to the issue

NotExpectedYet, 2020-09-08:
OctoFarm state log please, oh and OctoFarm and OctoPrint versions. 

It's indicating that it's either not finished scanning your files/failed at some point sorry the UI doesn't react properly to the eventuality there seems to be an edge case getting through. 

If you Re-Sync the files of that specific printer for me before sending the logs it should be easy to spot.

bretterer, 2020-09-08:
OctoFarm: 1.1.5.6
OctoPrint: 1.4.2 
Python: 2.7.16 
OctoPi: 0.17.0

State Log:
[OctoFarm-State.log](https://github.com/NotExpectedYet/OctoFarm/files/5189027/OctoFarm-State.log)

Printer in question is ip `10.12.13.142`

Ran re-sync, but no UI update happened to show that re-sync was working or triggered at all

NotExpectedYet, 2020-09-08:
Thanks for that 

Checking through the logs it seems the client in question lost connection and become unreachable:
`Mon, 07 Sep 2020 21:27:08 GMT | ERROR | OctoFarm-State.log | Error: connect EHOSTUNREACH`
When it came back online it looks like OctoPrint wasn't allowing the connection back in:
`Mon, 07 Sep 2020 21:38:36 GMT | ERROR | OctoFarm-State.log | Error: read ECONNRESET` which could just be OctoPrint booting back up. 

I believe this eventually resolved itself and octofarm never re-scanned the files due to them already existing in the database. 

If you'd like to resolve this, you should be able to delete the printer and re-add it again with no adverse effects. History will all still be linked and that as it uses your ip/printer name. 

This is certainly a bug on OctoFarms side. I'm hoping to push the next release this weekend if not Friday so I'll make sure this doesn't happen again in the future.

bretterer, 2020-09-09:
Would you like me to test your `attempted fix`  on my end, or were you able to re-produce it?

Thank you for the quick turn around on this!

NotExpectedYet, 2020-09-09:
If your happy to run the latest development build then please do :) 

I tried to reproduce by deleting the storage key from the database so i believe it should solve the issue. I just havent had chance to investigae how the situation can possibly arrise as those values should be pulled when files are synced. Your system had files but was complaining the storage didnt exist, but i also tweaked the triggers for functions that run the information cleaners for the client with the last push so it shouldnt happen again. 

Thanks for been so quick on this i hadnt gotten around to asking ahaha.

NotExpectedYet, 2020-09-09:
Your most welcome as well!

NotExpectedYet, 2020-09-20:
This should be resolved now, closing please let me know if it's not the case

closed 2020-09-20 by NotExpectedYet