OctoFarm/OctoFarm #109: [BUG] All Printers Offline (Out of Synch Error)

bolsoncerrado, 2020-09-10:
Hi,

I actually dont know how to accurately report this, but this is the scenario:

I bought a small miniPC to install Ubuntu on it, set up Octofarm and everything is now up and running. I leave the PC on all the time, and all the printers connected to it display their ongoing jobs fine... I do my daily job on other computers, and sometimes launch prints using octoprint on other computers instead of using octofarm. I even shutdown this or that printer if I dont plan to use all the farm in the weekend, for example.

Then, when I return to the octofarm server to see how things are going, I find out ALL printers show up as OFFLINE even if I left some online and printing, and I need to RESYNCH manually the Octofarm list of printers in order for it to stay on synch with my farm :? :? :? just not sure why? Isn't it its duty to keep track of whats going on with each printer at every time? Aint this the fun of a server like this?

JUst not sure in which moment it determined all printers went offline when it was never the case.

Any clues?

Thanks!

NotExpectedYet, 2020-09-10:
There are certain conditions where OctoFarm won't re-establish the connection automatically. If a connection attempt is denied by OctoPrint for example. This is usually hit when the Pi's are shutdown. 

If you send me your state log I can tell you what conditions arrose for it to shutdown. 

The printers which you left online I'd need logs too. If they're online then it should stay connected, the only thing that would trigger a disconnect is if the ping/pong was never returned and that's usually due to faulty/intermittened network issues.

bolsoncerrado, 2020-09-10:
Well obviously I ordered certain Pis to shutdown along with the printers I shutdown. But I thought Octofarm would report only those offline, not ALL of them :?

NotExpectedYet, 2020-09-10:
> 
> 
> Well obviously I ordered certain Pis to shutdown along with the printers I shutdown. But I thought Octofarm would report only those offline, not ALL of them :?

This is why I'm asking for the state logs. It's either a bug with a state/response that I've missed or it's intended and it's responding to network issues or even possibly some other unknown thing.

bolsoncerrado, 2020-09-10:
[OctoFarmStates.zip](https://github.com/NotExpectedYet/OctoFarm/files/5200956/OctoFarmStates.zip)

NotExpectedYet, 2020-09-10:
> 
> 
> [OctoFarmStates.zip](https://github.com/NotExpectedYet/OctoFarm/files/5200956/OctoFarmStates.zip)

Thanks. 

It's looking like network issues going off of that. OctoFarm is getting back "Host unreachable" errors so it can't establish a connection with the host device ie the Pi your running it on. That's either because they're offline or there is an issue on the network somewhere between octofarm and your Pi's.

bolsoncerrado, 2020-09-10:
NO, everything is etherneted on a LAN. :???

NotExpectedYet, 2020-09-10:
> 
> 
> NO, everything is etherneted on a LAN. :???

That's even more confusing.... 

Have you got any Octoprint logs around the time when you noticed it?

bolsoncerrado, 2020-09-10:
But which ones, the ones from the machines I turned off? The ones I kept alive? They're still alive and printing.

NotExpectedYet, 2020-09-10:
The ones that are online and printing. 

I want to see if I can see OctoFarms connection attempts going to the instance on the Pi.

bolsoncerrado, 2020-09-10:
this was last night's for example...

[octoprint.log.2020-09-09.zip](https://github.com/NotExpectedYet/OctoFarm/files/5201579/octoprint.log.2020-09-09.zip)

bolsoncerrado, 2020-09-10:
And this is one other it was left online....

[yetanother.zip](https://github.com/NotExpectedYet/OctoFarm/files/5201601/yetanother.zip)

NotExpectedYet, 2020-09-10:
> 
> 
> this was last night's for example...
> 
> [octoprint.log.2020-09-09.zip](https://github.com/NotExpectedYet/OctoFarm/files/5201579/octoprint.log.2020-09-09.zip)

Ah! You've hit the little issue I was notified about in Issue #99. 

For some reason, and I haven't been able to reproduce this but when OctoFarm's passively login in to establish the websocket connection + you also login with the API from the same user on the web interface it seems to close the connection for OctoFarm and prefer the login from the OctoPrint Client. OctoFarm see's this as a hard failure which would give exactly the results you've described as you mentioned using both OctoFarm and the OctoPrint client interface. 

I haven't been able to reproduce this personally on my own systems yet!. Could you give me some information about your current Users on OctoPrint and which Key your using? 

I'll do some further testing on this over the weekend but I believe that's the issue your seeing. 

For now, if I'm correct.... you should just be able to connect with OctoFarm and use that interface, just be aware if you login to the OctoPrint client then you'll need to re-sync on OctoFarm for now. 

Sorry about that like, the original way I was authenticating with OctoPrint got stopped in 1.4.1 and I haven't had time (work/life) to get a better solution in place as of yet. 

If it's a quick fix I'll get it in the next release for Friday/Saturday. If I manage to get a fix before then would you be available to test?

bolsoncerrado, 2020-09-10:
That would make sense, although I only have 1 user (same user) in every and all PIs in the LAN ("admin" user by all means), and Octofarm is using that admins API key (each admin its own API key) to get into each printer's PI...

NotExpectedYet, 2020-09-10:
Thanks, that should be working fine. I'll double check with my instances and see what is happening.

NotExpectedYet, 2020-09-16:
Hello again, 

Would you be able to run a check for me please? 

I've spoken to the developer of OctoPrint and my way of loggin into the device is fine. So it could very well be a plugin that's causing the issue. Could you put OctoPrint into safe mode, and attempt a OctoPrint browser session and login with OctoFarm at the same time and see if it disconnects?

bolsoncerrado, 2020-09-17:
Problem is I cant reproduce the original issue repeatedly... so imagine your proposition!

Also I try to run the system as bare as possible, the only few plugins I run on it are cosmethic, besides touchUI.

[image]

NotExpectedYet, 2020-09-19:
I'm unsure which one of those would specifically cause it but I don't believe this to be an OctoFarm/OctoPrint issue at the moment due to a clean system not showing these symptoms. 

The only one I could gather would possibly do something like that would be TouchUI. Strange one like. If I get chance I'll install TouchUI on one of my testing instances as I'd at least like to make a recommendation regarding this to future users.

bolsoncerrado, 2020-09-21:
What if in the meantime you add a RESYNC button on the Dashboard view? 👍

bolsoncerrado, 2020-09-23:
Well at least I was able to partially replicate the issue: when you install/uninstall any plugin that requires restarting/reloading the server, Octofarm won't reconnect by itself and will consider the printer offline for the remains...

NotExpectedYet, 2020-09-25:
That I haven't personally noticed any difference with either on a clean setup... hmmm... I gather the websocket is closed from OctoPrints side which OctoFarm will see that as a hard failure so not attempt re-syncing. 

Regarding the button, I have plans to update the views with a re-sync button but not the dashboard. I never really intended that for control but I have requests too add other buttons there so I'll see what I can do.

closed 2020-12-05 by github-actions[bot]