OctoFarm/OctoFarm #164: [BUG] After Restart of Host OctoFarm can't start

commanderf, 2020-09-30:
Hi,

I installed octofarm on Windows with Portainer (like in the wiki). It works until I restart the host. If I try to start octofarm I get the following log:

2020-09-30T07:47:17.795418269Z docker/entrypoint.sh: 34: cd: can't cd to app/

2020-09-30T07:47:18.016717367Z pm2 launched in no-daemon mode (you can add DEBUG="*" env variable to get more messages)

2020-09-30T07:47:18.181553165Z 2020-09-30T07:47:18: PM2 log: Launching in no daemon mode

2020-09-30T07:47:18.212555665Z 2020-09-30T07:47:18: PM2 log: [PM2] Starting /app/app.js in fork_mode (1 instance)

2020-09-30T07:47:18.215416465Z 2020-09-30T07:47:18: PM2 log: App [OctoFarm:0] starting in -fork mode-

2020-09-30T07:47:18.224246265Z 2020-09-30T07:47:18: PM2 log: App [OctoFarm:0] online

2020-09-30T07:47:18.232858165Z 2020-09-30T07:47:18: PM2 log: [PM2] Done.

2020-09-30T07:47:18.276332964Z 2020-09-30T07:47:18: PM2 log: ┌─────┬─────────────┬─────────────┬─────────┬─────────┬──────────┬────────┬──────┬───────────┬──────────┬──────────┬──────────┬──────────┐

2020-09-30T07:47:18.276365564Z │ id  │ name        │ namespace   │ version │ mode    │ pid      │ uptime │ ↺    │ status    │ cpu      │ mem      │ user     │ watching │

2020-09-30T07:47:18.276370564Z ├─────┼─────────────┼─────────────┼─────────┼─────────┼──────────┼────────┼──────┼───────────┼──────────┼──────────┼──────────┼──────────┤

2020-09-30T07:47:18.276374664Z │ 0   │ OctoFarm    │ default     │ 1.1.5   │ fork    │ 20       │ 0s     │ 0    │ online    │ 0%       │ 28.0mb   │ root     │ disabled │

2020-09-30T07:47:18.276378064Z └─────┴─────────────┴─────────────┴─────────┴─────────┴──────────┴────────┴──────┴───────────┴──────────┴──────────┴──────────┴──────────┘

2020-09-30T07:47:18.276647064Z 2020-09-30T07:47:18: PM2 log: [--no-daemon] Continue to stream logs

2020-09-30T07:47:18.277421264Z 2020-09-30T07:47:18: PM2 log: [--no-daemon] Exit on target PM2 exit pid=9

2020-09-30T07:47:18.871957459Z 07:47:18 0|OctoFarm  | 2020-09-30T07:47:18: Wed, 30 Sep 2020 07:47:18 GMT | INFO | OctoFarm-InformationCleaning.log | Running history Cleaning | 

2020-09-30T07:47:19.882905849Z 07:47:19 0|OctoFarm  | 2020-09-30T07:47:19: HEAT MAP ISSUE TypeError: Cannot read property '0' of null

2020-09-30T07:47:19.883259449Z 07:47:19 0|OctoFarm  | 2020-09-30T07:47:19:     at Function.heatMapping (/app/lib/dataFunctions/printerClean.js:1084:22)

2020-09-30T07:47:19.883419849Z 07:47:19 0|OctoFarm  | 2020-09-30T07:47:19:     at Function.sortCurrentOperations (/app/lib/dataFunctions/printerClean.js:439:30)

2020-09-30T07:47:19.883511649Z 07:47:19 0|OctoFarm  | 2020-09-30T07:47:19:     at Timeout._onTimeout (/app/lib/dataFunctions/printerClean.js:1185:22)

2020-09-30T07:47:19.883574549Z 07:47:19 0|OctoFarm  | 2020-09-30T07:47:19:     at listOnTimeout (internal/timers.js:531:17)

2020-09-30T07:47:19.883725349Z 07:47:19 0|OctoFarm  | 2020-09-30T07:47:19:     at processTimers (internal/timers.js:475:7)

Can anyone help how I get it work again.

NotExpectedYet, 2020-09-30:
Seems like a bug with the heatmap generation. 

Could you let me know the steps you took as is described in the Issue Template please, I'll need to try and reporduce... i.e. have you added any printers before the restart?, did they connect and show online?

commanderf, 2020-09-30:
To reproduce -> restart host. Didn't do anything else. Then I get this log output (Docker is automatically restarting after reboot including the containers). 
I added 2 printers which worked well.

Using recreation of the container using portainer produces the same error.

But if I delete it and create a new one, the container works and has all stored printers from mongodb

NotExpectedYet, 2020-09-30:
I just meant had you taken any actions on the server, like adding printers before you restarted. I just needed a hint as to where that was in the code, that will do find thank you. I'll double check now with a fresh instances and get back to you shortly.

commanderf, 2020-09-30:
> I just meant had you taken any actions on the server, like adding printers before you restarted. I just needed a hint as to where that was in the code, that will do find thank you. I'll double check now with a fresh instances and get back to you shortly.

No, just used the farm to upload files, start printing and watch the printers

NotExpectedYet, 2020-09-30:
Hello 

I can't currently reproduce this. 

Could you pass me your portainer settings please?

commanderf, 2020-09-30:
I posted you the log from portainer -> log: https://pastebin.com/vf56hMMK

I hope these are all settings you need: https://ibb.co/y8x56d1

NotExpectedYet, 2020-09-30:
Thank you, those pm2 logs arn't the greatest and the original one didn't show this line below: 

09:36:39 0|OctoFarm  | 2020-09-30T09:36:39: Wed, 30 Sep 2020 09:36:39 GMT | ERROR | OctoFarm-Server.log | MongooseServerSelectionError: connect ECONNREFUSED 172.17.0.4:27017 | 

This is your issue, your mongodb instance is refusing the connection causing OctoFarm to not be able to read the database. Is it alive and running before OctoFarm starts?

commanderf, 2020-10-02:
Thanks for your help. After switching to WSL2 and back to hyper-v, the problem is gone (for the moment).

closed 2020-12-23 by NotExpectedYet