OctoFarm/OctoFarm #174: rpi 3

moobi08, 2020-10-05:
good afternoon. when installed on rpi3, it returns the following errors.tell me what is wrong done?

pi@octopi:~/OctoFarm $ npm install

> dtrace-provider@0.8.8 install /home/pi/OctoFarm/node_modules/dtrace-provider
> node-gyp rebuild || node suppress-error.js

make: Entering directory '/home/pi/OctoFarm/node_modules/dtrace-provider/build'
  TOUCH Release/obj.target/DTraceProviderStub.stamp
make: Leaving directory '/home/pi/OctoFarm/node_modules/dtrace-provider/build'

> core-js@2.6.11 postinstall /home/pi/OctoFarm/node_modules/core-js
> node -e "try{require('./postinstall')}catch(e){}"

Thank you for using core-js ( https://github.com/zloirock/core-js ) for polyfilling JavaScript standard library!

The project needs your help! Please consider supporting of core-js on Open Collective or Patreon:
> https://opencollective.com/core-js
> https://www.patreon.com/zloirock

Also, the author of core-js ( https://github.com/zloirock ) is looking for a good job -)

> gifsicle@5.1.0 postinstall /home/pi/OctoFarm/node_modules/gifsicle
> node lib/install.js

  ⚠ Response code 404 (Not Found)
  ⚠ gifsicle pre-build test failed
  ℹ compiling from source
  ✖ Error: Command failed: /bin/sh -c autoreconf -ivf
/bin/sh: 1: autoreconf: not found

    at /home/pi/OctoFarm/node_modules/bin-build/node_modules/execa/index.js:231:11
    at processTicksAndRejections (internal/process/task_queues.js:97:5)
    at async Promise.all (index 0)

> mozjpeg@6.0.1 postinstall /home/pi/OctoFarm/node_modules/mozjpeg
> node lib/install.js

  ⚠ Command failed: /home/pi/OctoFarm/node_modules/mozjpeg/vendor/cjpeg -version
/home/pi/OctoFarm/node_modules/mozjpeg/vendor/cjpeg: 1: /home/pi/OctoFarm/node_modules/mozjpeg/vendor/cjpeg:ELF: not found
/home/pi/OctoFarm/node_modules/mozjpeg/vendor/cjpeg: 2: /home/pi/OctoFarm/node_modules/mozjpeg/vendor/cjpeg: �: not found
/home/pi/OctoFarm/node_modules/mozjpeg/vendor/cjpeg: 3: /home/pi/OctoFarm/node_modules/mozjpeg/vendor/cjpeg: �: not found
/home/pi/OctoFarm/node_modules/mozjpeg/vendor/cjpeg: 4: /home/pi/OctoFarm/node_modules/mozjpeg/vendor/cjpeg: �j�jI�: not found
/home/pi/OctoFarm/node_modules/mozjpeg/vendor/cjpeg: 5: /home/pi/OctoFarm/node_modules/mozjpeg/vendor/cjpeg: �j�j�@@DDP�tdxK: not found
/home/pi/OctoFarm/node_modules/mozjpeg/vendor/cjpeg: 6: /home/pi/OctoFarm/node_modules/mozjpeg/vendor/cjpeg: Syntax error: ")" unexpected

  ⚠ mozjpeg pre-build test failed
  ℹ compiling from source
  ✖ Error: Command failed: /bin/sh -c autoreconf -fiv
/bin/sh: 1: autoreconf: not found

    at /home/pi/OctoFarm/node_modules/bin-build/node_modules/execa/index.js:231:11
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:97:5)
    at async Promise.all (index 0)

> optipng-bin@6.0.0 postinstall /home/pi/OctoFarm/node_modules/optipng-bin
> node lib/install.js

  ⚠ Command failed: /home/pi/OctoFarm/node_modules/optipng-bin/vendor/optipng --version
/home/pi/OctoFarm/node_modules/optipng-bin/vendor/optipng: 1: /home/pi/OctoFarm/node_modules/optipng-bin/vendor/optipng: Syntax error: word unexpected (expecting ")")

  ⚠ optipng pre-build test failed
  ℹ compiling from source
  ✔ optipng built successfully

> nodemon@2.0.4 postinstall /home/pi/OctoFarm/node_modules/nodemon
> node bin/postinstall || exit 0

Love nodemon? You can now support the project via the open collective:
 > https://opencollective.com/nodemon/donate

npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.13 (node_modules/glob-watcher/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.13: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.13 (node_modules/@babel/cli/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.13: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@2.1.3 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.1.3: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: gifsicle@5.1.0 (node_modules/gifsicle):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: gifsicle@5.1.0 postinstall: `node lib/install.js`
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1

added 2043 packages from 895 contributors and audited 2062 packages in 396.627s

78 packages are looking for funding
  run `npm fund` for details

found 1 low severity vulnerability
  run `npm audit fix` to fix them, or `npm audit` for details
pi@octopi:~/OctoFarm $

NotExpectedYet, 2020-10-06:
Next time please don't delete the issue template, it has stuff I request on purpose. 

Your running on octopi? which is a rasbian based software package and doesn't suppose 64-bit for mongoDB for one which is mentioned in the readme/wiki. 

Those are just dependencies, development ones which arn't needed. I'll clean that up for the next release.

moobi08, 2020-10-06:
that is, until the 32 will not work? need a 64-bit package?

NotExpectedYet, 2020-10-06:
> that is, until the 32 will not work? need a 64-bit package?

You can use the Ubuntu 64-bit version for installation on a Pi3B+. 

The database system MongoDB which is a hard requirement for OctoFarm doesn't support 32-bit which is what the rasibian octopi image uses. 

It's also not recommended to install OctoFarm on the same device that you are running OctoPrint from. It's quite low resource use but it has the chance of interfering with a print. 

https://ubuntu.com/download/raspberry-pi

When installing you need to make sure NodeJS is on versions 12.18.4 as the default with Ubuntu currently is much lower. I support the LTS version of node currently. 
MongoDB should be fine with version 3.6+. 

Hope that helps

NotExpectedYet, 2020-11-11:
I am closing due to inactivity please re-open if that's not the case

closed 2020-11-11 by NotExpectedYet