OctoFarm/OctoFarm #175: [BUG] Fresh Debian 10 install, unable to access Octofarm - Incorrect mongo install?

TheKaese, 2020-10-12:
On a fresh debian install, I am unable to access Octofarm. This appears to be due to the method of mongo being installed. Removing the version that was installed via the install guides (https://octofarm.net/installation and the wiki) and replacing it with an install from official instructions https://docs.mongodb.com/manual/tutorial/install-mongodb-on-debian/ resolves this.

Also, the wiki instructions just don't work in general. The scripts are incorrectly formatted and cannot be run. I had to break each of the commands up by ;'s in order to get things to run, else it would just kick back "unable to execute binary file". I'm referring to this from the wiki "curl -sL https://deb.nodesource.com/setup_13.x | sudo -E bash - sudo apt-get install -y nodejs sudo apt install mongodb sudo apt install git"

Also also, Debian 10 doesn't require the last part of the instructions; "
```
...
Output: sudo env PATH=$PATH:/usr/bin /usr/lib/node_modules/pm2/bin/pm2 startup systemd -u pi --hp /home/pi
...
```
npm start ; npm run startup ; npm run saveStartup is enough.

**To Reproduce**
Steps to reproduce the behavior:
1. In a VM or wherever, install the latest version of Debian Buster.
2. Follow the install instructions from the install guide for Debian: https://octofarm.net/installation
3. Attempt to access Octofarm and note that it is unable to connect

**Expected behavior**
Mongo install correctly with the install instructions and Octofarm starts when requested

**Versions (please complete the following information):**
 - NodeJS Version 13.14.0
 - OctoFarm Version 1.1.5.7
 - OctoPrint Version n/a
 - OctoPrint Plugins n/a

NotExpectedYet, 2020-10-12:
Hello, did you resolve this? 

That's my mistake they are the Ubuntu docs. Didn't realise they'd be that different. 

The wiki instructions need cleaning up something for very messed up when copying them over. I'm planning to get those done this week I'll leave this up until I've sorted it. 

That command you said wasn't needed is a pm2 command and what you pasted is described on my installation page as just the output from the previous command. If you dont repaste the command in thats output from the start script generation it wont actually add anything to your start script on your system.

TheKaese, 2020-10-12:
Hey @NotExpectedYet  - Yes, I currently have Octofarm running. Though I am new to Octofarm, so I'm not sure if _all_ the features are functioning. I did realize that I am using Mongo 4.4.1, rather than 4.2.5 per the install scripts, so there is that too.  I can try to install and see if it works with 4.2.5 using the Mongo install. 

As far as the pm2 commands go, it looks like pm2 handles this already by creating a service for pm2:

[image]

NotExpectedYet, 2020-10-12:
> Hey @NotExpectedYet - Yes, I currently have Octofarm running. Though I am new to Octofarm, so I'm not sure if _all_ the features are functioning. I did realize that I am using Mongo 4.4.1, rather than 4.2.5 per the install scripts, so there is that too. I can try to install and see if it works with 4.2.5 using the Mongo install.
> 
> As far as the pm2 commands go, it looks like npm handles this already by creating a service for pm2:
> 
> [image]

No your fine on those version, anything above 3.8 for mongodb. 

Was not aware of that. Ubuntu wasn't as seemless so thanks for that as well I'll update that Debian documentation soon as I get chance.

NotExpectedYet, 2020-10-15:
I have updated the website, thanks again for pointing that out. 

Will close as the original issue has been resolved.

TheKaese, 2020-10-28:
Hey @NotExpectedYet,

You might want to take another look at the instructions that were updated.  They are missing a key element, actually  cloning the repo for Octofarm.  It looks like a copy and paste error :).

Also, in general those instructions still don't work (I tried again on a fresh install).

Better or more complete instructions would be to do this:

1. Install Mongo (https://docs.mongodb.com/manual/tutorial/install-mongodb-on-debian/)
`wget -qO - https://www.mongodb.org/static/pgp/server-4.4.asc | sudo apt-key add -`
 `**(Debian 10)** echo "deb http://repo.mongodb.org/apt/debian buster/mongodb-org/4.4 main" | sudo tee /etc/apt/sources.list.d/mongodb-org-4.4.list`
`**(Debian 9)** echo "deb http://repo.mongodb.org/apt/debian stretch/mongodb-org/4.4 main" | sudo tee /etc/apt/sources.list.d/mongodb-org-4.4.list`
`sudo apt-get update && sudo apt-get install -y mongodb-org`

2. Install the Pre-requisites
`curl -sL https://deb.nodesource.com/setup_13.x | sudo -E bash`
`sudo apt -y install git nodejs`

3. Clone Octofarm and edit /config/db.js

4. CD into the cloned directory and run npm install
`cd Octofarm && npm install`

5. Run Octofarm
`npm start`

Then there's also the stuff about enabling Octofarm to run at boot with npm save,

NotExpectedYet, 2020-10-28:
> Hey @NotExpectedYet,
>
> You might want to take another look at the instructions that were updated. They are missing a key element, actually cloning the repo for Octofarm. It looks like a copy and [...]

My god... seems I was drunk. 

Thanks for those! I'll get that updated shortly. I've re-opened the issue till I update. I need to learn piping commands on cli too as that's much cleaner.

NotExpectedYet, 2020-10-28:
Fixed... 

Thanks for the pointers!

closed 2020-10-28 by NotExpectedYet