OctoFarm/OctoFarm #176: [QUESTION] Klipper / Moonraker support

Arksine, 2020-10-14:
This is less of a question than it is an answer.  It was brought to my attention on the Voron discord that you were interested in supporting a direct connection to Klipper via Moonraker's remote API and you were looking for information and documentation.

The [project is located here](https://github.com/Arksine/moonraker) and you can find the [API documentation here](https://github.com/Arksine/moonraker/blob/master/docs/web_api.md).  There is also a hastily thrown together [test client](https://github.com/Arksine/moonraker/tree/master/test/client) that can be used as a reference.  If you are interested in looking at the source for [Mainsail, you can find it here](https://github.com/meteyou/mainsail).

I do need to create some documentation about Klipper "printer objects", as these are the items that you generally work with to query or subscribe to printer state.  I will be more than happy to go into more detail of that with you and provide you a list of the most useful objects that you will want to query.

If you need to contact me you can do so on Discord.  My user name `Arksine#0730`.  If you prefer you can create an issue in Moonraker's issue tracker, or we can continue the discussion here.

NotExpectedYet, 2020-10-15:
Hello! Thanks for reaching out. 

Yes I am, and I haven't gotten around to it but that saves me some time thanks! 

Here's totally fine for the discussion. 

At the moment OctoFarm isn't ready for another system but I'm working through that slowly but surely. I tagged this as a feature request so it doesn't get staled by my bot and stays open. 

I think I'll be looking into this further with Duet3D when the next release or two has all that work finalised. 

I've taken a quick look and I see no problems currently with how the API works, and you've got websocket support too! Which i see is supported for every command. That's really cool from my perspective as I could just technically stick with websockets unlike OctoPrint where I have to use both the API calls and websockets together. 

I think everything is there currently apart from some form of filament selection that I can currently see but that in this situation would be for OctoFarms purposes more anyway and it's not a requirement to use that feature in OctoFarm nor have it on the client.

NotExpectedYet, 2021-01-04:
Closing in favour of #207

closed 2021-01-04 by NotExpectedYet