OctoFarm/OctoFarm #177: [BUG]

jfkerekes, 2020-10-20:
### To better help the developer understand your issue, please fill in the information below. Failing to do so will waste the developer and your own time in sorting the issue. Thanks!

**Describe the bug**
Unable to add a new printer after update to 1.1.5.7

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Printers
2. Click on Add Printer and fill in all info. 
3. Click save button 
4. terminal window states 
16:3:37 | http://192.168.1.21:80 | Only absolute URLs are supported

**Expected behavior**
Expected to add new printer but never adds

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Versions (please complete the following information):**
 - NodeJS Version [e.g. 13]
 - OctoFarm Version [1.1.5.7]
 - OctoPrint Version [e.g. 1.4.2]
 - OctoPrint Plugins 

**Additional context**
Was able to add printer just before updating from version 1.1.5.5 to 1.1.5.7

NotExpectedYet, 2020-10-21:
Have you also updated OctoPrint at the same time? OctoPrint 1.4.2 requires the use of Application/User API key not the Global API Key.

jfkerekes, 2020-10-21:
I have not. I didn't realize that. I will go back and look at that again. I
just took the update when it showed up.

NotExpectedYet, 2020-10-21:
I've got some basic instructions on here on generating the keys. Either Application or User should work fine. 

https://octofarm.net/installation

I don't think it's documented anywhere for OctoPrint.

jfkerekes, 2020-10-21:
Thanks... I will give that a try and report back.

jfkerekes, 2020-10-21:
So still cannot create a new printer... am using a user API key now... my
other printers are stuck on trying to open websocket connection. I
generated a user APIkey for an existing printer and tried to replace it and
when I Click SAVE button on the Edit Printers menu the button says saving
and just clocks... never completes the save. Also I tried to nuke the
existing Printer database and it never completes. I am wondering if I have
a database connection problem.

jfkerekes, 2020-10-21:
So I have just done a clean installation of OctoFarm on Ubuntu Server 20.04
LTS... Just the base install of Ubuntu Server as a VM on my Unraid
Server... Install went off without an issue seemingly... I can start the
server, however, I cannot get to it from a browser on the local network. I
feel like Imight be missing a dependency but wondered if you have any
suggestions?

Thanks
JF Kerekes

NotExpectedYet, 2020-10-21:
That's usually the mongodb service not running, it doesn't auto start on installation for some weird reason.

NotExpectedYet, 2020-10-21:
Octofarm on the UnRaid community app store BTW :)

jfkerekes, 2020-10-21:
Started Mongodb service with startup and that fixed it. Thanks!!! Can I
still set up PM2 to start Octofarm automatically with server restart?

NotExpectedYet, 2020-10-21:
Brilliant! 

You certainly can. pm2 has documents on it or you can use my npm commands. 

Just copied from my website. 

1. Install pm2 globally (may need sudo)
npm install pm2 -g

2. Generate your system startup script. Some systems may not require the output command pasting back into your terminal, you can just skip to part 4.
npm run startup

Example Output: sudo env PATH=$PATH:/usr/bin /usr/lib/node_modules/pm2/bin/pm2 startup systemd -u pi --hp /home/pi

3. Copy and paste the output of this command into your terminal and run it. Do not copy the example above, this is specific to your system!

4. Save the configuration
npm run saveStartup

jfkerekes, 2020-10-21:
THank you!!! I really appreciate the time you took to answer my questions.

JFKerekes

NotExpectedYet, 2020-10-21:
Your most welcome 

I will close this now as it seems to be resolved, happy printing!

closed 2020-10-21 by NotExpectedYet