OctoFarm/OctoFarm #178: Mobile Browser Missing "File Manager" Tab. [BUG]

jag1251, 2020-10-21:
Don't know if it's just me, but when going to Octofarm on a mobile browser (Tested on Chrome/Android), the "File Manager" icon/link does not show up on the drop down list. When clicking on the drop down list, the list goes from "Printers" to "History". The actual URL works on mobile, it's just not showing up on the drop down list.

NotExpectedYet, 2020-10-21:
Thats intended, the UI doesnt work well on mobile devices so i disabled it. You can still access all of those functions on the printer control modal though, just not the multi file upload. 

Just out of curiousity whats the use case for that? I never expected people would be slicing and uploading from there phones so interested to know if ive missed something. Im planning a improving of the file section so would be good to know.

jag1251, 2020-10-21:
For the "File Management" on mobile, I'd use it more for easily selecting the different printers, seeing what files have been uploaded to each octoprint instance, and letting me pick and chose what printer and what file to start printing. It's a lot more user friendly (in my opinion) to be able to see all the printers/files in one spot, vs selecting the different printers and scrolling down in the "printer control" section.

 I agree that "uploading" from mobile is probably not going to be used often, but I do see some benefits in being able to start prints, move files between instances, and delete files all from one spot on mobile.

NotExpectedYet, 2020-10-21:
Thanks for that understanding perfectly now. 

Well I don't mind enabling it to make it easier for you all. Ill double check what its like and add it in to the next release. 

If you don't want to wait I can point out the two bits you can delete to get it to show.

jag1251, 2020-10-21:
I don't mind waiting for the next release :)

Thank you very much for being so responsive !

scottypres, 2020-12-22:
Hi! File manager is not coming up in the dropdown at all for me in Chrome or Firefox. This is on mobile and desktop. Going directly to the url does work however. Id love to get this working as managing files for 14 printers is time consuming. Thank you for the great program!

NotExpectedYet, 2020-12-22:
> 
> 
> Hi! File manager is not coming up in the dropdown at all for me in Chrome or Firefox. This is on mobile and desktop. Going directly to the url does work however. Id love to get this working as managing files for 14 printers is time consuming. Thank you for the great program!

It's getting added back into the next release as explained above.

scottypres, 2020-12-22:
> It's getting added back into the next release as explained above.

My mistake, I thought it was enabled for desktop and disabled for mobile

NotExpectedYet, 2020-12-22:
> My mistake, I thought it was enabled for desktop and disabled for mobile
My mistake, desktop should be showing File Manager certainly.

NotExpectedYet, 2020-12-22:
> My mistake, I thought it was enabled for desktop and disabled for mobile

I gather you we're using a smaller screen size? possibly 800x600? 

Sorry about that. I've just updated the view to stop disabling filemanager for any screensize. Should work fine now if you pull the latest master version. 

Will close this. 

Please re-open if it's not the case.

closed 2020-12-22 by NotExpectedYet

scottypres, 2020-12-22:
That seemed to solve it. Thank you!

NotExpectedYet, 2020-12-22:
> 
> 
> That seemed to solve it. Thank you!

Welcome, my fault thought I'd only blocked mobiles but I'd blocked smaller screens too aha.

NotExpectedYet, 2020-12-22:
@jag1251 should be working on Mobile now for you too on latest master :)

commit 3447685a (2020-11-12): Fixed #178