OctoFarm/OctoFarm #179: Error message on importing the MongoDB Public Key[BUG]

ambcmb, 2020-10-23:
### To better help the developer understand your issue, please fill in the information below. Failing to do so will waste the developer and your own time in sorting the issue. Thanks!

**Describe the bug**
This is the first time attempting to install OctoFarm. I installed a fresh version of Raspberry Pi Ubuntu 64-bit. I am following the steps for Linux (Ubuntu) on https://octofarm.net/installation.

On Step 5 I get: "Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8))."

On Step 5.1 (since I got an error) - sudo apt install gnupg: 
Reading package lists... Done
Building dependency tree
Reading state information... Done
gnupg is already the newest version (2.2.20-1ubuntu1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

Step 5.1 continued - wget -qO - https://www.mongodb.org/static/pgp/server-4.4.asc | sudo apt-key add -
Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).

Step 5.3 - 
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package mongodb-org

**Versions (please complete the following information):**
 - NodeJS Version [e.g. 13] - v12.19.0
 - OctoFarm Version [e.g. 1.5.5.7] - n/a (i have not gotten that far yet)
 - OctoPrint Version [e.g. 1.4.1] - n/a (i have not gotten that far yet)
 - OctoPrint Plugins [e.g. bed visualiser, octoklipper, printtimegenius] - n/a (i have not gotten that far yet)

NotExpectedYet, 2020-10-24:
Try this? 

`curl -fsSL https://www.mongodb.org/static/pgp/server-4.4.asc | sudo apt-key add -`

I'm confused as to why that doesn't work, unless the formatting got messed up when I copied from mongodb's website. I'll double check. 

I'll make the documentation more clear about that error it's not correct, the error specifically states about gnupg. 

Just for future reference this is a documentation issue, not an actual bug with the application too.

NotExpectedYet, 2020-10-24:
https://www.digitalocean.com/community/tutorials/how-to-install-mongodb-on-ubuntu-20-04

You may be better served following this until I update the site later today.

ambcmb, 2020-10-25:
Thanks for the response. Sorry for labeling it as a bug inappropriately.

Using the link provided I did get past that step. 

When I get to step 10 is get this:
npm start
npm ERR! code ENOENT
npm ERR! syscall open
npm ERR! path /home/ubuntu/package.json
npm ERR! errno -2
npm ERR! enoent ENOENT: no such file or directory, open '/home/ubuntu/package.json'
npm ERR! enoent This is related to npm not being able to find a file.
npm ERR! enoent

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/ubuntu/.npm/_logs/2020-10-25T00_42_50_796Z-debug.log

Any ideas?

NotExpectedYet, 2020-10-25:
No trouble! 

You need to CD in to the OctoFarm folder, the one that was git cloned. Your running the start commands from the home directory currently. I gather that's in /home/ubuntu/Octofarm from the error path in that output. Would depend where you git pulled originally. 

[image]

ambcmb, 2020-10-26:
Turns out I had the wrong version of 64bit ubuntu for the raspberry.

NotExpectedYet, 2020-10-26:
> 
> 
> Turns out I had the wrong version of 64bit ubuntu for the raspberry.

There anything I can point out in the steps to mitigate that in the future?

ambcmb, 2020-10-26:
I used the link off of the webpage. Maybe you can state to use the LTS version. 

Honestly, I was just going too fast and grabbed the latest version (highest number) without really reading it. After a few hours of troubleshooting, I slowed down and started over and caught my mistake.

NotExpectedYet, 2020-10-26:
> 
> 
> I used the link off of the webpage. Maybe you can state to use the LTS version.
> 
> Honestly, I was just going too fast and grabbed the latest version (highest number) without really reading it. After a few hours of troubleshooting, I slowed down and started over and caught my mistake.

No trouble, I shall add that too :) thanks for the feedback.

closed 2020-10-28 by NotExpectedYet