OctoFarm/OctoFarm #182: [BUG] Not installing on Ubuntu 20.04 for Raspberry Pi (missing dependencies)

ChiefGyk, 2020-11-09:
Following the guide here I realized build-essential, autoconf, zlib1g-dev, and some other things were left out

```
> optipng-bin@6.0.0 postinstall /home/chiefgyk3d/OctoFarm/node_modules/optipng-bin
> node lib/install.js

  ⚠ Command failed: /home/chiefgyk3d/OctoFarm/node_modules/optipng-bin/vendor/optipng --version
/home/chiefgyk3d/OctoFarm/node_modules/optipng-bin/vendor/optipng: 1: @@8�@@@@@@@��@@@@����: not found
/home/chiefgyk3d/OctoFarm/node_modules/optipng-bin/vendor/optipng: 1: ELF: not found
/home/chiefgyk3d/OctoFarm/node_modules/optipng-bin/vendor/optipng: 2: d: not found
/home/chiefgyk3d/OctoFarm/node_modules/optipng-bin/vendor/optipng: 1: Syntax error: ";" unexpected

  ⚠ optipng pre-build test failed
  ℹ compiling from source
  ✖ Error: Command failed: /bin/sh -c ./configure --with-system-zlib --prefix="/home/chiefgyk3d/OctoFarm/node_modules/optipng-bin/vendor" --bindir="/home/chiefgyk3d/OctoFarm/node_modules/optipng-bin/vendor"

Checking for gcc...
Using pre-configured libpng...
Checking for system zlib...
./configure: error: missing zlib or incorrect zlib version
./configure: note: zlib version 1.2.1 or higher is required

    at /home/chiefgyk3d/OctoFarm/node_modules/bin-build/node_modules/execa/index.js:231:11
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:97:5)
    at async Promise.all (index 0)

> gifsicle@5.1.0 postinstall /home/chiefgyk3d/OctoFarm/node_modules/gifsicle
> node lib/install.js

  ⚠ Response code 404 (Not Found)
  ⚠ gifsicle pre-build test failed
  ℹ compiling from source
  ✖ Error: Command failed: /bin/sh -c autoreconf -ivf
/bin/sh: 1: autoreconf: not found

    at /home/chiefgyk3d/OctoFarm/node_modules/bin-build/node_modules/execa/index.js:231:11
    at processTicksAndRejections (internal/process/task_queues.js:97:5)
    at async Promise.all (index 0)
npm WARN babel-loader@8.1.0 requires a peer of webpack@>=2 but none is installed. You must install peer dependencies yourself.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.13 (node_modules/@babel/cli/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.13: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@2.1.3 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.1.3: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.13 (node_modules/glob-watcher/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.13: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: optipng-bin@6.0.0 (node_modules/optipng-bin):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: optipng-bin@6.0.0 postinstall: `node lib/install.js`
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: gifsicle@5.1.0 (node_modules/gifsicle):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: gifsicle@5.1.0 postinstall: `node lib/install.js`
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1

audited 2062 packages in 42.941s

78 packages are looking for funding
  run `npm fund` for details

found 1 low severity vulnerability
  run `npm audit fix` to fix them, or `npm audit` for details
```

and even then it still failed after installing a few of the aforementioned packages.

NotExpectedYet, 2020-11-11:
This isn't a bug it's a documentation issue. 

I didn't write that documentation so I can't say for sure but I have Ubuntu 20.04 installation guide here: 
https://octofarm.net/installation

As far as I'm aware there should be no need for those extra packages on a base installation.

NotExpectedYet, 2020-11-11:
There is an error in that I've just found though. 

`curl -sL https://deb.nodesource.com/setup_12.x | sudo -E bash sudo apt -y install git nodejs`
should be: 
`curl -sL https://deb.nodesource.com/setup_12.x |  sudo apt -y install git nodejs`

NotExpectedYet, 2020-11-20:
Hello, there was a missing dependency in the package.json for requests.

closed 2020-12-23 by NotExpectedYet