OctoFarm/OctoFarm #183: Added a 5th printer - will not add

jfkerekes, 2020-11-11:
### To better help the developer understand your issue, please fill in the information below. Failing to do so will waste the developer and your own time in sorting the issue. Thanks!

**Describe the bug**
I created a new printer... would be number 5 in my farm... When I tried to save I got connection log message "Only absolute URLs are supported". The printer record is in the database I think but it does not show in the list. The printer count shows 5 but only 4 are showing. The addition of the 5th printer has created problems with the other printers. None of the camera feeds will show now and the other printers do not show in any of the views except for the printer list. I completed a print on one of the printers yesterday and it registered in the dashboard as a completed print which it let me harvest. I tried to NUKE the printers database but that does not seem to work either. Any help would be appreciated. What steps do I need to take to correct or remove the printer that is causing the issues or to just nuke the printers database and start over. Would really rather correct what is there.

**To Reproduce**
Steps to reproduce the behavior:
Create a new printer by copying and pasting info from one of the other printers and changing IP address info.. received error in log "Only absolute URLs are supported"

**Expected behavior**
a 5th Printer to be added
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Versions (please complete the following information):**
 - NodeJS Version 
 - OctoFarm Version 1.1.5.7
 - OctoPrint Version 1.4.2
 - OctoPrint Plugins 

**Additional context**
Add any other context about the problem here.
[image]

NotExpectedYet, 2020-11-11:
Hello thanks for all that. 

It looks like somethings gone weird when you added the printer but I'm not sure specifically what. Do you have the State.log file for the time that was added please? 

Yeah you will need to delete the database entry for Printers but silly thing is there is currently a bug in that section. It's a typo and I've just pushed a quick fix for it. If you pull the latest master and re-start Octofarm it should let you nuke the printer database. I'll leave this ticket open and edit the title more applicable to whats going on for other users as there seems to be something wrong when you copied and pasted and it's put in some HTML. The new version has those boxes removed for standard inputs already so it won't happen when that's released but it's going to be a bit before I can release it. For now I either recommend importing the IP's or typing them in to mitigate this in the future. It seems it's not cleaned the HTML from when you copied and pasted.

jfkerekes, 2020-11-11:
So I am still a bit new when it comes to pulling the latest updates... can
you tell me what the commands are to pull the latest master from Ubuntu
Server 20.04... sorry to ask

NotExpectedYet, 2020-11-11:
Hey it's no trouble. 

Cd into the OctoFarm directory

'git pull' 

'pm2 restart OctoFarm'

Should do it.

jfkerekes, 2020-11-11:
Should that be pm2 restart app.js?  When I start the service, it is always app.js... just making sure.

jfkerekes, 2020-11-11:
If that is the case, after pulling the latest and restarting,  I still cannot nuke the printer DB

NotExpectedYet, 2020-11-11:
Its all started through npm start that fires up the pm2 service that's installed so yeah you should be using that command if you want to restart the OctoFarm service when saving settings and such. 

Did you start the OctoFarm service differently?

jfkerekes, 2020-11-11:
When I did the clean install, I started Octofarm the first time using npm start... after that I installed PM2 and then ran pm2 start app.js... after that I stopped all services and ran pm2 -update and then pm2 start app.js so that I would have only the app.js instance running. pm2 save and the service starts automatically on restart.

jfkerekes, 2020-11-11:
Either way, I nuked everything as all I had since I did the fresh build on Ubuntu Server was a few prints in the history and 4 printers.... I have rebuilt the printers now and it is working normally again. Thanks again for all you help and patience.

NotExpectedYet, 2020-11-11:
No worries glad it's working. I've improve the printer manager a hell of a lot for the next release so should stop it happening in the future. 

I'd be careful with using any of the System stuff starting your app that way or at least restart the OctoFarm service manually as my mom start script specifies a name for pm2 which is used in the backend to control the service. 

Well you can but just do `pm2 app.js --name "OctoFarm" and you should be fine.

commit 86b8f042 (2020-11-12): Fixed #183
closed 2020-12-23 by NotExpectedYet