OctoFarm/OctoFarm #185: server start issue linux mint

fp30, 2020-11-15:
Hi,

I've followed the install instructions but am getting stuck at the npm start step. It looks like the server is starting but my last line isnt '> node app.js > production.log 2> productionError.log' as it describes in the instructions. Instead when I run npm start I see:
> octofarm@1.1.5 prestart /home/faheem/OctoFarm
> npm install --production

and it tries to install every time. By the end of the install I see:

[PM2] Applying action restartProcessId on app [OctoFarm](ids: [ 0 ])
[PM2] [OctoFarm](0) ✓
[PM2] Process successfully started
┌────┬────────────────────┬──────────┬──────┬───────────┬──────────┬──────────┐
│ id │ name               │ mode     │ ↺    │ status    │ cpu      │ memory   │
├────┼────────────────────┼──────────┼──────┼───────────┼──────────┼──────────┤
│ 0  │ OctoFarm           │ fork     │ 0    │ online    │ 0%       │ 23.5mb   │
└────┴────────────────────┴──────────┴──────┴───────────┴──────────┴──────────
which makes me think its running but when I go to the server address Firefox just tells me the connection was reset.

In the pm2.log file I see loads of this repeated:
2020-11-15T22:19:09: HEAT MAP ISSUE TypeError: Cannot read property '0' of null
2020-11-15T22:19:09:     at Function.heatMapping (/home/faheem/OctoFarm/lib/dataFunctions/printerClean.js:1088:22)
2020-11-15T22:19:09:     at Function.sortCurrentOperations (/home/faheem/OctoFarm/lib/dataFunctions/printerClean.js:443:30)
2020-11-15T22:19:09:     at Timeout._onTimeout (/home/faheem/OctoFarm/lib/dataFunctions/printerClean.js:1189:22)
2020-11-15T22:19:09:     at listOnTimeout (internal/timers.js:554:17)
2020-11-15T22:19:09:     at processTimers (internal/timers.js:497:7)

Any help would be appreciated. Thanks

gmccauley, 2020-11-16:
Do you have an OctoFarm-Server.log file in the logs directory?  If so, can you post it?

fp30, 2020-11-16:
I do... Just checked - turns out MongoDB service wasnt started - oops. Just started it and tried again but still not loading. server log attached.
Thanks

[OctoFarm-Server (copy).txt](https://github.com/NotExpectedYet/OctoFarm/files/5545130/OctoFarm-Server.copy.txt)

NotExpectedYet, 2020-11-16:
> I do... Just checked - turns out MongoDB service wasnt started - oops. Just started it and tried again but still not loading. server log attached.
> Thanks
> 
> [OctoFarm-Server (copy).txt](https://github.com/NotExpectedYet/OctoFarm/files/5545130/OctoFarm-Server.copy.txt)
> 

That log is still suggesting your database details are incorrect/failing to connect. Did you restart OctoFarm' after starting the database? 

What's your url look like in config.js?

fp30, 2020-11-16:
> That log is still suggesting your database details are incorrect/failing to connect. Did you restart OctoFarm' after starting the database?
>
> What's your url look like in config.js?

I did restart the Octofarm after starting the database. I dont know why it still runs npm install --production when I run npm start though. Is that normal?

config.js? Which config.js should I look in? In db.js I have the URI set to:
MongoURI: 'mongodb://localhost:27017/octofarm'

since both octoprint and mongodb are running on the same laptop

NotExpectedYet, 2020-11-16:
Hmm ok wasn't mentioned so just wanted to rule those out. 

Have you tried restarting the system? 

What are your nodejs and mongo versions too? 

Its certainly not connecting for some reason to the Mongodb database. Do you also have the pm2.error.log it should be in the /logs folder in OctoFarm.

fp30, 2020-11-16:
node version: v14.15.0
mongo version: v4.4.1

I have rebooted the system since installing Octofarm and Mongo

Fresh log files after restarting mongo and octofarm attached
[OctoFarm-Server.log.txt](https://github.com/NotExpectedYet/OctoFarm/files/5549560/OctoFarm-Server.log.txt)
[pm2.error.log.txt](https://github.com/NotExpectedYet/OctoFarm/files/5549561/pm2.error.log.txt)
[pm2.log.txt](https://github.com/NotExpectedYet/OctoFarm/files/5549562/pm2.log.txt)

NotExpectedYet, 2020-11-16:
Thanks I'll take a look at the logs in the morning for you I'm about to go to bed. 

One thing. Nodejs 14 isn't supported, I only support the LTS version of. 12 is recommended. There are funnily enough database issues with 14 as well so worth a try until I can look at the logs.

fp30, 2020-11-16:
> Thanks I'll take a look at the logs in the morning for you I'm about to go to bed.
> 
> One thing. Nodejs 14 isn't supported, I only support the LTS version of. 12 is recommended. There are funnily enough database issues with 14 as well so worth a try until I can look at the logs.

downgraded to nodejs 12 still the same. Hopefully you can make some sense of it in the morning!

NotExpectedYet, 2020-11-17:
You have two versions of OctoFarm running man, or something else on your server starting on port 4000. 

I don't think you correctly stopped your old instance by the looks. 

Try 
`pm2 list`

You should see a OctoFarm instance, if so use 
`npm run stop`

Then do the `npm start` and you should be able to access the interface. Apart from that the server seemed to start fine.

fp30, 2020-11-17:
It seems you are right. Something is running on port 4000. A command called nxd - is that node? If not, I cant figure out what it is! And it's restarting automatically.

faheem@faheem: sudo lsof -i :4000
COMMAND  PID USER   FD   TYPE  DEVICE SIZE/OFF NODE NAME
nxd     2353   nx    3u  IPv4 5928504      0t0  TCP *:4000 (LISTEN)
nxd     2353   nx    4u  IPv6 5927461      0t0  TCP *:4000 (LISTEN)
faheem@faheem:~$ sudo kill -9 2353
faheem@faheem:~$ sudo lsof -i :4000
COMMAND   PID USER   FD   TYPE   DEVICE SIZE/OFF NODE NAME
nxd     24513   nx    3u  IPv6 11516857      0t0  TCP *:4000 (LISTEN)
nxd     24513   nx    4u  IPv4 11517317      0t0  TCP *:4000 (LISTEN)

pm2 list shows the old instance is stopped (I think):

faheem@faheem-HP-EliteBook-8470p:~/OctoFarm$ pm2 list
┌────┬────────────────────┬──────────┬──────┬───────────┬──────────┬──────────┐
│ id │ name               │ mode     │ ↺    │ status    │ cpu      │ memory   │
├────┼────────────────────┼──────────┼──────┼───────────┼──────────┼──────────┤
│ 0  │ OctoFarm           │ fork     │ 15   │ stopped   │ 0%       │ 0b       │
└────┴────────────────────┴──────────┴──────┴───────────┴──────────┴──────────┘
thanks for the help. I guess if I can figure out what the nxd command is that's listening on port 4000 I might be able to fix it. Alternatively, is there a way to make OctoFarm listen on a different port?

fp30, 2020-11-17:
> It seems you are right. Something is running on port 4000. A command called nxd - is that node? If not, I cant figure out what it is! And it's restarting automatically.
>
> faheem@faheem:~ $ sudo lsof [...]

I found it! I have some software called NoMachine running and it turns out that is the nxd process that was listening on port 4000. I've reset that to listen on a different port and OctoFarm works!

Thanks again for the help. I did follow your instructions on https://octofarm.net/ the only thing missing was teh start mongodb service but that might have been just because I rebooted and it hadnt restarted automatically.

closed 2020-11-17 by fp30

NotExpectedYet, 2020-11-17:
> I found it! I have some software called NoMachine running and it turns out that is the nxd process that was listening on port 4000. I've reset that to listen on a different port and OctoFarm works! [...]

I use the same software! It does my nut in it runs on 4000 ahahaha. Brilliant, glad you got it sorted. 

Yeah my instructions are a bit muddled at minute I appologise for that. That should also have the section on restarting MongoDB but I've only updated that in the past week.