OctoFarm/OctoFarm #186: [BUG] Octofarm not working with Octoprint 1.5.0 (RC2)

Prutsium, 2020-11-18:
### To better help the developer understand your issue, please fill in the information below. Failing to do so will waste the developer and your own time in sorting the issue. Thanks!

**Describe the bug**
Octofarm cant login runs into websocket api issues

Running Ocotoprint 1.5.0(RC2) on a Raspberry Pi and Octofarm 1.1.5.7 on a Ubuntu Server.
Having 2 Octoprint installations one with 1.4.x and this works perfect but the one running 1.5.0(RC) seems not to be able to get connection via the Websocket

I have tried Global & User Api key but none work.
[pm2.log](https://github.com/NotExpectedYet/OctoFarm/files/5558569/pm2.log)
[OctoFarm-Server.log](https://github.com/NotExpectedYet/OctoFarm/files/5558572/OctoFarm-Server.log)
[OctoFarm-API.log](https://github.com/NotExpectedYet/OctoFarm/files/5558574/OctoFarm-API.log)

Attached various log.

Crowlord, 2020-11-18:
Just to be awkward can confirm no issues on RC2 for my test rig as yet

Octopi 0.17, Octoprint 1.5.0rc2, Python 3.7.3

Octofarm 1.1.5.7 Windows 10

NotExpectedYet, 2020-11-18:
This isn't with 1.5.0 the pm2 logs is riddled with errors. 

You've got no requests module and somethings failing with the heat map. 

Where did you get the installation from and how did you set it up?

Prutsium, 2020-11-18:
> 
> 
> This isn't with 1.5.0 the pm2 logs is riddled with errors.
> 
> You've got no requests module and somethings failing with the heat map.
> 
> Where did you get the installation from and how did you set it up?

I installed according to the Ubuntu guide you made.
I had to install Requests module manually so the log might be clogged with old stuff. Please take a look from only today in the log.

It runs and works on the 1.4.2 Octoprint but at the moment i add the RC2 version it get's stuck on the Web Api socket.

As you can see Prusa works as described and Ender is stuck.
[image]

Prutsium, 2020-11-18:
Here is a bit more recent part of the "State" log:
Wed, 18 Nov 2020 08:25:59 GMT | INFO | OctoFarm-State.log | Session Response | data:{"size":0,"timeout":0} | 
Wed, 18 Nov 2020 08:25:59 GMT | INFO | OctoFarm-State.log | sessionKey JSON: | data:{"_is_external_client":false,"active":true,"admin":true,"apikey":null,"groups":["admins","users"],"name":"_api","needs":{"group":["users","admins"],"role":["files_delete","monitor_terminal","timelapse_download","connection","plugin_announcements_manage","plugin_logging_manage","timelapse_delete","files_upload","settings_read","status","timelapse_list","gcodeviewer","plugin_pluginmanager_manage","files_select","plugin_backup_access","plugin_action_command_notification_clear","plugin_action_command_notification_show","settings","plugin_pluginmanager_install","webcam","admin","files_download","plugin_appkeys_admin","plugin_announcements_read","control","plugin_firmware_check_display","system","plugin_softwareupdate_check","plugin_action_command_prompt_interact","print","plugin_softwareupdate_update","files_list","slice","plugin_pi_support_check","timelapse_admin"]},"permissions":[],"roles":["user","admin"],"session":"BC7126F4FDBA43B09D1D1C015B9FB001","settings":{},"user":true} | 
Wed, 18 Nov 2020 08:25:59 GMT | INFO | OctoFarm-State.log | Attempting to connect to API: system/commands | http://192.168.99.6 | timeout: 10000 | 
Wed, 18 Nov 2020 08:25:59 GMT | INFO | OctoFarm-State.log | Successfully grabbed System Information for...: http://192.168.99.6 | 
Wed, 18 Nov 2020 08:25:59 GMT | INFO | OctoFarm-State.log | Attempting to connect to API: settings | http://192.168.99.6 | timeout: 10000 | 
Wed, 18 Nov 2020 08:25:59 GMT | INFO | OctoFarm-State.log | Successfully grabbed Settings for...: http://192.168.99.6 | 
Wed, 18 Nov 2020 08:25:59 GMT | INFO | OctoFarm-State.log | Attempting to connect to API: printerprofiles | http://192.168.99.6 | timeout: 10000 | 
Wed, 18 Nov 2020 08:25:59 GMT | INFO | OctoFarm-State.log | Successfully grabbed Profiles.js for...: http://192.168.99.6 | 
Wed, 18 Nov 2020 08:25:59 GMT | INFO | OctoFarm-State.log | Attempting to connect to API: connection | http://192.168.99.6 | timeout: 10000 | 
Wed, 18 Nov 2020 08:25:59 GMT | INFO | OctoFarm-State.log | Successfully grabbed Current State for...: http://192.168.99.6 | 
Wed, 18 Nov 2020 08:25:59 GMT | INFO | OctoFarm-State.log | Successfully grabbed Files for...: http://192.168.99.6 | 
Wed, 18 Nov 2020 08:25:59 GMT | INFO | OctoFarm-State.log | Closed websocket connection for: http://192.168.99.4:5000 | 
Wed, 18 Nov 2020 08:25:59 GMT | INFO | OctoFarm-State.log | Attempting to connect to API: users | http://192.168.99.4:5000 | timeout: 10000 | 
Wed, 18 Nov 2020 08:26:00 GMT | INFO | OctoFarm-State.log | WebSocketClient: open | data:{"0":0} | 
Wed, 18 Nov 2020 08:26:00 GMT | INFO | OctoFarm-State.log | Sending Auth to Websocket: 0: ws://192.168.99.6/sockjs/websocket  | data:{"auth":"admin:BC7126F4FDBA43B09D1D1C015B9FB001"} | 
Wed, 18 Nov 2020 08:26:00 GMT | INFO | OctoFarm-State.log | users:  | data:{"users":[{"active":true,"admin":true,"apikey":"32D0301EF6C74009A90F504BE6B2B177","groups":["admins","users"],"name":"Octofarm","needs":{"group":["admins","users"],"role":["plugin_pluginmanager_manage","slice","plugin_firmware_check_display","webcam","status","settings_read","timelapse_delete","files_delete","files_list","plugin_softwareupdate_check","timelapse_admin","plugin_logging_manage","plugin_action_command_notification_show","system","plugin_softwareupdate_update","plugin_announcements_read","plugin_backup_access","control","monitor_terminal","plugin_pi_support_check","admin","settings","plugin_action_command_notification_clear","plugin_pluginmanager_install","files_upload","plugin_announcements_manage","plugin_action_command_prompt_interact","print","gcodeviewer","timelapse_download","plugin_softwareupdate_configure","connection","timelapse_list","files_select","plugin_appkeys_admin","files_download"]},"permissions":["CONTROL","FILES_DELETE","FILES_UPLOAD","ADMIN","FILES_DOWNLOAD","CONNECTION","FILES_LIST","FILES_SELECT"],"roles":["user","admin"],"settings":{},"user":true},{"active":true,"admin":true,"apikey":"9404608DD7174EC7B31B34C270F10E5E","groups":["admins","users"],"name":"St0fzuiger","needs":{"group":["admins","users"],"role":["plugin_pluginmanager_manage","slice","plugin_firmware_check_display","webcam","status","settings_read","timelapse_delete","files_delete","files_list","plugin_softwareupdate_check","timelapse_admin","plugin_logging_manage","plugin_action_command_notification_show","system","plugin_softwareupdate_update","plugin_announcements_read","plugin_backup_access","control","monitor_terminal","plugin_pi_support_check","admin","settings","plugin_action_command_notification_clear","plugin_pluginmanager_install","files_upload","plugin_announcements_manage","plugin_action_command_prompt_interact","print","gcodeviewer","timelapse_download","plugin_softwareupdate_configure","connection","timelapse_list","files_select","plugin_appkeys_admin","files_download"]},"permissions":[],"roles":["user","admin"],"settings":{"interface":{"language":"_default"}},"user":true}]} | 
Wed, 18 Nov 2020 08:26:00 GMT | INFO | OctoFarm-State.log | Chosen user: | data:"St0fzuiger" | 
Wed, 18 Nov 2020 08:26:00 GMT | INFO | OctoFarm-State.log | Session Response | data:{"size":0,"timeout":0} | 
[... cut ...] ings","currentUser","octoPrintVersion","currentIdle","settingsApperance","camURL","apikey","group","storage","fileList"]} | 
Wed, 18 Nov 2020 08:26:42 GMT | INFO | OctoFarm-State.log | Error Saving Counters, Safe to ignore... | data:{"version":0,"modifiedPaths":["dateAdded","powerSettings","currentIdle","currentActive","selectedFilament","tempTriggers","feedRate","flowRate","sortIndex","printerURL","costSettings","camURL","octoPrintVersion","currentOffline","currentUser","storage","fileList"]} |

NotExpectedYet, 2020-11-18:
Nothing really jumps out at me at the moment. 

How come your using port 5000 for one and not the other out of interest? 

One is also login in with the admin user so I gather that doesn't have access control enabled? The other one does? 

Considering other users have it all working fine with 1.5.0 RC2 I'm leaning to some configurational difference / error. 

Or there's something gone awry in OctoFarms edit function and it's not connecting with the updated details. Have you tried removing the printer and adding it again at all?

Prutsium, 2020-11-18:
Have tried to remove the printer and add it again was exact the same.

The port is different as it was a manual install of Octoprint but i freed up the Pi now and can re-install it with Octopi OS since this also runs on the second one that works..

Will report back when i have done that but yes assume it would work then.

NotExpectedYet, 2020-11-18:
> Have tried to remove the printer and add it again was exact the same.
>
> The port is different as it was a manual install of Octoprint but i freed up the Pi now and can re-install it with Octopi OS [...]

Well that at least rules out OctoFarm. 

It shouldn't make a difference either way like but that's makes sense. 

Aye let me know. If it doesn't could you pass the octoprint logs of around the time when OctoFarm tries to connect?

Prutsium, 2020-11-18:
Ok made a fresh install of the concerning Pi (Octoprint / Octopi) and run into the same issue :)
But was also due digging out the logs to sort it out.

It seems in 1.5 the need is there for application key's and especially to the user that logs into Octofarm the same user must be added to Octoprint and an Application key under that user must be generated.
Not sure if i missed those steps in the manual before but now i have the same user on Octofarm and Octoprint and within Octoprint wihlst been logged in as that user i created an User App key and all started to work.

NotExpectedYet, 2020-11-18:
The user you create on OctoFarm has no bearing on OctoPrint. It's just used to log you in. That Application keys requirement has been there since 1.4.1 too so I'm not too sure what happened.  OctoFarm when it tries to log on grabs a list of users and looks for the first admin user it can find and use that to generate the websocket handshake. So as long as you've only a single user on OctoPrint and you use that user to generate an application key/user key you should be fine. Although I haven't fully tested this on 1.5.0 it seems to still be the case from previous reports.

Prutsium, 2020-11-18:
There might be a difference when upgrading vs Fresh install as in RC3 user login is forced / mandatory.
Not sure if you come from 1.4.x towards 1.5.x if those things changes or kept as legacy.

NotExpectedYet, 2020-11-18:
That's a good point. Think you might be right there and I've got my 1.5.0 update ready to test that but I haven't coded the function into OctoFarm yet aha. 

I was just reading about it and if you don't have a user originally on 1.4.2 when you upgrade it forces you to create a user supposedly. Dunno if that maybe didn't trigger for you because I gather that would leave it in limbo.

NotExpectedYet, 2020-11-20:
Hi I'm going to close this as I believe it has been resolved. 

This is something people need to be aware in with OctoPrint not OctoFarm really. Feel free to re-open if this isn't the case.

closed 2020-11-20 by NotExpectedYet