OctoFarm/OctoFarm #187: [BUG] Printer Edit/Add buttons gone

Trogdor04, 2020-11-23:
### To better help the developer understand your issue, please fill in the information below. Failing to do so will waste the developer and your own time in sorting the issue. Thanks!

**Describe the bug**
I switched the network my OctoFarm Server was on and the Add/Edit Printer button was gone.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to OctoFarm
2. Click on Printers

**Expected behavior**
Be able to add printers and/or edit

**Screenshots**
[image]

**Versions (please complete the following information):**
 - NodeJS Version [e.g. 13]
 - OctoFarm Version 1.1.5.7
 - OctoPrint Version 1.4.2
 - OctoPrint Plugins N/A

**Additional context**
Add any other context about the problem here.

NotExpectedYet, 2020-11-23:
"I switched the network my OctoFarm Server was on and the Add/Edit Printer button was gone."

I am not sure how that is relevant but could you explain a little more about this and your network setup please? it seems something isn't right with your install. Maybe missing files, F12 in the browser may show some more information if you could post that as well please?

Trogdor04, 2020-11-23:
Sure, I am a teacher and I am trying to set a print farm.  I used a small wireless router to test everything before I put it on our school's network.  Everything was working great so I shut it down and switched it to the school's network and that is when the buttons disappeared.  Here is the inspector from my browser.
[image]

NotExpectedYet, 2020-11-23:
Did you login to a different account by anychance? The first user that get's created becomes admin the other users can't access those buttons although the printers should be showing so I only mention to double check as I've never seen this before. 

What you described with your network shouldn't make a difference unless your school is actively blocking Javascript or something similar. 

If you haven't changed your installation at all, ie.. MongoDB and OctoFarm haven't moved then I don't think this is an OctoFarm issue sadly. 

If you could click the console tab on that screenshot sorry and send that through I might be able to advise further.

Trogdor04, 2020-11-23:
Is there a way to create a new admin?

[image]

Trogdor04, 2020-11-23:
It was the Admin issue...  I'm so sorry for wasting your time!

NotExpectedYet, 2020-11-23:
No trouble at all! Glad it's working. 

I've made that more clear in the UI for the next release as I don't specifically state it so you couldn't have known.

Trogdor04, 2020-11-23:
I do keep getting this error when trying to connect with my Pi's.  I've only gotten to connect with no issues once.

[image]

NotExpectedYet, 2020-11-23:
Trailing slash, remove that should be fine

Trogdor04, 2020-11-23:
Your Awesome!!!

closed 2020-11-23 by Trogdor04