OctoFarm/OctoFarm #189: [BUG] Bad configuration assumption regarding websockets.

gothrek22, 2020-11-27:
**Describe the bug**
You probably shouldn't assume that websockets run unencrypted here: https://github.com/NotExpectedYet/OctoFarm/blob/2ff38bc967469e8a415ce9da551c34b08c09e182/runners/state.js#L858

A better way would be to either make websocket endpoint configurable to use ssl (wss://) via a checkbox or handle 301 redirects correctly.

**To Reproduce**
Have everything redirected to ssl on a reverse proxy web server and use OctoFarm. Try to configure a printer. Have it error out with `Whoopsy! Big error...` ....

**Expected behavior**
Follow redirect. Seems to be a simple flag provided to `ws` lib, called `followRedirects`, default is false, switch to true in constructor.

**Screenshots**
``` wscat -c ws://test.example.com/sockjs/websocket
error: Unexpected server response: 301```

**Versions (please complete the following information):**
Doesn't matter

**Additional context**
Nope

gothrek22, 2020-11-27:
probably changing line #785 is enough and the line should be like this:
`const ws = new WebSocketClient({followRedirects: true});`

NotExpectedYet, 2020-11-28:
Hey man thanks for that! I'm quite new to a lot of all this so I wasn't aware of the follow redirects option. 

I'll probably plan this into the re-write of the websockets system because I want to add in support for more devices and it's currently pretty locked into OctoPrints configurations.

NotExpectedYet, 2020-11-28:
Thinking about it that's if the clients arn't also ask which wouldn't be the case for you. Should be able to drop that follow redirects in just need some time to test it.

gothrek22, 2020-11-29:
Cheers man! I did my change manually but haven't had the time to test it yet. 

CORS settings are pretty simple on OctoPrint side, if you're using Octopi just add the following 4 lines to `frontend public`:

```
  http-response set-header Access-Control-Allow-Origin "*"
  http-response set-header Access-Control-Allow-Headers "Origin, X-Requested-With, Content-Type, Accept, Authorization, JSNLog-RequestId, activityId, applicationId, applicationUserId, channelId, senderId, sessionId"
  http-response set-header Access-Control-Max-Age 3628800
  http-response set-header Access-Control-Allow-Methods "GET, DELETE, OPTIONS, POST, PUT"```

Keep in mind that setting Access-Control-Allow-Origin to "*" is wildly insecure and should be set to the domain name of octofarm only.

NotExpectedYet, 2020-11-30:
> Cheers man! I did my change manually but haven't had the time to test it yet.
>
> CORS settings are pretty simple on OctoPrint side, if you're using Octopi just add the following 4 lines to `frontend [...]

Oh yeah I know! OctoFarm should not be put anywhere on the internet, it's certainly not safe for that at the minute. I plan on re-working it but ti's just getting time at the moment. I've responded on reddit anyway.

commit 98c92d8f (2020-11-28): Fixed #189 add follow redirect option to web sockets
commit cec58972 (2020-11-28): Fixed #189 actual fix for that bug... added to completely the wrong place
closed 2020-12-23 by NotExpectedYet