OctoFarm/OctoFarm #191: [BUG] on Dev 1.1.6 : Failure to create user on new install

yeldarb420, 2020-12-15:
**Describe the bug**
On a fresh install of development branch v1.1.6, There is a failure on submission of the new user form.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'home page of fresh install (no users)'
2. Click on 'register'
3. Fill out form & Submit
4. See error

**Expected behavior**
`_ReferenceError: /home/ubuntu/OctoFarm/views/register.ejs:8
    6|       </h1>
    7|       <%- include('./partials/messages') %>
 >> 8|       <% if(userCount === 0){ %>
    9|         <div class="alert alert-warning" role="alert">
    10|           This looks like your first user... This user will become the Administrator and is the only one allowed to add/remove/edit printers. Any subsequent users created will have the User permission set. (this will be enhanced at a later date.)
    11|         </div>

userCount is not defined
    at eval (/home/ubuntu/OctoFarm/views/register.ejs:13:8)
    at register (/home/ubuntu/OctoFarm/node_modules/ejs/lib/ejs.js:691:17)
    at tryHandleCache (/home/ubuntu/OctoFarm/node_modules/ejs/lib/ejs.js:272:36)
    at View.exports.renderFile [as engine] (/home/ubuntu/OctoFarm/node_modules/ejs/lib/ejs.js:489:10)
    at View.render (/home/ubuntu/OctoFarm/node_modules/express/lib/view.js:135:8)
    at tryRender (/home/ubuntu/OctoFarm/node_modules/express/lib/application.js:640:10)
    at Function.render (/home/ubuntu/OctoFarm/node_modules/express/lib/application.js:592:3)
    at ServerResponse.render (/home/ubuntu/OctoFarm/node_modules/express/lib/response.js:1012:7)
    at ServerResponse.res.render (/home/ubuntu/OctoFarm/node_modules/express-ejs-layouts/lib/express-layouts.js:77:18)
    at /home/ubuntu/OctoFarm/server_src/routes/users.js:58:13
    at Layer.handle [as handle_request] (/home/ubuntu/OctoFarm/node_modules/express/lib/router/layer.js:95:5)
    at next (/home/ubuntu/OctoFarm/node_modules/express/lib/router/route.js:137:13)
    at Route.dispatch (/home/ubuntu/OctoFarm/node_modules/express/lib/router/route.js:112:3)
    at Layer.handle [as handle_request] (/home/ubuntu/OctoFarm/node_modules/express/lib/router/layer.js:95:5)
    at /home/ubuntu/OctoFarm/node_modules/express/lib/router/index.js:281:22
    at Function.process_params (/home/ubuntu/OctoFarm/node_modules/express/lib/router/index.js:335:12)_`

**Screenshots**
[image]

**Versions (please complete the following information):**
 - NodeJS Version [12.20]
 - OctoFarm Version [1.1.6]

NotExpectedYet, 2020-12-15:
Cheers for that! Snafu, will get it sorted.

NotExpectedYet, 2020-12-15:
Sorted. There's a dev 13 with the following changes: 

1.1.6-dev13
`Fixed issue with been unable to create new user on fresh install`

commit 2da481fd (2020-12-15): Fixed #191
closed 2020-12-23 by NotExpectedYet