OctoFarm/OctoFarm #192: [FEATURE REQUEST] Down machine descriptive note

ihateu3, 2020-12-24:
As with others, many times when a printer goes down due to a clog, shorting thermistor wire, etc. that printer ends up staying down until I get the time to repair it, or sometimes it is waiting on parts.

I would like an easy way to attach a down status note (like a sticky note of sorts) to any of the machines so that I could put things in such as "clogged", "bed thermistor issues", "bed needs to be leveled" "waiting on heartbreak to arrive", etc.   Possibly even add the ability to put hyperlinks in, so you could attach a tracking number in the note for easy reference.  This note would ideally be automatically dated and also automatically have an entry made in the historical notes log.  The note would be visible for the machine it is attached to from the list, panel view, printers tab, etc. 

The purpose of this request is to quickly and easily identify which machines are down, from what cause, and possibly add a tracking number for ordered parts.  This will also very beneficial for those of us that have "helpers" or employees, to let them know not to start a certain machine due to its "down" status, and also a place for them to leave a status when needed.

NotExpectedYet, 2020-12-24:
I think this would be perfect for #144

NotExpectedYet, 2021-01-04:
Closing in favour of #204

closed 2021-01-04 by NotExpectedYet