OctoFarm/OctoFarm #194: [BUG] printers & groups lists empty on printer admin screens

tonysurma, 2020-12-28:
### To better help the developer understand your issue, please fill in the information below. Failing to do so will waste the developer and your own time in sorting the issue. Thanks!

**Describe the bug**
Just upgraded to 1.1.6.  Cannot Edit Printers or any of these actions on the printers screen as no printers are listed in the pop up screen and the group list is empty.

Screen of Actions
[image]
Pop up Screen (edit as example)
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Printers'
2. Click on any action like 'Edit'
3. Screen that pops up lists no printers and the groups drop down is empty.  Everywhere else in system the printers are listed and the groups are named.
4. See error

**Expected behavior**
Printers would be listed (as would groups) and I could perform the next steps.

**Versions (please complete the following information):**
 - octofarm/octofarm latest docker image 1.1.6
 - OctoFarm Version 1.1.6
 - OctoPrint Version 1.5.2

tonysurma, 2020-12-29:
appears to be something related to loading from the database for display as I am also not seeing any history shown.  All worked until I upgraded to 1.1.6

NotExpectedYet, 2020-12-29:
Gonna need some server logs man, pm2.error.log/state and server will do. Something's gone awry but not a clue what. What OS and how did you update also? 

Not experienced this myself personally no other reports as of yet.

tonysurma, 2020-12-29:
Thanks, this is boggling me as well.

Log Files (wasn't sure which ones so let me know what I missed) Note: I also truncated them to just this week when I upgraded as they were kinda big otherwise
[pm2.error.log](https://github.com/NotExpectedYet/OctoFarm/files/5751384/pm2.error.log)
[OctoFarm-Server.log](https://github.com/NotExpectedYet/OctoFarm/files/5751389/OctoFarm-Server.log)
[OctoFarm-HistoryCollection.log](https://github.com/NotExpectedYet/OctoFarm/files/5751390/OctoFarm-HistoryCollection.log)

What OS: Debian Linux (Synology)

How I upgraded: I did the same thing I've done for all the upgrades here so was surprised to have an issue. I download the latest image of octofarm/octofarm.  Stop the container, clear it (resets running image) and then start it up.

console log:
GET http://nas.surma.family:4000/images/bg.jpg 404 (Not Found)
printerManagementRunner.min.js:5559 Uncaught (in promise) TypeError: e.selectedFilament.forEach is not a function
    at printerManagementRunner.min.js:5559
    at Array.forEach (<anonymous>)
    at printerManagementRunner.min.js:5554
    at l (printerManagementRunner.min.js:17125)
    at Generator._invoke (printerManagementRunner.min.js:17355)
    at Generator.next (printerManagementRunner.min.js:17180)
    at i (printerManagementRunner.min.js:5454)
    at a (printerManagementRunner.min.js:5456)

NotExpectedYet, 2020-12-29:
Can you grab earlier docker image tag for now? 

I won't get chance to look till morning, but that seems likes it's something to do with filament manager but I ain't changed anything there so even more confused.

NotExpectedYet, 2020-12-29:
Its not causing the issue but it's not finding the bg image strangely as well. 

Have you got a storage mounted for that?

tonysurma, 2020-12-29:
I have /app/images mounted (saw item in the new version pages about that but didn't have it before)

I have never set the bg image so presume it is using the default?

I will try reverting to earlier docker image as you said.  Thanks again

NotExpectedYet, 2020-12-29:
Ahh makes sense, if your bothered upload one in the UI. For some reasons it's not downloaded the default. 

Yeah well just thought it keeps you running I've got enough info to go off from man thanks for the reports. I'll give you a shout hopefully in the morning with a quick fix. 

Do you have any filament on your printers? I have a decent idea already if you don't aha

Your welcome too.

tonysurma, 2020-12-29:
I have filament manager 1.6.3 installed on both printers hooked to this instance of octofarm.  I have filaments 'loaded' into its database as well as set via octoprint UI to be loaded into each tool (one is a prusa w/ MMU the other is just a prusa)

Okay I went back to previous version.  Most of the issues above were with current version (the new printers popup and the plugins etc.) so can't really test in this version but no errors in console.  However, I still see no history...

NotExpectedYet, 2020-12-29:
I'm even more confused now. Nothings changed in the database really on this version so the history one must be unrelated. 

I also use filament manager with the plugin and don't experience this so my first assumption might not be right either. 

Thanks for the extra information.

tonysurma, 2020-12-30:
I am going to try some stuff on my side today.  As it is boggling me as well.  I am going to to run through what changed on my side since the last update as I had downtime from printing on these printers for a month ish.  I'll report back if I learn anything.

tonysurma, 2020-12-30:
Here is my current theory.  If you stole the webcams off of your octoprint instances (let's say because more folks are working/learning from home at your house than normal) and you never unchecked the webcam button in octoprint it appears that octofarm gets all kinds of confused with that.  As it (rightfully so) expects images and cameras and such to appear and when they don't it gets bad data.

I am only 66% sure that is the issue but it is the one thing I just now changed and everything seemed to be starting to work again

NotExpectedYet, 2020-12-31:
Really? 

I recently put some checks into that area to actually react to octoprints settings. 

Would be interested in hearing the exact steps to reproduce.

NotExpectedYet, 2020-12-31:
I've checked through your logs and should have resolved everything with the next push. 

If that's not the case or anything is still amis then re-open.

closed 2020-12-31 by NotExpectedYet

tonysurma, 2021-01-05:
Thanks and I will try the next push you made to see if that works.

FYI once I made sure I had webcams hooked up again it made everything work better (history etc.) both on Current - 1 version as well as then updating to the Current version.

I have had to print items so I didn't test if I later removed the cameras to see if it regresses for sure but I imagine it would go back to not working.

I'll update if I see something more once I try with missing cameras but for not the newest version and all the features are working great.

thanks again

commit f13c8789 (2020-12-31): Fixed #194