OctoFarm/OctoFarm #195: Missing GCode File

botbuilder1, 2020-12-29:
Just set up OF 1.1.6 on a RPI 4.  Connected to my first two open printers, each with OP 1.5.2.

On one of the printers, the file list on OF was missing one of my most recent GCODE files that I have been printing.  I can start the print job on the OP console, but it still does not show up on the OF file list even when type the file name in the filter:

[image]

ihateu3, 2020-12-30:
Did you click "Re-Sync"?

NotExpectedYet, 2020-12-31:
Could you please not delete the issue template and fill it out correctly and I will help you with the issue.

botbuilder1, 2020-12-31:
Its there now after a "ReSync"

closed 2020-12-31 by botbuilder1