OctoFarm/OctoFarm #196: Can't connect to OctoFarm Server After Installing PM2 and Generating the Startup Script[BUG]

brooksware2000, 2020-12-30:
### To better help the developer understand your issue, please fill in the information below. Failing to do so will waste the developer and your own time in sorting the issue. Thanks!

**Describe the bug**
I get an error that there is an issue with the MongoDB Database after installing PM2, generating the startup script, and restarting the server.  I am running the 20.04.1 LTS 64-bit version of the Ubuntu server on a RaspberryPi 4.

**To Reproduce**
Steps to reproduce the behavior:
1. Follow the steps to setup OctoFarm on a RaspberryPi 4 Ubuntu Server.
2. Install OctoFarms dependencies and start the service.  (OctoFarm can be accessed from a browser).
3. Install PM2 globally, generate the startup script, and save it.  (OctoFarm is not accessible but, is showing as started and running after a system reboot).
4. Get an error that there is an issue with the MongoDB Database and I cannot connect to the OctoFarm server.

**Versions (please complete the following information):**
 - NodeJS Version [14.15.3]
 - OctoFarm Version [1.1.6]
 - OctoPrint Version [1.5.2]

NotExpectedYet, 2020-12-30:
I think that's because your running nodejs 14 not 12 LTS like is specified in the instructions.

Could you make sure the Mongodb service is actually running and try downgrading your node instance please? 

It should resolve it. 

If not I'll need the pm2.error.log file.

NotExpectedYet, 2020-12-30:
Also be worth to double check and see if OctoFarm is running twice or not.

NotExpectedYet, 2020-12-30:
I gather something didn't happen right when you generated the start script as that should have no bearing on an already running service it just takes over managing it.

brooksware2000, 2020-12-30:
> I think that's because your running nodejs 14 not 12 LTS like is specified in the instructions.
> 
> Could you make sure the Mongodb service is actually running and try downgrading your node instance please?
> 
> It should resolve it.
> 
> If not I'll need the pm2.error.log file.

Ok, I'll downgrade nodejs and post results.

brooksware2000, 2020-12-31:
Ok, I redid the setup steps and sort of got things to work but, I cannot get the startup script to function.  Here is what I have to do in order to get OctoFarm to function:

1.  After a server restart or power off, I have to log back in (obvious step)
2. The MongoDB database does not autostart so, I have to enter the following: sudo systemctl start mongod
3. The OctoFarm dependencies does not autostart so, I have to enter the following: npm start
4. PM2 gives me a message saying what's loaded in memory is out of date (In memory PM2 version: 4.5.1) (Local PM2 version 4.5.0) and recommends I update it.
5. I enter the following: pm2 update
6. Finally, I have to restart app.js before OctFarm connects to the MongoDB server and functions.  I enter the following: sudo pm2 restart app.js

Here are the steps I attempted to create the autostart script:
1.  I Followed these steps from the OctoFarm instructions:  
  npm install pm2 -g
  npm run startup

I do not receive any output to copy and paste

npm run saveStartup

This method does not work for me so, I attempt the following:

pm2 start app.js
pm2 startup systemd

I copy and paste the results

pm2 save

I reboot after attempting both methods to get the autostart script to work and neither works.

Any recommendations?  I appreciate your assistance.

Curtis

NotExpectedYet, 2020-12-31:
I've just double checked the installation instructions for you and there we're some errors. I'm just updating the site but I would do the following: 

Restart the server - Should clear out any errant items
Make sure mongodb is running - `sudo systemctl status mongod` - start it if not. 

cd into the OctoFarm folder you had before. 

`sudo npm install pm2 -g` (shouldn't error out this time)
`sudo npm run startup` (should generate the code, shouldn't need to copy and paste the output. )

`pm2 list` (confirm you see OctoFarm running)

`sudo run saveStartup` (save the startup script)

Hopefully that should get you up and running, just done the same for another user :) let me know if not.

brooksware2000, 2020-12-31:
> I've just double checked the installation instructions for you and there we're some errors. I'm just updating the site but I would do the following:
> 
> Restart the server - Should clear out any errant items
> Make sure mongodb is running - `sudo systemctl status mongod` - start it if not.
> 
> Try this if mongo isn't starting correctly on boot:
> `sudo rm /var/lib/mongodb/mongod.lock`
> `mongod --repair`
> `sudo service mongodb start`
> 
> cd into the OctoFarm folder you had before.
> 
> `sudo npm install pm2 -g` (shouldn't error out this time)
> `sudo npm run startup` (should generate the code, shouldn't need to copy and paste the output. )
> 
> `pm2 list` (confirm you see OctoFarm running)
> 
> `sudo run saveStartup` (save the startup script)
> 
> Hopefully that should get you up and running, just done the same for another user :) let me know if not.
> 
> Sorry about those issues too, my fault!

Attempted your steps and here are the results:

I restarted the server, checked to see if mongodb was running and it wasn't, I got some weird output but was able to clear it by starting a new SSH session, and using the following command to start up mongodb:

sudo systemctl start mongod

Then verified it was functioning with:

sudo systemctl status mongod

Next, I tried your recommended steps if mongo isn't starting correctly at boot:

sudo rm /var/lib/mongodb/mongod.lock
mongod --repair
sudo service mongodb start  (I got an error "Failed to start mongodb.service:  Unit mongodb.service not found."

So, I attempted the remainder of your recommendations:

cd into the OctoFarm folder you had before.

sudo npm install pm2 -g (shouldn't error out this time)
sudo npm run startup (should generate the code, shouldn't need to copy and paste the output. )

pm2 list (confirm you see OctoFarm running)

sudo run saveStartup (save the startup script) -- This did not work so I entered: sudo npm run saveStartup

I rebooted the server and I am still having the same issues as before.

brooksware2000, 2020-12-31:
No luck with the procedure you recommended.  I have been able to get OctoFarm to work after restarting the server by logging in and entering the following commands:

sudo service mongod start
cd OctoFarm   (changing to the OctoFarm directory)
sudo pm2 restart app.js  (restarting app.js and it states the version of pm2 is not the same as what's loaded in memory)
pm2 update  (enter this command to update to what's loaded in memory and I am able to get OctoFarm loaded).

I'm not sure where else to go from here but, I appreciate your software and assistance.  In fact, I signed up as a patreon backer yesterday.

Thanks,

Curtis

NotExpectedYet, 2020-12-31:
It's appreciated man. Will get it sorted and I can only apologise. 

You are in luck though, I have managed to get myself into exactly the same state on my own system so I will have a play in the morning and try and advise. 

What I believe is the case is my commands when run are using a different version of pm2, as I'm technically running them as a nodejs script. 

What I think I should be advising is using the plane pm2 scripts. 

https://pm2.keymetrics.io/docs/usage/startup/

If my theory is correct then I would check what's listed with `pm2 list` and stop that if anything is running. 

Then start OctoFarm with `npm install` 

Then follow the instructions from pm2. 

I won't get a chance to try till morning so thought I'd message if you have more time. 

Hope it helps.

brooksware2000, 2020-12-31:
No apologies are necessary.  The community is benefiting from your efforts.

I'm going to mess around a bit more and see if I can figure anything out.  Have a good night.

Curtis

NotExpectedYet, 2021-01-01:
Right so mine is now working on reboot. 

For mongodb I did this: 
`sudo systemctl enable mongod.service`
That will enable the service on start up. I'll add it to my instructions. 

For OctoFarm I did the following: 

`pm2 list` - Nothing should be in here, you should have an id table with nothing under it. 
If something is there, possibly your "app.js" will be under name. Delete this if it exists. 
`pm2 stop {service name}`
`pm2 delete {service name}`

cd into the OctoFarm folder. 

`npm start`

After this has run (will install dependencies) then should start the OctoFarm service. You should see an OctoFarm named service when running `pm2 list` 

`pm2 startup` 

We already installed that globally so shouldn't need to now. This should give you a url to past in like mine below:
`sudo env PATH=$PATH:/usr/bin /usr/lib/node_modules/pm2/bin/pm2 startup systemd -u octoprint --hp /home/octoprint` 

`pm2 save` - This will save that setup. 

`sudo reboot` - Reboot to confirm. 

I've confirmed this with my own system and the previous messed up situation with the install so hopefully that should get you up and running.

brooksware2000, 2021-01-01:
Success!  Well, I did need to perform one additional step:

I modified this command:

npm install pm2 -g

to this command:

sudo npm install pm2 @latest -g

I needed to do this as the version of pm2 that was installed with the first command did not match what was in memory - 
"In memory PM2 version: 4.5.1
Local PM2 version: 4.5.0"

After running npm install pm2 @latest -g, everything worked after a server reboot.

Your help has been much appreciated.

V/r

Curtis

NotExpectedYet, 2021-01-01:
Epic! Good to hear.

closed 2021-01-01 by NotExpectedYet

C0lemanjr, 2025-02-19:
Hi All,
Following this thread I also encountered some trouble. Seems my pm list is showing OctoFarm running (see image)

[image]

, but I am unable to access it from an ipaddress (image attached). 

[image]

Two errors I was able to identify were/are:
1) sudo pm2 start app.js\ script error
         [PM2][ERROR] Script not found: /home/alton/OctoFarm/app.js

2)sudo systemctl status mongod showing an error running
          × mongod.service - MongoDB Database Server
     Loaded: loaded (/lib/systemd/system/mongod.service; disabled; vendor preset: enabled)
     Active: failed (Result: core-dump) since Wed 2025-02-19 17:52:47 UTC; 7s ago
       Docs: https://docs.mongodb.org/manual
    Process: 42099 ExecStart=/usr/bin/mongod --config /etc/mongod.conf (code=dumped, signal=ILL)
   Main PID: 42099 (code=dumped, signal=ILL)
        CPU: 52ms

Feb 19 17:52:47 ubuntuserver systemd[1]: Started MongoDB Database Server.
Feb 19 17:52:47 ubuntuserver systemd[1]: mongod.service: Main process exited, code=dumped, status=4/ILL
Feb 19 17:52:47 ubuntuserver systemd[1]: mongod.service: Failed with result 'core-dump'.

Any suggestions?