OctoFarm/OctoFarm #197: Continual loss/retry of printer connection

botbuilder1, 2020-12-31:
I just stood up a new OF 1.1.6 installation.  

OF is constantly dropping and reconnecting to two of my six printers, even after I resync.  

[image]

I am getting the messages below in the log.  Any advice on solving this would be appreciated!

Thu, 31 Dec 2020 13:10:11 GMT | ERROR | OctoFarm-State.log | WebSocket hard failure: 3: ws://192.168.1.18//sockjs/websocket | 
Thu, 31 Dec 2020 13:10:11 GMT | INFO | OctoFarm-State.log | WebSocketClient: retry in 5000ms | data:"Error: Unexpected server response: 4043: ws://192.168.1.18//sockjs/websocket" | 
Thu, 31 Dec 2020 13:10:11 GMT | INFO | OctoFarm-State.log | Re-Opening Websocket: 4: ws://192.168.1.15//sockjs/websocket | 
Thu, 31 Dec 2020 13:10:11 GMT | ERROR | OctoFarm-State.log | Error: Unexpected server response: 404 | data:"4: ws://192.168.1.15//sockjs/websocket" | 
Thu, 31 Dec 2020 13:10:11 GMT | ERROR | OctoFarm-State.log | WebSocket hard failure: 4: ws://192.168.1.15//sockjs/websocket | 
Thu, 31 Dec 2020 13:10:11 GMT | INFO | OctoFarm-State.log | WebSocketClient: retry in 5000ms | data:"Error: Unexpected server response: 4044: ws://192.168.1.15//sockjs/websocket" | 
Thu, 31 Dec 2020 13:10:13 GMT | INFO | OctoFarm-State.log | Re-Opening Websocket: 4: ws://192.168.1.15//sockjs/websocket | 
Thu, 31 Dec 2020 13:10:13 GMT | ERROR | OctoFarm-State.log | Error: Unexpected server response: 404 | data:"4: ws://192.168.1.15//sockjs/websocket" | 
Thu, 31 Dec 2020 13:10:13 GMT | ERROR | OctoFarm-State.log | WebSocket hard failure: 4: ws://192.168.1.15//sockjs/websocket | 
Thu, 31 Dec 2020 13:10:13 GMT | INFO | OctoFarm-State.log | WebSocketClient: retry in 5000ms | data:"Error: Unexpected server response: 4044: ws://192.168.1.15//sockjs/websocket" | 
Thu, 31 Dec 2020 13:10:15 GMT | INFO | OctoFarm-State.log | Re-Opening Websocket: 3: ws://192.168.1.18//sockjs/websocket | 
Thu, 31 Dec 2020 13:10:15 GMT | INFO | OctoFarm-State.log | Re-Opening Websocket: 3: ws://192.168.1.18//sockjs/websocket | 
Thu, 31 Dec 2020 13:10:16 GMT | INFO | OctoFarm-State.log | Re-Opening Websocket: 3: ws://192.168.1.18//sockjs/websocket | 
Thu, 31 Dec 2020 13:10:16 GMT | ERROR | OctoFarm-State.log | Error: Unexpected server response: 404 | data:"3: ws://192.168.1.18//sockjs/websocket" | 
Thu, 31 Dec 2020 13:10:16 GMT | ERROR | OctoFarm-State.log | WebSocket hard failure: 3: ws://192.168.1.18//sockjs/websocket | 
Thu, 31 Dec 2020 13:10:16 GMT | INFO | OctoFarm-State.log | WebSocketClient: retry in 5000ms | data:"Error: Unexpected server response: 4043: ws://192.168.1.18//sockjs/websocket" | 
Thu, 31 Dec 2020 13:10:16 GMT | ERROR | OctoFarm-State.log | Error: Unexpected server response: 404 | data:"3: ws://192.168.1.18//sockjs/websocket" | 
Thu, 31 Dec 2020 13:10:16 GMT | ERROR | OctoFarm-State.log | WebSocket hard failure: 3: ws://192.168.1.18//sockjs/websocket | 
Thu, 31 Dec 2020 13:10:16 GMT | INFO | OctoFarm-State.log | WebSocketClient: retry in 5000ms | data:"Error: Unexpected server response: 4043: ws://192.168.1.18//sockjs/websocket" | 
Thu, 31 Dec 2020 13:10:16 GMT | ERROR | OctoFarm-State.log | Error: Unexpected server response: 404 | data:"3: ws://192.168.1.18//sockjs/websocket" | 
Thu, 31 Dec 2020 13:10:16 GMT | ERROR | OctoFarm-State.log | WebSocket hard failure: 3: ws://192.168.1.18//sockjs/websocket | 
Thu, 31 Dec 2020 13:10:16 GMT | INFO | OctoFarm-State.log | WebSocketClient: retry in 5000ms | data:"Error: Unexpected server response: 4043: ws://192.168.1.18//sockjs/websocket" |

NotExpectedYet, 2021-01-01:
You have a forward slash in your url. If you've removed it, I think there's a bug and you'll need to restart the service.

NotExpectedYet, 2021-01-06:
This ha just been pushed to master. Please re-open if required.

closed 2021-01-06 by NotExpectedYet