OctoFarm/OctoFarm #199: OctoFarm - Lost Connection to Server...

AndyKoeppe, 2021-01-03:
Hallo ich bin neu in der OctoFarm/OctoPrint (3d Welt).

Ich habe ein Problem mit Octofarm. Habe es schon mehrfach versucht zu installieren auf unterschiedliche Systeme. Aber unter allen Systemen bekommt Octofarm keine Verbindung zur Datenbank. 

Mich würde auch interessieren wie ich das Filament auf mehreren OctoPrint Instanzen synchronisieren kann.

Beste grüße

Andreas

___________________________

Hello, I am new to OctoFarm / OctoPrint (3d world).

I have a problem with Octofarm. Have tried several times to install on different systems. But under all systems Octofarm has no connection to the database.

I would also be interested in how I can synchronize the filament on multiple OctoPrint instances.

Best regards

Andreas

AndyKoeppe, 2021-01-03:
Ich habe OctoFarm jetzt am laufen. Habe die die Zip Daten von GitHub auf mein System geschoben und dann war OctoFarm erreichbar.

Lediglich fehlt mir noch eine Verbindung vom Octoprint zu Octoprint um das Filament zu synchronisieren.

______________________________________________________

I have OctoFarm up and running now. I pushed the zip data from GitHub onto my system and OctoFarm was available.

The only thing missing is a connection from Octoprint to Octoprint to synchronize the filament.

NotExpectedYet, 2021-01-03:
I'll need some more information here regarding the database connection issue. What system? and how did you install specifically? 

Usually it's the mongodb service not starting correctly if you followed the instructions on the website I've updated them recently as there was some issues with them initially. 

What do you mean by syncronise filament? The main Filament manager in OctoFarm has no baring on OctoPrint and nothing get's sent to it. You'd need to setup Filament Manager plugin which OctoFarm supports there's a fair few tutorials on Google for how to correctly set that up with a database.

AndyKoeppe, 2021-01-03:
Is installed
Debian GNU / Linux 10 (Buster)
Linux iobroker-IP 4.19.0-13-amd64 # 1 SMP Debian 4.19.160-2 (2020-11-28) x86_64

I worked according to the instructions and got everything installed for a while. mongodb npm pm2 run.
node --version v13.14.0

sudo systemctl status mongod
● mongod.service - MongoDB Database Server
   Loaded: loaded (/lib/systemd/system/mongod.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2021-01-03 11:10:44 CET; 11min ago
     Docs: https://docs.mongodb.org/manual
 Main PID: 914 (mongod)
   Memory: 216.3M
   CGroup: /system.slice/mongod.service
           └─914 / usr / bin / mongod --config /etc/mongod.conf

Jan 03 11:10:44 iobroker-IP systemd [1]: Started MongoDB Database Server.
Jan 03 11:21:06 iobroker-IP systemd [1]: /lib/systemd/system/mongod.service:11: PIDFile = references path below legacy dir

When I pushed the zip folder onto my system from GitHub, there was a brief connection. Account set up my printer and then the connection was gone again.

Filament.
I use the Filament Manager on OctoPrint, the filament has also been added to OctoFarm, but it is not returned to OctoPrint (printer 2 without filament specified).

AndyKoeppe, 2021-01-03:
`[PM2] Flushing /home/andreas/.pm2/pm2.log
[PM2] Flushing:
[PM2] /home/andreas/.pm2/logs/OctoFarm-out.log
[PM2] /home/andreas/OctoFarm/logs/pm2.error.log
[PM2] Flushing:
[PM2] /home/andreas/.pm2/logs/app-out.log
[PM2] /home/andreas/.pm2/logs/app-error.log
[PM2] Logs flushed
[PM2][ERROR] Script already launched, add -f option to force re-execution
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! octofarm@1.1.6 start: `pm2 flush && pm2 start app.js --name OctoFarm -e './logs/pm2.error.log' --time`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the octofarm@1.1.6 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/andreas/.npm/_logs/2021-01-03T10_24_38_398Z-debug.log`

AndyKoeppe, 2021-01-03:
[PM2] Flushing /home/andreas/.pm2/pm2.log
[PM2] Flushing:
[PM2] /home/andreas/.pm2/logs/OctoFarm-out.log
[PM2] /home/andreas/OctoFarm/logs/pm2.error.log
[PM2] Flushing:
[PM2] /home/andreas/.pm2/logs/app-out.log
[PM2] /home/andreas/.pm2/logs/app-error.log
[PM2] Logs flushed
[PM2][ERROR] Script already launched, add -f option to force re-execution
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! octofarm@1.1.6 start: `pm2 flush && pm2 start app.js --name OctoFarm -e './logs/pm2.error.log' --time`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the octofarm@1.1.6 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/andreas/.npm/_logs/2021-01-03T10_24_38_398Z-debug.log

AndyKoeppe, 2021-01-03:
~/OctoFarm$ pm2 flush && pm2 start app.js --name OctoFarm -e
[PM2] Flushing /home/andreas/.pm2/pm2.log
[PM2] Flushing:
[PM2] /home/andreas/.pm2/logs/OctoFarm-out.log
[PM2] /home/andreas/OctoFarm/logs/pm2.error.log
[PM2] Flushing:
[PM2] /home/andreas/.pm2/logs/app-out.log
[PM2] /home/andreas/.pm2/logs/app-error.log
[PM2] Logs flushed

  error: option `-e --error <path>' argument missing

NotExpectedYet, 2021-01-03:
Well for one your using nodejs 13 which isn't recommended nor specified in the instructions. I recommend down grading that because I don't develop against anything higher than 12 currently. 

/home/andreas/OctoFarm/logs/pm2.error.log

Can you send me the contents of that file please? 

It's getting confusing referencing the filament manager here as well, can you open another ticket for that please?

AndyKoeppe, 2021-01-03:
I followed the instructions exactly, version 13 is always installed. I have now installed version 12.18.3. There is nothing in the log.

NotExpectedYet, 2021-01-04:
What does `pm2 list` show?

AndyKoeppe, 2021-01-04:
pm2 list
┌────┬────────────────────┬──────────┬──────┬───────────┬──────────┬──────────┐
│ id │ name               │ mode     │ ↺    │ status    │ cpu      │ memory   │

AndyKoeppe, 2021-01-04:
npm start
npm ERR! code ENOENT
npm ERR! syscall open
npm ERR! path /home/andreas/package.json
npm ERR! errno -2
npm ERR! enoent ENOENT: no such file or directory, open '/home/andreas/package.json'
npm ERR! enoent This is related to npm not being able to find a file.
npm ERR! enoent

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/andreas/.npm/_logs/2021-01-04T14_39_42_145Z-debug.log

NotExpectedYet, 2021-01-04:
Your not running npm start from the OctoFarm folder?

AndyKoeppe, 2021-01-04:
Yes

NotExpectedYet, 2021-01-04:
Well it's looking for package.json in your home directory so I'm not sure what you've done

[image]

AndyKoeppe, 2021-01-04:
[image]

AndyKoeppe, 2021-01-04:
[image]

AndyKoeppe, 2021-01-04:
[image]

NotExpectedYet, 2021-01-04:
Need some logs please, there should be something in pm2.error.log now the service is starting.

AndyKoeppe, 2021-01-04:
there is still nothing in the log

NotExpectedYet, 2021-01-04:
What about OctoFarm-Server.log?

AndyKoeppe, 2021-01-04:
on, 04 Jan 2021 17:08:43 GMT | ERROR | OctoFarm-Server.log | MongooseServerSelectionError: connect ECONNREFUSED 192.168.178.53:27017 | 
Mon, 04 Jan 2021 17:08:43 GMT | INFO | OctoFarm-Server.log | HTTP server started... | 
Mon, 04 Jan 2021 17:08:43 GMT | INFO | OctoFarm-Server.log | You can now access your server on port: 4000 |

NotExpectedYet, 2021-01-04:
Your mongo server is refusing the connection man. It's either going to be the URL or the mongod server configuration that's at fault.

AndyKoeppe, 2021-01-04:
However, I've already reinstalled everything several times and don't know how to change the data.

NotExpectedYet, 2021-01-04:
Is the mongo server running?

AndyKoeppe, 2021-01-04:
Yes Mongo-Server runs on the same system but in the main folder not in the OctoFarm folder.

AndyKoeppe, 2021-01-04:
When I enter the command: sudo systemctl start mongod, however, I don't get any response from the console.

NotExpectedYet, 2021-01-04:
Oh no that's fine, just wanted to confirm the default database url will work. 

Honestly I'm not really sure what to suggest at this point. It's something with the mongodb instance that's running. 

If `sudo systemctl status mongod` shows active, then we'd have to take a look at mongo's access logs to see what it's saying. 

Those should be stored in `/var/log/mongodb`, they are on Ubuntu anyway not sure about Debian. 

Apart from this I'm not really sure what to suggest.

NotExpectedYet, 2021-01-04:
> 
> 
> When I enter the command: sudo systemctl start mongod, however, I don't get any response from the console.

That maybe the issue. I don't think the server is installed correctly. Were there any errors when running the installation commands?

NotExpectedYet, 2021-01-04:
I know this is cliche but have you tried rebooting?

NotExpectedYet, 2021-01-04:
`Jan 03 11:21:06 iobroker-IP systemd [1]: /lib/systemd/system/mongod.service:11: PIDFile = references path below legacy dir` 

Think I've got an idea. How much disk space is on the system? That's actually an error code from the mongo service saying there isn't enough space to setup the collections.

NotExpectedYet, 2021-01-04:
Well it might not specifically be disk space but your mongodb is moaning about something aha. 

If you send through the /var/logs/mongod log file from that folder should be able to advise.

AndyKoeppe, 2021-01-04:
I have no access to the mongodb logs. when I enter mongo in the console the following comes up:

MongoDB shell version v4.4.2
connecting to: mongodb: //127.0.0.1: 27017 /? compressors = disabled & gssapiServiceName = mongodb
Error: couldn't connect to server 127.0.0.1:27017, connection attempt failed: SocketException: Error connecting to 127.0.0.1:27017 :: caused by :: Connection refused:
connect@src/mongo/shell/mongo.js: 374: 17
@ (connect): 2: 6
exception: connect failed
exiting with code 1

There is still around 100GB of free space on the system and 6GB of Ram. The system was restarted several times.

NotExpectedYet, 2021-01-04:
netstat -an | grep 27017 

What does that output? 

Confusing because this is all suggesting the service isn't running. There's nothing blocking ports like a firewall or anything like that?

AndyKoeppe, 2021-01-04:
-bash: netstat: Kommando nicht gefunden.

NotExpectedYet, 2021-01-04:
Could you also try (sudo if required) 

service mongod stop

systemctl restart mongod

NotExpectedYet, 2021-01-04:
> -bash: netstat: Kommando nicht gefunden.

You may need to install it, not sure how on Debian.

NotExpectedYet, 2021-01-28:
I'm closing this as there has been no response for a while.

closed 2021-01-28 by NotExpectedYet