OctoFarm/OctoFarm #221: [FEATURE] Messaged from Patreon regarding adding some extra features. These we're quick and could utilise systems already implemented so think they are perfect for a little release treat with the next 1.1.7 bug fixes.

NotExpectedYet, 2021-01-06:
Messaged from Patreon regarding adding some extra features. These we're quick and could utilise systems already implemented so think they are perfect for a little release treat with the next 1.1.7 bug fixes. 

All thanks for Scott for these.

_Originally posted by @NotExpectedYet in https://github.com/NotExpectedYet/OctoFarm/discussions/218_

scottypres, 2021-01-06:
Being able to send gcode macros to selected printers would be awesome. That could presumably take the place of start, pause, preheat etc.

NotExpectedYet, 2021-01-06:
Thanks for the response. 

I've done Pre-Heat as a text box for tool, chamber and bed for now due to me not having the Pre-Heat presets setup yet. 

I've done Control as a pop up, will have start, pause, cancel, restart, resume and also move/home commands. These are all buttons to press. 

I've also added a gcode option, this will work same as terminal really and allow for multiple commands to be queued and sent at once. 

I think that should cover most basis for what you wanted without having to use gcode for everything let me know if you've any more suggestions for it.

commit aa62ea7c (2021-01-06): Fixed #221 - New bulk printer actions for controlling.
closed 2021-01-06 by NotExpectedYet