OctoFarm/OctoFarm #222: Lost connection to server (using Docker)

marmao78, 2021-01-13:
Dear OctoFarm team,
How are you?
I try to install OctoFarm in Docker. I succeed in installing Mongo DB and Octofarm but when I try to open the application on the browser it shows trying to connect to DB (lost connection to Server).
[image]
and keep trying...
Please find logs below:
******************** Logs ***********************
2021-01-13T22:27:56.790747162Z 
2021-01-13T22:27:56.790773266Z > core-js@2.6.11 postinstall /app/node_modules/core-js
2021-01-13T22:27:56.790776536Z > node -e "try{require('./postinstall')}catch(e){}"
2021-01-13T22:27:56.790779122Z 
2021-01-13T22:27:56.832267152Z Thank you for using core-js ( https://github.com/zloirock/core-js ) for polyfilling JavaScript standard library!
2021-01-13T22:27:56.832287154Z 
2021-01-13T22:27:56.832290240Z The project needs your help! Please consider supporting of core-js on Open Collective or Patreon: 
2021-01-13T22:27:56.832293600Z > https://opencollective.com/core-js 
2021-01-13T22:27:56.832296259Z > https://www.patreon.com/zloirock 
2021-01-13T22:27:56.832298734Z 
2021-01-13T22:27:56.832301021Z Also, the author of core-js ( https://github.com/zloirock ) is looking for a good job -)
2021-01-13T22:27:56.832303556Z 
2021-01-13T22:27:57.401016929Z 
2021-01-13T22:27:57.401039659Z > gifsicle@5.1.0 postinstall /app/node_modules/gifsicle
2021-01-13T22:27:57.401042984Z > node lib/install.js
2021-01-13T22:27:57.401045396Z 
2021-01-13T22:27:57.801689680Z   ⚠ Response code 404 (Not Found)
2021-01-13T22:27:57.802144926Z   ⚠ gifsicle pre-build test failed
2021-01-13T22:27:57.802588374Z   ℹ compiling from source
2021-01-13T22:28:07.055695169Z   ✔ gifsicle built successfully
2021-01-13T22:28:07.062627046Z 
2021-01-13T22:28:07.062646960Z > mozjpeg@6.0.1 postinstall /app/node_modules/mozjpeg
2021-01-13T22:28:07.062650671Z > node lib/install.js
2021-01-13T22:28:07.062653237Z 
2021-01-13T22:28:07.856582187Z   ✔ mozjpeg pre-build test passed successfully
2021-01-13T22:28:07.868823276Z 
2021-01-13T22:28:07.868970690Z > optipng-bin@6.0.0 postinstall /app/node_modules/optipng-bin
2021-01-13T22:28:07.868977459Z > node lib/install.js
2021-01-13T22:28:07.868980198Z 
2021-01-13T22:28:08.198459871Z   ✔ optipng pre-build test passed successfully
2021-01-13T22:28:08.223099382Z 
2021-01-13T22:28:08.223122913Z > nodemon@2.0.4 postinstall /app/node_modules/nodemon
2021-01-13T22:28:08.223126098Z > node bin/postinstall || exit 0
2021-01-13T22:28:08.223128341Z 
2021-01-13T22:28:08.283243398Z Love nodemon? You can now support the project via the open collective:
2021-01-13T22:28:08.283263976Z  > https://opencollective.com/nodemon/donate
2021-01-13T22:28:08.283266856Z 
2021-01-13T22:28:09.832554907Z npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.13 (node_modules/glob-watcher/node_modules/fsevents):
2021-01-13T22:28:09.833126963Z npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.13: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})
2021-01-13T22:28:09.833861959Z npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.13 (node_modules/@babel/cli/node_modules/fsevents):
2021-01-13T22:28:09.834446936Z npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.13: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})
2021-01-13T22:28:09.835103430Z npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@2.1.3 (node_modules/fsevents):
2021-01-13T22:28:09.835698770Z npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.1.3: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})
2021-01-13T22:28:09.836155320Z 
2021-01-13T22:28:09.841623561Z added 2083 packages from 943 contributors and audited 2089 packages in 77.776s
2021-01-13T22:28:09.842427022Z found 18 vulnerabilities (15 low, 2 moderate, 1 high)
2021-01-13T22:28:09.842437007Z   run `npm audit fix` to fix them, or `npm audit` for details
2021-01-13T22:28:09.864254553Z /app
2021-01-13T22:28:10.057455491Z 
2021-01-13T22:28:10.057490172Z                         -------------
2021-01-13T22:28:10.057493543Z 
2021-01-13T22:28:10.057495873Z __/\\\\\\\\\\\\\____/\\\\____________/\\\\____/\\\\\\\\\_____
2021-01-13T22:28:10.057498583Z  _\/\\\/////////\\\_\/\\\\\\________/\\\\\\__/\\\///////\\\___
2021-01-13T22:28:10.057501122Z   _\/\\\_______\/\\\_\/\\\//\\\____/\\\//\\\_\///______\//\\\__
2021-01-13T22:28:10.057503709Z    _\/\\\\\\\\\\\\\/__\/\\\\///\\\/\\\/_\/\\\___________/\\\/___
2021-01-13T22:28:10.057506364Z     _\/\\\/////////____\/\\\__\///\\\/___\/\\\________/\\\//_____
2021-01-13T22:28:10.057508807Z      _\/\\\_____________\/\\\____\///_____\/\\\_____/\\\//________
2021-01-13T22:28:10.057511290Z       _\/\\\_____________\/\\\_____________\/\\\___/\\\/___________
2021-01-13T22:28:10.057513799Z        _\/\\\_____________\/\\\_____________\/\\\__/\\\\\\\\\\\\\\\_
2021-01-13T22:28:10.057527087Z         _\///______________\///______________\///__\///////////////__
2021-01-13T22:28:10.057529656Z 
2021-01-13T22:28:10.057531645Z 
2021-01-13T22:28:10.057533610Z                           Runtime Edition
2021-01-13T22:28:10.057535776Z 
2021-01-13T22:28:10.057537783Z         PM2 is a Production Process Manager for Node.js applications
2021-01-13T22:28:10.057539893Z                      with a built-in Load Balancer.
2021-01-13T22:28:10.057541975Z 
2021-01-13T22:28:10.057543923Z                 Start and Daemonize any application:
2021-01-13T22:28:10.057546012Z                 $ pm2 start app.js
2021-01-13T22:28:10.057548075Z 
2021-01-13T22:28:10.057550058Z                 Load Balance 4 instances of api.js:
2021-01-13T22:28:10.057552118Z                 $ pm2 start api.js -i 4
2021-01-13T22:28:10.057554202Z 
2021-01-13T22:28:10.057556142Z                 Monitor in production:
2021-01-13T22:28:10.057558200Z                 $ pm2 monitor
2021-01-13T22:28:10.057560311Z 
2021-01-13T22:28:10.057562246Z                 Make pm2 auto-boot at server restart:
2021-01-13T22:28:10.057564343Z                 $ pm2 startup
2021-01-13T22:28:10.057583624Z 
2021-01-13T22:28:10.057588366Z                 To go further checkout:
2021-01-13T22:28:10.057590453Z                 http://pm2.io/
2021-01-13T22:28:10.057592525Z 
2021-01-13T22:28:10.057594498Z 
2021-01-13T22:28:10.057596858Z                         -------------
2021-01-13T22:28:10.057598924Z 
2021-01-13T22:28:10.060681999Z pm2 launched in no-daemon mode (you can add DEBUG="*" env variable to get more messages)
2021-01-13T22:28:10.274453340Z 2021-01-13T22:28:10: PM2 log: Launching in no daemon mode
2021-01-13T22:28:10.296244874Z 2021-01-13T22:28:10: PM2 log: [PM2] Starting /app/app.js in fork_mode (1 instance)
2021-01-13T22:28:10.298380532Z 2021-01-13T22:28:10: PM2 log: App [OctoFarm:0] starting in -fork mode-
2021-01-13T22:28:10.305354052Z 2021-01-13T22:28:10: PM2 log: App [OctoFarm:0] online
2021-01-13T22:28:10.316199260Z 2021-01-13T22:28:10: PM2 log: [PM2] Done.
2021-01-13T22:28:10.342135512Z 2021-01-13T22:28:10: PM2 log: ┌─────┬─────────────┬─────────────┬─────────┬─────────┬──────────┬────────┬──────┬───────────┬──────────┬──────────┬──────────┬──────────┐
2021-01-13T22:28:10.342163436Z │ id  │ name        │ namespace   │ version │ mode    │ pid      │ uptime │ ↺    │ status    │ cpu      │ mem      │ user     │ watching │
2021-01-13T22:28:10.342178265Z ├─────┼─────────────┼─────────────┼─────────┼─────────┼──────────┼────────┼──────┼───────────┼──────────┼──────────┼──────────┼──────────┤
2021-01-13T22:28:10.342181842Z │ 0   │ OctoFarm    │ default     │ 1.1.7   │ fork    │ 1551     │ 0s     │ 0    │ online    │ 0%       │ 28.4mb   │ root     │ disabled │
2021-01-13T22:28:10.342184730Z └─────┴─────────────┴─────────────┴─────────┴─────────┴──────────┴────────┴──────┴───────────┴──────────┴──────────┴──────────┴──────────┘

2021-01-13T22:28:10.342747422Z 2021-01-13T22:28:10: PM2 log: [--no-daemon] Continue to stream logs
2021-01-13T22:28:10.343305016Z 2021-01-13T22:28:10: PM2 log: [--no-daemon] Exit on target PM2 exit pid=1540
2021-01-13T22:28:11.021863473Z 22:28:11 0|OctoFarm  | 2021-01-13T22:28:11: mongodb://172.17.0.2:27017/octofarm
2021-01-13T22:28:41.045474984Z 22:28:41 0|OctoFarm  | 2021-01-13T22:28:41: You can now access your server on port: 4000

I would like your help if you can guide me on what steps to fix it.
Thank you for the great tool.

NotExpectedYet, 2021-01-14:
Hello, I am well thanks hope you are too. 

Could you fill out the issue report please? That's going to save a lot of back and forth. Also could you add OctoFarm-Server.logs please? 

If your new to git, create a new issue please and I'll clean this one up or you can close it thanks.

marmao78, 2021-01-14:
Sorry for the dummy question. I didn't find the issue report form? Please share the link, I will be happy to fill it. 
Please find my Octofarm logs.
[pm2.log](https://github.com/NotExpectedYet/OctoFarm/files/5816109/pm2.log)
[OctoFarm-Server.log](https://github.com/NotExpectedYet/OctoFarm/files/5816110/OctoFarm-Server.log)
[pm2.error.log](https://github.com/NotExpectedYet/OctoFarm/files/5816111/pm2.error.log)
If there is any information that I can share please let me know.
Thank you for your help.

NotExpectedYet, 2021-01-18:
It's when you open a new ticket, there's some options to choose from. It's fine, you have provided enough info.

NotExpectedYet, 2021-01-18:
It seems your mongoDB service has authentication turned on and you haven't provided any in the connection string. Do you know if this is the case? 

Could you detail abit more about how you installed mongodb?

NotExpectedYet, 2021-01-18:
How is your system setup too? You mentioned installing in docker. I gather OctoFarm is running on Docker and mongodb is installed where? 

I have a monolithic image pre-setup with mongodb and OctoFarm if you were not aware?

marmao78, 2021-01-19:
Hi James, Thank you for your feedback.
I found a docker command at Discord using monolithic and it works fine.
**************************
docker run -d --name octofarm -e "MONGO=mongodb://127.0.0.1/octofarm" -p 4000:4000 octofarm/octofarm:monolithic-latest -e "TZ=America/Chicago" -v '{path}/OctoFarm/logs/':'/app/logs/':'rw' -v '{path}/OctoFarm/scripts/':'/scripts/':'rw' -v '{path}/OctoFarm/data/':'/data/':'rw'
****************************
Thank you for your help.

NotExpectedYet, 2021-01-19:
No worries at all glad your up and running

closed 2021-01-23 by NotExpectedYet