OctoFarm/OctoFarm #223: [BUG] Shutting down printers will show them online but disconnected

Chrismettal, 2021-01-15:
### To better help the developer understand your issue, please fill in the information below. Failing to do so will waste the developer and your own time in sorting the issue. Thanks!

**Describe the bug**
-When a connected printer gets turned off, it will still be shown online but failed to connect, so it will count towards "online percentages" and such. To have it actually shown disconnected I need to resync the printers after I shut them down.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect and sync printer
2. Have a print finish
3. Turn off printer 
4. Check OctoFarm after a week and see that the printer apparently was "completed" all the time.

**Expected behavior**
-I expected to after a certain amount of attempts have the printer be shown offline without the need of manually resyncing so my stats actually reflect the printer state.

**Screenshots**
[image]
Right now I only have a screenshot of the faulty stats. If needed I can create more screenshots of a turned off printer that's still connected tomorrow.

[image]

**Versions (please complete the following information):**
- Docker container monolithic-latest, pulled 2020-12-23 (Is there a container image version number to find if I used latest?)

NotExpectedYet, 2021-01-18:
Hello, 

How are you shutting down your printer? I gather you mean OctoPrint/OctoPi itself not the printer. Just wanted to clarify. 

What should happen is those should be counted as "Offline" as you state because the "Complete" flag should be getting wiped. I'll have a look and see if I can re-produce. Will be something in the latest changes more than likely. 

Regarding the docker version, I just require what's on the system tab :) 
[image]
Containers on the latest tag just update to the latest master commit.

NotExpectedYet, 2021-01-22:
Could you upload some state logs please? I'm not sure how this is occurring.

NotExpectedYet, 2021-01-22:
Think I've found it, will push a fix shortly.

Chrismettal, 2021-01-22:
Sorry, only now I am seeing your reply. I completely unpower Octoprint and the printer, basically pulling the plug.

Can I assist you with any more logs or did you find the cause for this?

NotExpectedYet, 2021-01-22:
I've just put a fix on Dev if you'd like to try it.

Chrismettal, 2021-01-23:
Thank you!
I have just switched my container to monolithic-dev and see Version 1.1.8 in the System Information tab now. Will run some prints now and get back to you!

NotExpectedYet, 2021-01-23:
Epic, no worries. Hopefully should sort it but let me know and I'll have another look.

Chrismettal, 2021-02-09:
Sorry for taking so long. While it seems that disconnecting when powering off the Pis is more stable now, I think now that the actual problem is me not pressing "Print harvested?" after a print, since when a print finishes right before I go to bed I will usually not open up Octofarm again and just cut the printers power directly.

Doing this, finishing a print, having it say "completed" and then unexpectedly turning off the Pi without "harvesting" seems to get the printer state stuck in "completed" even though it could very well jump to "offline" I would think.

NotExpectedYet, 2021-02-09:
No worries at all man thanks for getting back to me. 

It's all been pushed to master now with 1.1.10 and I'm annoyed that didn't fix it for you. 

So my websockets stay connected with the listener, but due websockets just stopping sending in your case, I have to setup a ping/pong to catch those events. That ping/pong should have captured the printer missing and I'm wondering why it didn't. When that happens it should fire off an error as it didn't get the pong return from it's initial ping. That error should correctly replicate through the rest of the information as the graphs just run on a timer and just grab the latest information. 

Somewhere in that chain... I'm not updating the latest information. 

Could you send me through your state logs where you've got a printer shutdown? Might be able to spot where it's not following through with the data.

Chrismettal, 2021-02-09:
For Information, I have two machines running, IPs ending in 224 and 225. 225 is my main work horse and most often gets disconnected unexpectedly.

The logs sadly don't go far enough back to the last unexpected shutdown of 225 on Saturday. I have a print running right now on machine 225 and will provoke it for you in the evening. What I can see in the logs on sunday where no printer was running is this:

```
Sun, 07 Feb 2021 09:27:45 GMT | ERROR | OctoFarm-State.log | Error: timeout | 
Sun, 07 Feb 2021 09:27:45 GMT | ERROR | OctoFarm-State.log | Error: timeout | 
Sun, 07 Feb 2021 09:27:45 GMT | INFO | OctoFarm-State.log | Attempting to re-connect to API: api/users | http://192.168.2.224 | timeout: 10000 | 
Sun, 07 Feb 2021 09:27:45 GMT | INFO | OctoFarm-State.log | Attempting to connect to API: api/users | http://192.168.2.224 | timeout: 10000 | 
Sun, 07 Feb 2021 09:27:45 GMT | INFO | OctoFarm-State.log | Attempting to re-connect to API: api/users | http://192.168.2.224 | timeout: 10000 | 
Sun, 07 Feb 2021 09:27:45 GMT | INFO | OctoFarm-State.log | Attempting to connect to API: api/users | http://192.168.2.224 | timeout: 10000 | 
Sun, 07 Feb 2021 09:27:47 GMT | ERROR | OctoFarm-State.log | FetchError: request to http://192.168.2.224/api/users failed, reason: connect EHOSTUNREACH 192.168.2.224:80 | 
Sun, 07 Feb 2021 09:27:47 GMT | ERROR | OctoFarm-State.log | FetchError: request to http://192.168.2.224/api/users failed, reason: connect EHOSTUNREACH 192.168.2.224:80 | 
Sun, 07 Feb 2021 09:27:47 GMT | INFO | OctoFarm-State.log | Timeout Exceeded: api/users | http://192.168.2.224 | 
Sun, 07 Feb 2021 09:27:47 GMT | ERROR | OctoFarm-State.log | request to http://192.168.2.224/api/users failed, reason: connect EHOSTUNREACH 192.168.2.224:80 | data:"Couldn't grab initial connection for Printer: http://192.168.2.224" | 
Sun, 07 Feb 2021 09:27:47 GMT | INFO | OctoFarm-State.log | Timeout Exceeded: api/users | http://192.168.2.224 | 
Sun, 07 Feb 2021 09:27:47 GMT | ERROR | OctoFarm-State.log | request to http://192.168.2.224/api/users failed, reason: connect EHOSTUNREACH 192.168.2.224:80 | data:"Couldn't grab initial connection for Printer: http://192.168.2.224" | 
Sun, 07 Feb 2021 09:27:51 GMT | INFO | OctoFarm-State.log | Attempting to connect to API: api/users | http://192.168.2.224 | timeout: 1000 | 
Sun, 07 Feb 2021 09:27:51 GMT | INFO | OctoFarm-State.log | Attempting to connect to API: api/users | http://192.168.2.224 | timeout: 1000 | 
Sun, 07 Feb 2021 09:27:52 GMT | ERROR | OctoFarm-State.log | Error: timeout | 
Sun, 07 Feb 2021 09:27:52 GMT | ERROR | OctoFarm-State.log | Error: timeout | 
Sun, 07 Feb 2021 09:27:52 GMT | INFO | OctoFarm-State.log | Attempting to re-connect to API: api/users | http://192.168.2.224 | timeout: 10000 | 
Sun, 07 Feb 2021 09:27:52 GMT | INFO | OctoFarm-State.log | Attempting to connect to API: api/users | http://192.168.2.224 | timeout: 10000 | 
Sun, 07 Feb 2021 09:27:52 GMT | INFO | OctoFarm-State.log | Attempting to re-connect to API: api/users | http://192.168.2.224 | timeout: 10000 | 
Sun, 07 Feb 2021 09:27:52 GMT | INFO | OctoFarm-State.log | Attempting to connect to API: api/users | http://192.168.2.224 | timeout: 10000 | 
Sun, 07 Feb 2021 09:27:54 GMT | ERROR | OctoFarm-State.log | FetchError: request to http://192.168.2.224/api/users failed, reason: connect EHOSTUNREACH 192.168.2.224:80 | 
Sun, 07 Feb 2021 09:27:54 GMT | ERROR | OctoFarm-State.log | FetchError: request to http://192.168.2.224/api/users failed, reason: connect EHOSTUNREACH 192.168.2.224:80 | 
Sun, 07 Feb 2021 09:27:54 GMT | INFO | OctoFarm-State.log | Timeout Exceeded: api/users | http://192.168.2.224 | 
Sun, 07 Feb 2021 09:27:54 GMT | ERROR | OctoFarm-State.log | request to http://192.168.2.224/api/users failed, reason: connect EHOSTUNREACH 192.168.2.224:80 | data:"Couldn't grab initial connection for Printer: http://192.168.2.224" | 
Sun, 07 Feb 2021 09:27:54 GMT | INFO | OctoFarm-State.log | Timeout Exceeded: api/users | http://192.168.2.224 | 
Sun, 07 Feb 2021 09:27:54 GMT | ERROR | OctoFarm-State.log | request to http://192.168.2.224/api/users failed, reason: connect EHOSTUNREACH 192.168.2.224:80 | data:"Couldn't grab initial connection for Printer: http://192.168.2.224" | 
Sun, 07 Feb 2021 09:27:57 GMT | INFO | OctoFarm-State.log | Attempting to connect to API: api/users | http://192.168.2.224 | timeout: 1000 | 
Sun, 07 Feb 2021 09:27:57 GMT | INFO | OctoFarm-State.log | Attempting to connect to API: api/users | http://192.168.2.224 | timeout: 1000 | 
```

While every line appears twice, as it should for two disconnected printers, it seems to ping machine 224 twice and never even try to reach machine 225 which was last disconnected?

Chrismettal, 2021-02-09:
```
Tue, 09 Feb 2021 13:01:36 GMT | INFO | OctoFarm-State.log | PrintDone1: ws://192.168.2.225/sockjs/websocket | 
Tue, 09 Feb 2021 13:03:27 GMT | INFO | OctoFarm-State.log | PrintDone0: ws://192.168.2.224/sockjs/websocket | 
Tue, 09 Feb 2021 13:19:43 GMT | INFO | OctoFarm-State.log | Error with websockets... resetting up! | 
Tue, 09 Feb 2021 13:19:43 GMT | INFO | OctoFarm-State.log | Attempting to connect to API: api/users | http://192.168.2.224 | timeout: 10000 | 
Tue, 09 Feb 2021 13:19:46 GMT | ERROR | OctoFarm-State.log | FetchError: request to http://192.168.2.224/api/users failed, reason: connect EHOSTUNREACH 192.168.2.224:80 | 
Tue, 09 Feb 2021 13:19:46 GMT | INFO | OctoFarm-State.log | Timeout Exceeded: api/users | http://192.168.2.224 | 
Tue, 09 Feb 2021 13:19:46 GMT | ERROR | OctoFarm-State.log | request to http://192.168.2.224/api/users failed, reason: connect EHOSTUNREACH 192.168.2.224:80 | data:"Couldn't grab initial connection for Printer: http://192.168.2.224" | 
Tue, 09 Feb 2021 13:20:16 GMT | INFO | OctoFarm-State.log | Attempting to connect to API: api/users | http://192.168.2.224 | timeout: 1000 | 
Tue, 09 Feb 2021 13:20:17 GMT | ERROR | OctoFarm-State.log | Error: timeout | 
Tue, 09 Feb 2021 13:20:17 GMT | INFO | OctoFarm-State.log | Attempting to re-connect to API: api/users | http://192.168.2.224 | timeout: 10000 | 
Tue, 09 Feb 2021 13:20:17 GMT | INFO | OctoFarm-State.log | Attempting to connect to API: api/users | http://192.168.2.224 | timeout: 10000 | 
Tue, 09 Feb 2021 13:20:19 GMT | ERROR | OctoFarm-State.log | FetchError: request to http://192.168.2.224/api/users failed, reason: connect EHOSTUNREACH 192.168.2.224:80 | 
Tue, 09 Feb 2021 13:20:19 GMT | INFO | OctoFarm-State.log | Timeout Exceeded: api/users | http://192.168.2.224 | 
Tue, 09 Feb 2021 13:20:19 GMT | ERROR | OctoFarm-State.log | request to http://192.168.2.224/api/users failed, reason: connect EHOSTUNREACH 192.168.2.224:80 | data:"Couldn't grab initial connection for Printer: http://192.168.2.224" | 
Tue, 09 Feb 2021 13:20:49 GMT | INFO | OctoFarm-State.log | Attempting to connect to API: api/users | http://192.168.2.224 | timeout: 1000 | 
Tue, 09 Feb 2021 13:20:50 GMT | ERROR | OctoFarm-State.log | Error: timeout | 
```

This is the point where both printers finished at the same time and I shut down 224, which popped up as an error, was marked offline and did not get stuck even though I just cut the power.

225 was still running. 
Somewhere around 19:15 GMT 225 was finished, I removed the print and turned off the printer, but nothing shows up in the log for that timeframe at all. In the end I manually did a resync at 19:44, finding both printers offline.

```
Tue, 09 Feb 2021 18:55:01 GMT | INFO | OctoFarm-State.log | Attempting to re-connect to API: api/users | http://192.168.2.224 | timeout: 10000 | 
Tue, 09 Feb 2021 18:55:01 GMT | INFO | OctoFarm-State.log | Attempting to connect to API: api/users | http://192.168.2.224 | timeout: 10000 | 
Tue, 09 Feb 2021 18:55:03 GMT | ERROR | OctoFarm-State.log | FetchError: request to http://192.168.2.224/api/users failed, reason: connect EHOSTUNREACH 192.168.2.224:80 | 
Tue, 09 Feb 2021 18:55:03 GMT | INFO | OctoFarm-State.log | Timeout Exceeded: api/users | http://192.168.2.224 | 
Tue, 09 Feb 2021 18:55:03 GMT | ERROR | OctoFarm-State.log | request to http://192.168.2.224/api/users failed, reason: connect EHOSTUNREACH 192.168.2.224:80 | data:"Couldn't grab initial connection for Printer: http://192.168.2.224" | 
Tue, 09 Feb 2021 18:55:33 GMT | INFO | OctoFarm-State.log | Attempting to connect to API: api/users | http://192.168.2.224 | timeout: 1000 | 
Tue, 09 Feb 2021 18:55:34 GMT | ERROR | OctoFarm-State.log | Error: timeout | 
Tue, 09 Feb 2021 18:55:34 GMT | INFO | OctoFarm-State.log | Attempting to re-connect to API: api/users | http://192.168.2.224 | timeout: 10000 | 
Tue, 09 Feb 2021 18:55:34 GMT | INFO | OctoFarm-State.log | Attempting to connect to API: api/users | http://192.168.2.224 | timeout: 10000 | 
Tue, 09 Feb 2021 18:55:35 GMT | ERROR | OctoFarm-State.log | Could not Establish API Connection: 502http://192.168.2.224 | data:"Couldn't grab initial connection for Printer: http://192.168.2.224" | 
Tue, 09 Feb 2021 19:44:44 GMT | INFO | OctoFarm-State.log | Attempting to connect to API: api/users | http://192.168.2.224 | timeout: 10000 | 
Tue, 09 Feb 2021 19:44:54 GMT | ERROR | OctoFarm-State.log | Error: timeout | 
Tue, 09 Feb 2021 19:44:54 GMT | INFO | OctoFarm-State.log | Timeout Exceeded: api/users | http://192.168.2.224 | 
Tue, 09 Feb 2021 19:44:54 GMT | ERROR | OctoFarm-State.log | timeout | data:"Couldn't grab initial connection for Printer: http://192.168.2.224" | 
Tue, 09 Feb 2021 19:44:54 GMT | INFO | OctoFarm-State.log | Closed websocket connection for: http://192.168.2.225 | 
Tue, 09 Feb 2021 19:44:54 GMT | INFO | OctoFarm-State.log | Attempting to connect to API: api/users | http://192.168.2.225 | timeout: 1000 | 
Tue, 09 Feb 2021 19:44:55 GMT | ERROR | OctoFarm-State.log | Error: timeout | 
Tue, 09 Feb 2021 19:44:55 GMT | INFO | OctoFarm-State.log | Attempting to re-connect to API: api/users | http://192.168.2.225 | timeout: 10000 | 
Tue, 09 Feb 2021 19:44:55 GMT | INFO | OctoFarm-State.log | Attempting to connect to API: api/users | http://192.168.2.225 | timeout: 10000 | 
Tue, 09 Feb 2021 19:45:05 GMT | ERROR | OctoFarm-State.log | Error: timeout | 
Tue, 09 Feb 2021 19:45:05 GMT | INFO | OctoFarm-State.log | Timeout Exceeded: api/users | http://192.168.2.225 | 
Tue, 09 Feb 2021 19:45:05 GMT | ERROR | OctoFarm-State.log | timeout | data:"Couldn't grab initial connection for Printer: http://192.168.2.225" | 
```

As expected, before the resync the printer 225 was stuck operational:

[image]

[image]

The local time shown in the screenshot is GMT+1 so no mention of 225 after I turned it off. The last pong message might have been close to where I turned it off, but it never even tried another ping after that?

NotExpectedYet, 2021-02-11:
It certainly seems like the ping/pong or something isn't firing thanks for those detailed reports. 

I'll scrub the API key from the logs next release for future too I agree it's not needed at all for any diagnostics.

Chrismettal, 2021-02-11:
Thank you!

I think for number 225 after I shut it down it will just never fire a ping again for that machine. Even if I "harvested" it. Since I only have 2 machines connected I can not really say if its a problem with machine 225 or something like "Any machine after the first machine won't be pinged" or something like that. Haven't actually looked at the code at all though sadly.

Slightly related, is there a way to turn off the "harvesting"? I never remember to do that and sometimes start a new print without harvesting, where the printer will say it's "finished" and have no temperature, even though it is running fine:

[image]

[image]

That's machine 224 running about 60% into the print in the shots. 225 is turned on, idle and was harvested.

NotExpectedYet, 2021-02-11:
Save your eyes looking at the code. I was very inexperienced with Javascript and nodejs last year when I wrote that section. I'm itching to get working on 1.2 to resolve all those issues and actually give someone a chance at doing a PR. 

So the harvesting doesn't actually do anything really, all that button does is select the file again to clear the internal state as OctoPrint doesn't have a deselect command available through the API. I suppose it doesn't need it tbh in it's position. The Completed state is only triggered when it detects a 100% or 0 time left so this is seemingly like a specific issue with the ping/pong/web socket connection. 

Which makes me wonder why it's not clearing after you start another print as a new print would reset the conditions which create the Completed state in the first place. 

The temperature is weird as well, because even if it's stuck in the Complete state it should still be updating temperatures. The only thing I can think of that would cause what your seeing, and not be able to replicate it on my own system is the ping/pong again. The only way that could be an issue is if OctoPrint stops sending from a network outage or something similar. I'm not 100% sure on what it would do in that case. When I tested the loosing network connection functional stuff last year by disconnecting my own instances it never re-established connection so I'm probably safe assuming that that's still the case and this is related to the ping/pong. That would be completely blown out of the water though if you've got no network issues, and lead me to believe it's something in the backend not updating the correct state. 

If you hide that Current Operations bar you in effect hide the print harvested but that's not really a solution for someone who wants that Current Operations card there displayed. What "should" happen is that button shouldn't cause any issues at all, pressed or not pressed. This certainly seems like back end from everything you've said anyway, I can tell because you've got the correct "Operational" state designation and a green bar. The Operational State comes from the websocket listener and is then run through a function that generates the colours for that state, somehow that's been missed.

Chrismettal, 2021-02-11:
Thanks for your explanation. So my "problems" with harvesting aren't even the intended way. 😆 Does the listener react differently depending on from where a print is started? I mostly start prints via the Prindoid app, but monitor the printers using Octofarm. Only rarely I start prints via Octofarm since Printoid is a little bit faster from a smartphone.

My connection log today does not even show any pingponging to 224 at all. Both machines were turned on at 6:30, but only 225 shows up in pingponging connection logs for the whole day, even though 224 is running since then.. So the temps not updating might just be since 224 isn't even talked to anymore right now.

While I haven't noticed network failures actively, I agree that Wifi in that room isn't the best. It miiiight timeout sometimes, but I have never seen drops in the camera feeds or couldn't reach a machine.

NotExpectedYet, 2021-02-11:
> Thanks for your explanation. So my "problems" with harvesting aren't even the intended way.  Does the listener react differently depending on from where a print is started? I mostly start prints via [...]

Welcome. Certainly not XD, I gather somethings got out of whack with the recent changes or I've got some very oblivious users as this has been in place for a while. To confirm as well it makes no difference where you start prints, or it "shouldn't" aha. Once that websocket is connected on OctoFarm is should stay there and keep checking. I agree printoid is much faster on a smart phone. The new layout will make that better and if someone hasn't by that time I do want to dabble in Android and iOS again. 

Honestly.. if somethings weird in the OctoFarm neighbourhood, it's something to do with this ginger right here XD.  

Thanks for that extra info, so OctoFarm should now at least recover from that perfectly fine. Automatically as well. What your indicating does indeed sound like the network has dropped and it only takes a millisecond of connection drop for that to occur. 

I remember testing, when I first learnt of this annoyance with websockets, with WIFI and putting my router on a lower power setting, something could walk past it and suddenly my OctoPrint would stop communicating even if it didn't move. I did have it right at the edge of lowered range like but that's how sensitive they "can" be anyway. I don't think that sort of connection would fail over HTTP. I think it's enough for OctoPrint to think OctoFarm disconnected and OctoFarm to still think it's receiving data anyway. 

Well I'm honestly glad to at least have a lead as I was lost before.

commit b5cd6c9f (2021-01-22): Fixed #223 - Shutting down printers whilst still in Complete state will log both Offline and Complete on Utilisation graph
commit 46d08d17 (2021-02-13): Fixed #223 -  Incorrect error message when firing power settings.
closed 2021-02-18 by NotExpectedYet