OctoFarm/OctoFarm #227: Prusa Printer + OctoPrint - API Issues on Sync

MythikWolf, 2021-01-20:
**Describe the bug**
Following the direction on the website, I added a Prusa i3 MK3S+ running OctoPrint 1.5.3 as its head.  Created an app key and added it to the autoscanned printer listing in OctoFarm.   OctoFarm is running on a Pi 4. I get the following wehn trying to sync the Prusa:

22:46:13 | http://10.0.1.166 | Could not Establish API Connection: 403http://10.0.1.166: API issues... halting!

The user associated with the key is an admin on the OctoPrint Server.

**To Reproduce**
Steps to reproduce the behavior:
1. Add Printer to Octo Farm.
2. Generate and Use App Key from OctoPrint.
3. Re-Sync in OctoFarm.

**Expected behavior**
I expected it to sync.

**Screenshots**
[image]

**Versions (please complete the following information):**
 - NodeJS Version : v12.20.1
 - OctoFarm Version : 1.1.7
 - OctoPrint Version : 1.5.3
 - OctoPrint Plugins : Access Anywhere - The Spaghetti Detective, DeleteAfterPrint

**Additional context**
None

NotExpectedYet, 2021-01-20:
403 is forbidden. Your OctoPrint instance is denying the connection. 

Have you enabled CORS and restarted the OctoPrint instance?

MythikWolf, 2021-01-20:
I have in fact. I should have included that. A little sleepy here.

[image]

NotExpectedYet, 2021-01-20:
No drama, ditto as I've not long been up. 

Hmmm strange. It's something with the OctoPrint instance for sure. Do you have any OctoPrint logs from when it tries to connect? 

You could also possibly try the user key although both should authenticate the websocket fine.

NotExpectedYet, 2021-01-20:
Could you also pass me OctoFarms state logs? I'll double check OctoFarm is sending the correct information.

MythikWolf, 2021-01-20:
[OctoFarm-State.log](https://github.com/NotExpectedYet/OctoFarm/files/5840770/OctoFarm-State.log)
[octoprint.log](https://github.com/NotExpectedYet/OctoFarm/files/5840787/octoprint.log)
[serial.log](https://github.com/NotExpectedYet/OctoFarm/files/5840788/serial.log)

Let me know if these don't have what you need.  I grabbed the most recent.

NotExpectedYet, 2021-01-20:
Brilliant thank you, I will take a look shortly for you.

NotExpectedYet, 2021-01-20:
It's your url resolutions. It's not liking the self signed certificate. That's a different issue to your initial one though. I can't see the original IP on there.

Could you explain a little more about your network and DNS setup please?

NotExpectedYet, 2021-01-20:
Ahh spoke too soon it's at the end. Disregard that last one.

NotExpectedYet, 2021-01-20:
Have you tried restarting OctoFarm? 

Just wondering if there is still an issue with edit.

MythikWolf, 2021-01-20:
It is a flat 10.0.1.x/24.  I run my own DNS servers, but I am not going to OctoPrint via https (or at least I shouldn't be).  I do have my TAZ6 and its OctoPrint server connected (Pi 3B v1.2).  

When I originally set things up I was using https and the fully qualified domain.  I saw it was unhappy with https so I deleted everything and let it autoscan for Octoprint servers.

[image]

The above screen is how it is attempting to connect now.  So on http://10.0.1.166.

MythikWolf, 2021-01-20:
Same issue as before after a npm restart.

NotExpectedYet, 2021-01-20:
Yeah OctoFarm updated and corrected itself by the looks when you changed over. 

It's looking like OctoPrint is denying the http connection. There's a lot in the logs I'm unsure about from OctoPrint. You've got some root errors at the start of it, but I don't think they are related. 

How is OctoPrint setup for Https? My usual method is reverse proxy on my home network, but it seems OctoPrint is caching stuff for the client on Https://. Again I'm not eve sure this is related, it's just an unknown at the moment. 

Somethings I would try until I get chance to investigate more. 

1. Restart the OctoFarm service. (Just to clear out any possible issues in the backend). npm restart shouldn't work if you've setup the pm2 service. You'd have to do `pm2 restart OctoFarm`. You can confirm with the uptime in System. 

2. Put Octoprint into safe mode and attempt a resync. (I've not had any reports of those plugins causing issues and they shouldn't but just to confirm).

MythikWolf, 2021-01-20:
[image]

[image]

No dice.

NotExpectedYet, 2021-01-20:
Not sure at minute. It's certainly something with your OctoPrint instance.

MythikWolf, 2021-01-20:
Nothing really odd about it.  It works fine and without error.  I could
reinstall it if you believe that'll help.

NotExpectedYet, 2021-01-20:
That's what I don't understand. 

d, 20 Jan 2021 07:26:46 GMT | INFO | OctoFarm-State.log | Attempting to connect to API: api/users | http://10.0.1.166 | timeout: 10000 | 
Wed, 20 Jan 2021 07:26:46 GMT | ERROR | OctoFarm-State.log | Could not Establish API Connection: 403http://10.0.1.166 | data:"Couldn't grab initial connection for Printer: http://10.0.1.166" | 
Wed, 20 Jan 2021 07:27:16 GMT | INFO | OctoFarm-State.log | Attempting to connect to API: api/users | http://10.0.1.166 | timeout: 10000 | 
Wed, 20 Jan 2021 07:27:16 GMT | ERROR | OctoFarm-State.log | Could not Establish API Connection: 403http://10.0.1.166 | data:"Couldn't grab initial connection for Printer: http://10.0.1.166" | 

It's certainly getting denied as far as OctoFarm is concerned. I can't understand it as that isn't been reflected in the OctoPrint logs at all. 

2021-01-20 07:22:38,941 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:10.0.1.188
2021-01-20 07:22:38,979 - octoprint.server.util.flask - INFO - Passively logging in user glaberge from ::ffff:10.0.1.188
2021-01-20 07:22:38,981 - octoprint.access.users - INFO - Logged in user: glaberge
2021-01-20 07:22:39,161 - octoprint.server.util.sockjs - INFO - User glaberge logged in on the socket from client ::ffff:10.0.1.188

NotExpectedYet, 2021-01-20:
There should be some 403 errors in your OctoPrint logs but it isn't the case.

NotExpectedYet, 2021-01-20:
What is your base_url set to on the proxy? It's reading as https on your logs...

MythikWolf, 2021-01-20:
I have another octoprint server on my network that I havent added because
it is on day 2 of a 5 day print.  I can see if it reacts the same way or if
it connects once the print is done.

In the mean time, I'll reinstall octoprint on the Pi 4 and see if it works
after that. I'll report back.

MythikWolf, 2021-01-20:
No proxy involved.  Unless there is something I am missing on the Octoprint
server.

NotExpectedYet, 2021-01-20:
Funny I always get these issues with Prusa! aha. 

No worries, hopefully that resolves whatever it is. 

It's probably my mis-understanding of this in your logs. 

`2021-01-20 06:47:34,256 - octoprint.server.preemptive_cache - INFO - Preemptively caching / (ui _default) for {'query_string': 'l10n=en', 'path': '/', 'base_url': 'https://mk3s.mythikal.local/'}`

`2021-01-20 07:22:20,730 - octoprint.server.preemptive_cache - INFO - Preemptively caching / (ui _default) for {'query_string': 'l10n=en', 'path': '/', 'base_url': 'https://10.0.1.26/'}` 

Aye it is... I've just checked that on my local instances and that's whatever your accessing it from.

MythikWolf, 2021-01-20:
Reinstalled.  No plugins.  Same error.

Manually added the api/users to the host and got the following from the Prusa:

{
  "users": [
    {
      "active": true, 
      "admin": true, 
      "apikey": null, 
      "name": "glaberge", 
      "settings": {}, 
      "user": true
    }
  ]
}

And this is what the TAZ6 gave me.  Same with the CR10.

{"users":[{"active":true,"admin":true,"apikey":null,"groups":["admins","users"],"name":"glaberge","needs":{"group":["admins","users"],"role":["plugin_pluginmanager_manage","slice","plugin_firmware_check_display","webcam","status","settings_read","timelapse_delete","files_delete","files_list","plugin_softwareupdate_check","timelapse_admin","plugin_logging_manage","plugin_action_command_notification_show","system","plugin_softwareupdate_update","plugin_announcements_read","plugin_backup_access","control","monitor_terminal","plugin_pi_support_check","admin","settings","plugin_action_command_notification_clear","plugin_pluginmanager_install","files_upload","plugin_announcements_manage","plugin_action_command_prompt_interact","print","gcodeviewer","timelapse_download","plugin_softwareupdate_configure","connection","timelapse_list","files_select","plugin_appkeys_admin","files_download"]},"permissions":[],"roles":["user","admin"],"settings":{},"user":true}]}

MythikWolf, 2021-01-21:
And third time is the charm.  Got a new SD card, installed on that and well... this time it worked. No idea why it is working now but it is.  So I am going to take it as a win!

closed 2021-01-21 by MythikWolf

NotExpectedYet, 2021-01-21:
Hmmm that's interesting. I would have never pinned it on the SD card. Glad it's working for you anyway I will keep that in mind of any similar issues arrise.