OctoFarm/OctoFarm #229: [BUG] Connect buttons not working in Printers Tab

GrenadeApe, 2021-01-20:
### To better help the developer understand your issue, please fill in the information below. Failing to do so will waste the developer and your own time in sorting the issue. Thanks!

**Describe the bug**
When using the Connect button either on the individual printer entry or the global button towards the top, there is an error and it doesn't connect. The disconnect function works correctly in both locations.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Printers tab on top menu
2. With the printer disconnected, click on either the global connect button or quick connect button on printer entry
3. See error popup

**Expected behavior**
When you click on the connect button, OctoPrint should connect to the printer.

**Versions (please complete the following information):**
 - NodeJS Version [e.g. 13]  12.20.1
 - OctoFarm Version [e.g. 1.5.5.7]   1.1.7
 - OctoPrint Version [e.g. 1.4.1]   1.5.3
 - OctoPrint Plugins [e.g. bed visualiser, octoklipper, printtimegenius]  Fresh install, none added

**Additional context**
The disconnect function does work, it just won't connect

NotExpectedYet, 2021-01-22:
Was this resolved eventually after your setting started saving? I couldn't reproduce locally.

NotExpectedYet, 2021-01-29:
Fixed on development

commit 19616eb0 (2021-02-02): Fixed #229
closed 2021-01-29 by NotExpectedYet