OctoFarm/OctoFarm #234: [BUG] Adding Power and Custom Power commands won't save

MTrab, 2021-01-23:
### To better help the developer understand your issue, please fill in the information below. Failing to do so will waste the developer and your own time in sorting the issue. Thanks!

**Describe the bug**
When entering information in Power and Custom Power Commands fields, nothing is saved.
After pressing save and reloading the page, the fields are back to default text again.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Printer Settings'
2. Click on 'Power'
3. Add information in the fields
4. Press Save
5. Reload the page
6. Observe the fields are back to default

**Expected behavior**
Settings should be saved

**Screenshots**
Before Save/Reload:
[image]

After Save/Reload:
[image]

**Versions (please complete the following information):**
 - NodeJS Version [e.g. 13] (??)
 - OctoFarm Version [e.g. 1.5.5.7] 1.1.6 (Octofarm-Monolithic)
 - OctoPrint Version [e.g. 1.4.1]  1.5.3
 - OctoPrint Plugins [e.g. bed visualiser, octoklipper, printtimegenius] Enclosure Plugin

**Additional context**
N/A

NotExpectedYet, 2021-01-23:
Hello, could you update to 1.1.7 please and see if that rectifies the issue? 

I'll do some testing here in a minute see if I can reproduce.

NotExpectedYet, 2021-01-23:
Confirmed. Don't worry about updating. Will add it to the 1.1.8 release.

NotExpectedYet, 2021-01-23:
For a work around, save the settings and then press the re-sync button for the printer. It will grab them back from OctoPrint.

MTrab, 2021-01-25:
Updated to 1.1.8, and settings still not being saved. Issue still persists.

NotExpectedYet, 2021-01-25:
Yep just confirmed. What the hell was I smoking yesterday. I'll have another look through week, that's annoying I confirmed it saving yesterday grr.

NotExpectedYet, 2021-01-25:
Right certainly fixed this time. It was the OctoPrint update causing issues... The thing was sending null values back to it, rather than the saved ones then re-syncing itself and clearing. This might explain some other issues I've seen with people not having saved settings. Fixed for it clearing out settings/appearance too. Thanks for reporting, wouldn't have seen that at all otherwise.

NotExpectedYet, 2021-01-25:
It's on the master branch, I'll back fill development shortly.

MTrab, 2021-01-25:
Nice - looking forward test it when pushed into monolithic.

NotExpectedYet, 2021-01-25:
Just finished some tweaks to the new graphs too and pushed to both master and dev. 

Monolithic image on docker hub shouldn't be too long.

NotExpectedYet, 2021-01-25:
I believe it's live :)

MTrab, 2021-01-25:
Hmmm - the image did update, but still reads 1.1.8.

Unfortunately still not saving :(

NotExpectedYet, 2021-01-25:
I haven't formalised it as a version yet so will still read that. Grr, this one is been annoying. It's certainly saving on my end. Just tried it with saving blank (no edits to modal), and saving 1 line and leaving the others. 

All the monolithic builds have been complete, could you just confirm there isn't another?

MTrab, 2021-01-25:
Nope - no new images now

NotExpectedYet, 2021-01-25:
I've just forced a push through for the development version. 

Would you mind switching over and seeing if that works? I've updated the version to 1.1.9-dev1 so you can see the change. 

It sounds like the image isn't pulling the new one, but it should be so just rather confirm.

NotExpectedYet, 2021-01-25:
10.20 GMT was the last monolightic build I believe so might have just taken some time to go through.

MTrab, 2021-01-25:
Switched to octofarm/octofarm:monolithic-dev, but version still reads 1.1.8 :/

MTrab, 2021-01-25:
And still not saving

NotExpectedYet, 2021-01-25:
It's still building for the 1.1.9-dev1 shouldn't be too long

NotExpectedYet, 2021-01-25:
I can't reproduce at the moment with your original steps. Not sure what's going on at the moment.  

Could you also send through the octofarm-API logs as well please?

MTrab, 2021-01-25:
Custom power on command:
`[PrinterURL]/plugin/enclosure/setIO?status=true&index_id=1&apikey=[PrinterAPI]`

Custom power off command:
`[PrinterURL]/plugin/enclosure/setIO?status=false&index_id=1&apikey=[PrinterAPI]`

Just in case you'd want to try for your selv with the exact strings I use.

MTrab, 2021-01-25:
[OctoFarm-API.log](https://github.com/NotExpectedYet/OctoFarm/files/5865983/OctoFarm-API.log)

NotExpectedYet, 2021-01-25:
I was checking the restart commands... 

I'll take a look at those in a bit sorry, didn't realise it was those as well.

NotExpectedYet, 2021-01-25:
Can you confirm the restart commands are staying now please?

MTrab, 2021-01-25:
The restart commands doesn't save either, so just as well ... ;)

NotExpectedYet, 2021-01-25:
How are you updating the container? as I can't reproduce with the restart commands. 

Just the custom power commands currently. 

Monolithic has certainly built. You should see 1.1.9-dev1 when you update this time.

MTrab, 2021-01-25:
Using a custom update script: https://pastebin.com/yCWe0dUq

Will try an update again

NotExpectedYet, 2021-01-25:
> Using a custom update script: https://pastebin.com/yCWe0dUq
> 
> Will try an update again

No that's perfect, just wanted to confirm that it was updating properly but that should. Some people just restart the container aha. 

I'll check the custom power commands shortly, just let me know if restart still isn't persisting and I'll have a look at that as well.

NotExpectedYet, 2021-01-25:
10:42 that built so should certainly have a new pull. 

Just confused as hell why the restart ain't saving for you aha. Be my goof somewhere.

MTrab, 2021-01-25:
Okay - not it reads 1.1.9-dev1.
Still not saving restart commands thou :/

NotExpectedYet, 2021-01-25:
Grrr. Can't understand that at all. Sorry for the mess around. 

I'll do some tests in a bit see if I can't figure it out.

MTrab, 2021-01-25:
Let me know if you need anything

NotExpectedYet, 2021-01-25:
Some time would be nice aha. 

Be about mid-week I think I'll be able to check as I've got to go back into my day job tomorrow.

MTrab, 2021-01-25:
It's okay.
I've been doing some testing on 1.1.9-dev1 - and it looks like the restart commands get saved afterall, it just returns N/A in the text fields afterwards.
The restart command button becomes visible and works after a resync.

MTrab, 2021-01-25:
[image]
[image]

NotExpectedYet, 2021-01-25:
> It's okay.
> I've been doing some testing on 1.1.9-dev1 - and it looks like the restart commands get saved afterall, it just returns N/A in the text fields afterwards.
> The restart command button becomes visible and works after a resync.

Hmmm that's still weird because on mine it pulls back through... confusing! I wonder if it's something to do with me having long standing data but I tried with a fresh printer setup. 

Aye that makes sense as the push to OctoPrint was working and a re-sync just collects the commands from there and keeps them in memory. Before I was updating OctoPrint with blank values which is why it previously went in circles. I've probably missed something again.

At least those are working. 

The power commands below "might" work the same. The only difference there is they go back to my database rather than updating octoprint. I would make sure all are filled in though.

MTrab, 2021-01-25:
Just tried with the custom commands, those doesn't work still. Will be awaiting what you find :)

NotExpectedYet, 2021-01-25:
Thanks for the confirmation. 

Hopefully I should get some time tomorrow to have a look into it. It's my fault I should have paid more attention this morning so sorry about that.

MTrab, 2021-01-25:
No worry . just happy you'll take the time to look into this :)

NotExpectedYet, 2021-01-25:
Oh OctoFarm is my baby I don't like bugs 😅. I certainly will though.

NotExpectedYet, 2021-01-26:
Yo, just pushed another attempted fix to development if you get chance to try. 

I tested here and all was working fine and updating straight away after a refresh. Fingers crossed same on your end. Will be -dev2

MTrab, 2021-01-26:
Hmmm - Save button seems to not be working in -dev2.
Nothing happens when pressing Save.

NotExpectedYet, 2021-01-26:
Could you send the API log through please?

MTrab, 2021-01-26:
[OctoFarm-API (1).log](https://github.com/NotExpectedYet/OctoFarm/files/5871577/OctoFarm-API.1.log)

NotExpectedYet, 2021-01-26:
Hmm its certainly sending the information and receiving it. 

I'll take another look at the code in a bit and see if I missed something.

NotExpectedYet, 2021-01-26:
Does anything come up in the browser console window when you press save?

MTrab, 2021-01-26:
No, nothing comes up. And the commands seems not to be saved either.

NotExpectedYet, 2021-01-26:
Press F12 on your browser and click the console tab. 

Then press save, it will show errors if there are any.

MTrab, 2021-01-26:
Nothing happens in the console

MTrab, 2021-01-26:
When pressin save WITHOUT entering anything, the save button works

NotExpectedYet, 2021-01-26:
What nodejs version are you running?

MTrab, 2021-01-26:
Using 12.16.3-0017

NotExpectedYet, 2021-01-26:
Not that then hmmm

NotExpectedYet, 2021-01-28:
Hello, I have made some tweaks and pushed a new version to development. Will be -dev3. Would you mind testing? 

If it fails again this time I have bolstered the State log to show what's actually going on. Should at least either fix it or allude me to what is happening.

MTrab, 2021-01-28:
Still no luck :(
[OctoFarm-API (2).log](https://github.com/NotExpectedYet/OctoFarm/files/5886588/OctoFarm-API.2.log)
[OctoFarm-State.log](https://github.com/NotExpectedYet/OctoFarm/files/5886596/OctoFarm-State.log)

NotExpectedYet, 2021-01-28:
Your log provided only goes up to 25th of January, need the attempts today please

MTrab, 2021-01-28:
Strange - tried right after upgrade of the container.
I'll give it another try.

NotExpectedYet, 2021-01-28:
You need a OctoFarm-State#.log that file without the number is the oldest.

MTrab, 2021-01-28:
Remember - my save button doesn't seem to work when I have anything entered in the boxes.
[OctoFarm-State1.log](https://github.com/NotExpectedYet/OctoFarm/files/5886659/OctoFarm-State1.log)

NotExpectedYet, 2021-01-28:
Aye I know there we're no errors in the browser so that would be the server not responding correctly to the request. 

{
  "state": "Disconnected",
  "printer": {
    "printerName": "",
    "printerURL": "",
    "index": "60096307de188800262631e0",
    "cameraURL": "",
    "apikey": ""
  },
  "connection": {
    "preferredPort": "0",
    "preferredBaud": "115200",
    "preferredProfile": "_default"
  },
  "systemCommands": {
    "serverRestart": "sudo service octoprint restart",
    "systemRestart": "sudo shutdown -r now",
    "systemShutdown": "sudo shutdown -h now"
  },
  "powerCommands": {
    "powerOnCommand": "",
    "powerOnURL": "[PrinterURL]/plugin/enclosure/setIO?status=true&index_id=1&apikey=[PrinterAPI]",
    "powerOffCommand": "",
    "powerOffURL": "[PrinterURL]/plugin/enclosure/setIO?status=false&index_id=1&apikey=[PrinterAPI]",
    "powerToggleCommand": "",
    "powerToggleURL": "",
    "powerStatusCommand": "",
    "powerStatusURL": "",
    "wol": {
      "enabled": false,
      "ip": "",
      "port": "",
      "interval": "",
      "count": "",
      "MAC": ""
    }
  },
  "costSettings": {
    "powerConsumption": null,
    "electricityCosts": null,
    "purchasePrice": null,
    "estimateLifespan": null,
    "maintenanceCosts": null
  },
  "profileID": "_default",
  "profile": {
    "name": null,
    "color": "default",
    "model": null,
    "volume": {
      "formFactor": "rectangular",
      "width": null,
      "depth": null,
      "height": null
    },
    "heatedBed": true,
    "heatedChamber": false,
    "axes": {
      "x": {
        "speed": null,
        "inverted": false
      },
      "y": {
        "speed": null,
        "inverted": false
      },
      "z": {
        "speed": null,
        "inverted": false
      },
      "e": {
        "speed": null,
        "inverted": false
      }
    },
    "extruder": {
      "count": null,
      "nozzleDiameter": null,
      "sharedNozzle": false
    }
  },
  "gcode": {
    "afterPrintCancelled": "",
    "afterPrintDone": "",
    "afterPrintPaused": "",
    "afterPrinterConnected": "",
    "afterToolChange": "",
    "beforePrintResumed": "",
    "beforePrintStarted": "",
    "beforePrinterDisconnected": "",
    "beforeToolChange": ""
  },
  "other": {
    "enableCamera": true,
    "rotateCamera": false,
    "flipHCamera": false,
    "flipVCamera": false,
    "enableTimeLapse": true,
    "heatingVariation": "",
    "coolDown": ""
  }
}

I can see the request been sent to the server. 

But there is nothing in the state log with my new logging. Nothing at all. That seems to me like that hasn't updated, your certainly on -dev3?

[image]

`Thu, 28 Jan 2021 10:50:52 GMT | INFO | OctoFarm-State.log | Grabbed 2 for checking | `

That was the last reboot of OctoFarm for your server and is before the monolithic image actually pushed so as far as I can tell by the logs you haven't updated anything.

[image]

MTrab, 2021-01-28:
[image]
I'll try updating again, perhaps something missed?!

[63 more replies]

commit 7d6d3ac5 (2021-01-23): Fixed #234
commit aea8c41e (2021-01-25): fix for settings clearing out OctoPrints Fixed #234 again!
commit db781e46 (2021-01-26): Fixed #234 - Adding Power and Custom Power commands won't save
commit 41bfc856 (2021-01-29): Fixed #234 - Power settings. ACTUALL!
closed 2021-01-29 by NotExpectedYet