OctoFarm/OctoFarm #237: Changed port in OctoFarm settings from 4000 to 80 and it broke

JS3910, 2021-01-26:
I found the added port on the web interface server to be pointless and would rather have all the Pi's in the farm just be accessible on their own hostname alone. I changed this setting in the system settings and now the web interface won't start up.

pm2.error.log provides an endless spam of
```
2021-01-26T16:47:37: Error: listen EACCES: permission denied 0.0.0.0:80
2021-01-26T16:47:37:     at Server.setupListenHandle [as _listen2] (net.js:1299:21)
2021-01-26T16:47:37:     at listenInCluster (net.js:1364:12)
2021-01-26T16:47:37:     at Server.listen (net.js:1450:7)
2021-01-26T16:47:37:     at Function.listen (/home/ubuntu/OctoFarm/node_modules/express/lib/application.js:618:24)
2021-01-26T16:47:37:     at serverStart (/home/ubuntu/OctoFarm/app.js:116:9)
2021-01-26T16:47:37:     at processTicksAndRejections (internal/process/task_queues.js:97:5) {
2021-01-26T16:47:37:   code: 'EACCES',
2021-01-26T16:47:37:   errno: 'EACCES',
2021-01-26T16:47:37:   syscall: 'listen',
2021-01-26T16:47:37:   address: '0.0.0.0',
2021-01-26T16:47:37:   port: 80
2021-01-26T16:47:37: }
```

OctoFarm-Server.log endlessly repeats
```
Tue, 26 Jan 2021 16:48:28 GMT | INFO | OctoFarm-Server.log | Checking Server Settings... | 
Tue, 26 Jan 2021 16:48:28 GMT | INFO | OctoFarm-Server.log | Server settings already exist, loaded existing values... | 
Tue, 26 Jan 2021 16:48:28 GMT | INFO | OctoFarm-Server.log | MongoDB Connected... | 
Tue, 26 Jan 2021 16:48:28 GMT | INFO | OctoFarm-Server.log | Grabbing plugin list from OctoPrint.org | 
Tue, 26 Jan 2021 16:48:28 GMT | INFO | OctoFarm-Server.log | Updated Plugin List... | 
Tue, 26 Jan 2021 16:48:28 GMT | INFO | OctoFarm-Server.log | Farm information inititialised... | 
Tue, 26 Jan 2021 16:48:28 GMT | INFO | OctoFarm-Server.log | Checking Client Settings... | 
Tue, 26 Jan 2021 16:48:28 GMT | INFO | OctoFarm-Server.log | Client settings already exist, loaded existing values... | 
Tue, 26 Jan 2021 16:48:28 GMT | INFO | OctoFarm-Server.log | Printer Runner has been initialised... | data:"System Runner has checked over 4 printers..." | 
Tue, 26 Jan 2021 16:48:28 GMT | INFO | OctoFarm-Server.log | Starting System Information Runner... | 
Tue, 26 Jan 2021 16:48:31 GMT | INFO | OctoFarm-Server.log | Started system information collection... | 
Tue, 26 Jan 2021 16:48:32 GMT | INFO | OctoFarm-Server.log | No settings or disabled for influxdb export | 
```

There is nothing on this pi listening to or blocking port 80 that I can find, and I can't for the life of me find any config file where I can change the web interface port back to 4000 so that I can at least use it.

JS3910, 2021-01-26:
I'm guessing it's related to this:
https://stackoverflow.com/questions/60372618/nodejs-listen-eacces-permission-denied-0-0-0-080

I haven't tried to work around it yet but obviously this is something the software/web interface should handle more gracefully :p.

NotExpectedYet, 2021-01-26:
You'd need to change it in the database. Try mongodb compass.

NotExpectedYet, 2021-01-26:
That's system config by the looks, nothing to do with nodejs. It can't reach privileged it's blocked from at a system level. A Linux safe user ie.. your sudo user which will be Pi is denied access to port 80. Nothing I can do, unless you'd like to get on to the rasbian Devs you will have to remove the block yourself.

tideline3d, 2021-01-26:
A reverse proxy like HAProxy (similar to what OctoPi does) is the preferred approach to getting an app onto a privileged port (<1024) as you really don't want to be running random applications as root.

NotExpectedYet, 2021-01-27:
I'm going to close this as the original question has been answered.

closed 2021-01-27 by NotExpectedYet

JS3910, 2021-01-28:
> A reverse proxy like HAProxy (similar to what OctoPi does) is the preferred approach to getting an app onto a privileged port (<1024) as you really don't want to be running random applications as root.

Whatever octopi does seems reasonable, and should be part of the default install. For now just starting it elevated (sudo npm start) is a workaround. Though to be honest, Octofarm uses Docker already which almost by definition runs "random" apps as root in the first place so it's hard to take issue with it.

On a side note, is there any documentation at all? The front readme says "All user documentation is now moving to OctoFarm.net. Development documentation below." and the wiki says "Notice: This place will soon be depricated in favour of https://octofarm.net. These guides are mainly out of date.". There is however no apparent links or even remotest mention of documentation on octofarm.net apart from the installation instructions so that's a bit... confusing.

NotExpectedYet, 2021-01-28:
I don't have an image, it's a straight installed application so it's down to the user at the moment to setup their own methods of connecting outside of port 4000 as I don't control the OS you set it up on. OctoPi is a seperate image and not even maintained by Gina who does OctoPrint as a full time job as I'm aware. Me making an image similar to that is currently outside of any available time I have for the project. Docker would be a much better choice for what your trying to do anyway as that runs as sudo regardless so won't hit the same limitation. 

There's installation instructions on the website and currently what ever is on the wiki, it's been a last thought to be honest and I do agree with you it requires sorting out. The message was more just to allude to not using those installation instructions, but the alerts and plugin documents still stand. 

I'm eventually moving it all over to OctoFarm but as it's a side project at the moment it's when I get the time for it sadly. I make up for it by been rather responsive to questions asked so if there's anything you'd like to know more about, like a function or direction then please feel free to open a discussion board. 

I'm also happy to take commits towards documentation when I've finally gotten that setup behind it's own repo for OctoFarm.net to pull from and for someone to setup their own pre-made image but nobody's taken on the task as of yet. OctoFarm wouldn't be on docker unless a very helpful user didn't contribute it. OctoPi probably might not exist in it's same capacity as well, although I don't know the full story of how that came to be. 

I will update that message though, thanks for pointing that out.