OctoFarm/OctoFarm #239: OctoFarm is not working correctly

3DFans, 2021-01-28:
Hey
OctoFarm is installed so far and the printers all show up. However, when I enter the overview, it only says "Awaiting Websocket" and when I want to display the cameras, they load briefly and then disappear. No processes are visible in the dashboard itself, although the printers and APIs are correct.

NotExpectedYet, 2021-01-28:
Awaiting websockets means you've either not got CORS enabled, or the API key you've used is the wrong one. You need to generate an Application / User key now for anything later than OctoPrint 1.4.2. Instructions are on the website https://octofarm.net if you don't know how to create one.

3DFans, 2021-01-28:
CORS is enabled and API is also correct, but to no avail.

API checks successful

NotExpectedYet, 2021-01-28:
You'll need to open a bug report and provide some logs then please. 

Just to mention again, if your using the global API key that will not work and you will see exactly what your seeing.

3DFans, 2021-01-28:
Hey
I also created a new user and then assigned the API and then entered the "new API", but without success.
The printers are also all displayed offline, but I can restart them, so there is a connection, but the printers are not displayed.

closed 2021-01-28 by 3DFans

NotExpectedYet, 2021-01-28:
> haha problem solved. I didn't save it. :D:D

No worries XD