OctoFarm/OctoFarm #240: [BUG] Sending commands from Octofarm seems malformed

MTrab, 2021-01-29:
### To better help the developer understand your issue, please fill in the information below. Failing to do so will waste the developer and your own time in sorting the issue. Thanks!

**Describe the bug**
When sending a command from Octofarm, it looks like the string is formattet in some strange way, resulting in Octoprint responding with an error

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Printer'
2. Write G or M command in text field
3. Press send
4. See error

**Expected behavior**
M or G codes being sent correctly, as entered in the text field.

**Screenshots**
[image]

**Versions (please complete the following information):**
 - NodeJS Version [e.g. 13] 12.16.3-0017
 - OctoFarm Version [e.g. 1.5.5.7] 1.1.9-dev5
 - OctoPrint Version [e.g. 1.4.1] 1.5.3
 - OctoPrint Plugins [e.g. bed visualiser, octoklipper, printtimegenius] Enclosure Plugin

**Additional context**

NotExpectedYet, 2021-01-30:
Could you screenshot the console output please? Think something else is going on here as I can't reproduce with your command.

[image]

MTrab, 2021-01-30:
Console gives absolutely no output on this

NotExpectedYet, 2021-01-30:
I'll need to repro to figure it out as it's client side and that's all compiled for the user. 

When typing in the command for me it sends fine, even with mutli-lined sends. Same for enter fire, and Button press fire.

MTrab, 2021-02-03:
Issue is the same in all browser and on both printers.
Just a tought - could this be some kind of language interpretation?
My system language is Danish, so is my input locale.

NotExpectedYet, 2021-02-03:
> Issue is the same in all browser and on both printers.
> Just a tought - could this be some kind of language interpretation?
> My system language is Danish, so is my input locale.

That's a good thought. I gather nothing comes up in the console when you send it apart from the octoprint error? 

I will have a search and see if there's any issues with language parsing from an input. If that is the case it's probably my new line interpretation or that's my guess. 

I presumed it would be fine now a days. I'll put some more console outputs in a push tomorrow for you to try if you'd have time? 

Just trying to cover all basis for now, not sure which one of the above would be most fruitful.

NotExpectedYet, 2021-02-03:
What's confusing is theres that little u before the array quotes which I've never seen before.

MTrab, 2021-02-03:
I'll fire up my work laptop later today, it's with EN-US localization - the we can see if that makes a difference.

NotExpectedYet, 2021-02-03:
It's all sent and managed client side yet so fingers crossed. I'll see about putting some UTF-8 encoding on the field should resolve it.

MTrab, 2021-02-03:
Nope - still doesn't work on the laptop.
But now the terminal windows actually are not showing anything anymore :/

I can see on the Octoprint terminal pane that the error are the same.

NotExpectedYet, 2021-02-03:
It's something else then. I wonder if it's related to the profile bug. Could you fire me through your state and API logs please?

MTrab, 2021-02-03:
[OctoFarm-API (1).log](https://github.com/NotExpectedYet/OctoFarm/files/5920244/OctoFarm-API.1.log)
[OctoFarm-State5.log](https://github.com/NotExpectedYet/OctoFarm/files/5920247/OctoFarm-State5.log)

MTrab, 2021-02-06:
Unfortunately still an issue :(

From console
[192.168.1.9-1612643408065.log](https://github.com/NotExpectedYet/OctoFarm/files/5937590/192.168.1.9-1612643408065.log)

NotExpectedYet, 2021-02-07:
Well this one's a PIA, even worse the array is there formatted perfect in the error output :/

NotExpectedYet, 2021-02-13:
Right so I've double checked everything on this side of things, the function, the log... it for all intensive purposes is sending the correct array, and that's even outputted in your log you sent. I see the individual key press G, 2, 8 and then the array created on the input ["G28"]. 

I put another set of logging in to see what's getting fired. 

[image]

You should have something like the above.

If that is showing correctly, then I'd say this was something to do with OctoPrint specifically.

MTrab, 2021-02-21:
This is what I see in 1.1.11-dev12

[image]

NotExpectedYet, 2021-03-05:
I've found a bug, somehow that Array is getting double quoted before sending but I can't understand why that is only not working on your system. 

I'll update the title to reflect this and have a look.

commit 24ec1e62 (2021-02-06): Fixed #193
Fixed #240
commit a2561835 (2021-02-06): Fixed #193
Fixed #240
PR #368 (2021-03-17): Bugfix/double wrapped terminal commands
Fixed #240
closed 2021-04-24 by NotExpectedYet