OctoFarm/OctoFarm #242: [BUG] WebSocket permanently in tentative state

eovnu87435ds, 2021-02-02:
### To better help the developer understand your issue, please fill in the information below. Failing to do so will waste the developer and your own time in sorting the issue. Thanks!

**Describe the bug**
I made a fresh install of octofarm on an Ubuntu 20.04.1 x64 server VM running on a vSphere ESXi 6.7 instance. After spinning up the VM, I followed the instructions on the webpage to install OF and its dependencies. When adding printers, I can establish a connection, and commands like updating octoprint work successfully, but WebSocket Status stays at "Websocket Connected but in Tentative state until receiving data."

In addition, the navigation tabs Panel, List, and Camera all show elements for about half of a second before they disappear. 

I was able to use the "nuke all data" function to reset octofarm, and replicate the problem, so that there would be less noise in the logs.

I have 2 printers, each with Raspi3b+. One is on Octoprint 1.5.3 and the other on 1.4.2. Both are using application keys and have CORS enabled.

I have attached a video showing the issue, as well as the logs. 

_Yes, I revoked all API keys after uploading this._

**Versions (please complete the following information):**
 - NodeJS Version 12.20.1
 - OctoFarm Version 1.1.9
 - OctoPrint Version 1.5.3 and 1.4.2
[octofarm.mp4](https://drive.google.com/file/d/1z3cLKxoLnP4p2U8JX_SNRviLf55jjDo4/view?usp=sharing)
[OctoFarm-logs.zip](https://github.com/NotExpectedYet/OctoFarm/files/5908496/OctoFarm-logs.zip)

**Additional context**
Add any other context about the problem here.

NotExpectedYet, 2021-02-02:
Hey man, this seems like something on your OctoPrint instance. 

Basically the websocket is not getting any authenticated from OctoPrint, so I'm not receiving any confirmation response from it. 

data:{"auth":"evan:F0C8DA4C035243C4906B6E63D846840A"}

It's attempting with that user, and that websocket authentication key, but it's incorrect. 

Could you send through your OctoPrint logs for when this tries to connect? It's something on the instance. Honestly I'd say you we're still using the global key from my logs, but I get weird issues with some OctoPrint instances and this, you can check the git issue history. PIA. 

I had one user get a similar issue and worked when he swapped SD card, but let's start with octoprint logs aha

NotExpectedYet, 2021-02-02:
I think this might be down to your multiple users?, I can take a wild guess that the user OctoFarm chooses is not the user you generated the key from?

NotExpectedYet, 2021-02-02:
`Sending Auth to Websocket: 1: ws://192.168.6.52/sockjs/websocket  | data:{"auth":"evan:9C86399F1191431B9405325FF2FF2703"} | `

Even for 192.168.6.52
Sending Auth to Websocket: 0: ws://192.168.6.51:80/sockjs/websocket  | data:{"auth":"bfd:7783D1DB489F4D1B97CD3E1AFE371DD6"} | 

BFD for 182.168.6.51

eovnu87435ds, 2021-02-02:
Thanks for the investigation! I hadn't noticed that the application key was not matching the auto-populated user name. I have 3 admin and 5 users on each instance. As a backwards way of dealing with it, I made an app key with 1 username, populated it into octofarm, saw what username got populated there, made a new app key with that user name, and replaced the old key with that one.

NotExpectedYet, 2021-02-02:
> Thanks for the investigation! I hadn't noticed that the application key was not matching the auto-populated user name. I have 3 admin and 5 users on each instance. As a backwards way of dealing with [...]

No trouble at all! Glad it's working. 

I haven't really considered multiple user setups before sorry for not thinking of it on Discord. Basically I try to do the a lot of what would usually be UI work for the user and enter that in but there's always edge cases it misses like this. I've no idea how OctoPrint orders them as well but I will have a think of how to handle this more gracefully for future. What I am hoping is your work around will stick because I don't think OctoPrint would randomly change the array order it sends back. 

I don't really want to go to the lengths of supporting OctoPrints user system because I want to bring that feature in OctoFarm itself and whilst been a beefy undertaking itself I also plan to support marina, klipper, dwc and other interfaces that exist, so would just be too much to maintain across all those systems. An alert or at least a notification of what user was there been more obvious might help though. If you have any thoughts they are welcome.

eovnu87435ds, 2021-02-02:
Honestly, I think the 1 user limitation is totally fine. In fact, I was going to reduce the users anyways, as pretty much all maintenance can be done in OF. 

Perhaps if the username field is editable in octofarm, it could be manually made to match whatever app key you use? Between that and updating the octofarm popup instructions for adding a printer, I think many users would be covered.

NotExpectedYet, 2021-02-06:
I'll have a look at making the user field editable, it's not registered at login so I think I'll include this in my 1.2 changes for the refactor. 

I'm going to close this.

closed 2021-02-06 by NotExpectedYet