OctoFarm/OctoFarm #243: History not showing up [BUG]

bdw1980, 2021-02-02:
### To better help the developer understand your issue, please fill in the information below. Failing to do so will waste the developer and your own time in sorting the issue. Thanks!

**Describe the bug**
I have printed several prints since I recently did an install on pi4 and none of the History is showing up.

**To Reproduce**
None to Reproduce

**Expected behavior**
Should be collecting logs

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Versions (please complete the following information):**
 - NodeJS Version [e.g. 13]
 - OctoFarm Version 1.1.9
 - OctoPrint Version 
 - OctoPrint Plugins none
[OctoFarm-API.log](https://github.com/NotExpectedYet/OctoFarm/files/5913637/OctoFarm-API.log)
[OctoFarm-FilamentManager.log](https://github.com/NotExpectedYet/OctoFarm/files/5913638/OctoFarm-FilamentManager.log)
[OctoFarm-HistoryCollection.log](https://github.com/NotExpectedYet/OctoFarm/files/5913639/OctoFarm-HistoryCollection.log)
[OctoFarm-InformationCleaning.log](https://github.com/NotExpectedYet/OctoFarm/files/5913640/OctoFarm-InformationCleaning.log)
[OctoFarm-InformationCleaning1.log](https://github.com/NotExpectedYet/OctoFarm/files/5913641/OctoFarm-InformationCleaning1.log)
[OctoFarm-InformationCleaning2.log](https://github.com/NotExpectedYet/OctoFarm/files/5913642/OctoFarm-InformationCleaning2.log)

**Additional context**
Add any other context about the problem here.

NotExpectedYet, 2021-02-02:
Brilliant cheers man, will have a look.

NotExpectedYet, 2021-02-02:
Save your System settings man. Sorry seems I've still got a bug there. 

Go to System, click on Server and press save should start capturing history again.

NotExpectedYet, 2021-02-03:
I've updated the title to reflect the actual bug.

commit e92d4db2 (2021-02-06): Fixed #243
commit fde27343 (2021-02-06): Fixed #243
closed 2021-02-06 by NotExpectedYet