OctoFarm/OctoFarm #246: [BUG] Cancel and pause buttons inactive

MTrab, 2021-02-03:
### To better help the developer understand your issue, please fill in the information below. Failing to do so will waste the developer and your own time in sorting the issue. Thanks!

**Describe the bug**
During a print, the pause and cancel buttons in the detailed printer window is inactive.
To stop a print I need to go to the list or some other view showing all printers, and press stop.

**To Reproduce**
Steps to reproduce the behavior:
1. Start a print
2. Go to the detailed printer view
3. Note both pause and cancel buttons are not active

**Expected behavior**
Being able to press cancel and pause from the detailed view

**Screenshots**
[image]

**Versions (please complete the following information):**
- NodeJS Version [e.g. 13] 12.16.3-0017
- OctoFarm Version [e.g. 1.5.5.7] 1.1.9-dev5
- OctoPrint Version [e.g. 1.4.1] 1.5.3
- OctoPrint Plugins [e.g. bed visualiser, octoklipper, printtimegenius] Enclosure Plugin

**Additional context**
Add any other context about the problem here.

NotExpectedYet, 2021-02-03:
Hello. Could you open the printer control and show me the console output of the browser? F12 should open the Dev tools for you. 

There's something failing that's why those arn't hiding and activating.

MTrab, 2021-02-03:
```
listViewRunner.min.js:4901 Uncaught (in promise) TypeError: e[t].length.toFixed is not a function
    at listViewRunner.min.js:4901
    at Array.forEach (<anonymous>)
    at listViewRunner.min.js:4899
    at u (serverAliveCheck.min.js:6522)
    at Generator._invoke (serverAliveCheck.min.js:6752)
    at Generator.next (serverAliveCheck.min.js:6577)
    at f (listViewRunner.min.js:3369)
    at o (listViewRunner.min.js:3371)
    at listViewRunner.min.js:3371
    at new Promise (<anonymous>)
(anonymous) @ listViewRunner.min.js:4901
(anonymous) @ listViewRunner.min.js:4899
u @ serverAliveCheck.min.js:6522
(anonymous) @ serverAliveCheck.min.js:6752
(anonymous) @ serverAliveCheck.min.js:6577
f @ listViewRunner.min.js:3369
o @ listViewRunner.min.js:3371
(anonymous) @ listViewRunner.min.js:3371
(anonymous) @ listViewRunner.min.js:3371
(anonymous) @ listViewRunner.min.js:5059
(anonymous) @ listViewRunner.min.js:3464
u @ serverAliveCheck.min.js:6522
(anonymous) @ serverAliveCheck.min.js:6752
(anonymous) @ serverAliveCheck.min.js:6577
f @ listViewRunner.min.js:3369
o @ listViewRunner.min.js:3371
Promise.then (async)
f @ listViewRunner.min.js:3369
o @ listViewRunner.min.js:3371
Promise.then (async)
f @ listViewRunner.min.js:3369
o @ listViewRunner.min.js:3371
(anonymous) @ listViewRunner.min.js:3371
(anonymous) @ listViewRunner.min.js:3371
(anonymous) @ listViewRunner.min.js:3506
(anonymous) @ listViewRunner.min.js:5856
15listViewRunner.min.js:4901 Uncaught (in promise) TypeError: e[t].length.toFixed is not a function
    at listViewRunner.min.js:4901
    at Array.forEach (<anonymous>)
    at listViewRunner.min.js:4899
    at u (serverAliveCheck.min.js:6522)
    at Generator._invoke (serverAliveCheck.min.js:6752)
    at Generator.next (serverAliveCheck.min.js:6577)
    at f (listViewRunner.min.js:3369)
    at o (listViewRunner.min.js:3371)
    at listViewRunner.min.js:3371
    at new Promise (<anonymous>)
```

NotExpectedYet, 2021-02-03:
Thanks! I'll get that sorted.

NotExpectedYet, 2021-02-06:
This got fixed, think you just need to update your version. Could you check to see if the terminal command is still doing the same issue from #240 please? 

I'm wondering if those we're related: 
Version 1.1.10-dev4

I've added in more information to an error output as well, hopefully track it down if not.

MTrab, 2021-02-06:
Confirmed fixed in 1.1.10-dev4

NotExpectedYet, 2021-02-06:
Same for terminal commands?

closed 2021-02-06 by NotExpectedYet