OctoFarm/OctoFarm #254: [FEATURE REQUEST] Set Timezone in Octofarm Webinterface

OXERY, 2021-02-04:
Setting the correct timezone is possible in Docker using `TZ="Europe/Berlin"` but not in the Octofarm webinterface.. (regarding estimated time of arrival display)

NotExpectedYet, 2021-02-04:
Feature requests are in discussions now. But the client uses browser local time and the server uses system and you can already define the time zone in your docker command so could you explain a bit more please I am probably missing something?

OXERY, 2021-02-05:
Sorry, my brain was "offline" after 9 hours of work and some anger about my 3d printer.

Some people are not using octofarm in Docker, therefore aren't able to change the timezone, if their knowledge is not deep enough to do this in their respective linux OS. Just wanted to ask if it also could be set in the octofarm webinterface.

NotExpectedYet, 2021-02-05:
> Sorry, my brain was "offline" after 9 hours of work and some anger about my 3d printer.
> 
> Some people are not using octofarm in Docker, therefore aren't able to change the timezone, if their knowledge is not deep enough to do this in their respective linux OS. Just wanted to ask if it also could be set in the octofarm webinterface.
> 
> btw: I'm pretty happy with Octofarm :) Nice work

Aha no worries, I'm generally in that state myself so like to check. Currently the work required to update the interface to allow for that wouldn't be worth it, but 1.2 is re-writing a lot of stuff and generally doing a massive clean up with a new web interface and everything else. 

So just to confirm I'm understanding you (was rudely awoken this morning by work on my day off XD) you want OctoFarm to be able to set a timezone and everything conform to that. ie.. the backend and the client side. 

It would, as you said, currently require a system level change which not everyone is able to do which is no good for none technical people. Would you mind opening a Discussion regarding this? We can continue it there. I'm just trying to keep the issues as a list of planned work currently.

OXERY, 2021-02-05:
Mentioned it in discussion #255 :) Thanks!!

closed 2021-02-05 by OXERY