OctoFarm/OctoFarm #256: [QUESTION] Interested in merging printer map module

InfoMTB3D, 2021-02-05:
I've started working on a printer location map tab based on the latest OctoFarm.
[image]
Is there interest into merging this into OctoFarm?

NotExpectedYet, 2021-02-05:
I am interested totally, this sounds cool. 

I am trying to think currently how this will work with my current changes working towards 1.2. Initially I have had reservations of taking commits to the main code because the current base is a mess, it's not modular and it's got a lot of my learning errors in it. There's also maintainability, so I'd like to obviously understand the code to keep it up if you decided to no longer, or someone wanted an extra feature I could add myself and you weren't available.  

Do you have a branch I can look at and test out across my testing instances? I am happy to work with you on this totally but all the work on 1.2 will  be going on a private branch for patreons initially because it's about time I gave my supporters something extra. That's whilst it's in development then Private will stay a version or two in front. 

Would you like to open a discussion and we can flesh this out further. Happy for the help, and it sounds like a cool feature.

InfoMTB3D, 2021-02-05:
No worries, I will PR after I remove our private code. Also, I will consider to join in open-source wise because I have quite some experience I can share with you. I'll shoot you an email for that.

Discussion is made, let's talk!

closed 2021-02-05 by InfoMTB3D