OctoFarm/OctoFarm #265: [QUESTION]

Killajoedotcom, 2021-02-07:
I've attempted to run Octofarm on my windows PC. I can get through the install, however npm doesn't start. I continue to get the error message below. Could anyone help me understand what's going on and provide a way to correct this?

npm WARN EBADENGINE Unsupported engine {
npm WARN EBADENGINE   package: 'clinic@6.0.3',
npm WARN EBADENGINE   required: { node: '^14.0.0 || ^13.0.0 || ^12.0.0 || ^11.0.0 || ^10.12.0' },
npm WARN EBADENGINE   current: { node: 'v15.8.0', npm: '7.5.1' }
npm WARN EBADENGINE }

NotExpectedYet, 2021-02-07:
Install the correct version of node by the looks

NotExpectedYet, 2021-02-07:
Sorry was rushing, you'll need version 12 of node not 15. I only support LTS versions, and next will be V14.

Killajoedotcom, 2021-02-07:
I've downgraded to the node v12. I'm getting an entirely different error now.

C:\OctoFarm\OctoFarm-master\config>npm start

> octofarm@1.1.10 prestart C:\OctoFarm\OctoFarm-master
> npm install --production

npm WARN babel-loader@8.1.0 requires a peer of webpack@>=2 but none is installed. You must install peer dependencies yourself.

audited 1955 packages in 8.077s
found 1 low severity vulnerability
  run `npm audit fix` to fix them, or `npm audit` for details

> octofarm@1.1.10 start C:\OctoFarm\OctoFarm-master
> pm2 flush && pm2 start app.js --name OctoFarm -e 'logs/pm2.error.log' --time

'pm2' is not recognized as an internal or external command,
operable program or batch file.
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! octofarm@1.1.10 start: `pm2 flush && pm2 start app.js --name OctoFarm -e 'logs/pm2.error.log' --time`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the octofarm@1.1.10 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

NotExpectedYet, 2021-02-07:
That's actually my guff, I've only just updated the instructions. 

`npm install pm2 -g` 

Then run the `npm start` again you should be good to go. Sorry about that.

Killajoedotcom, 2021-02-07:
James, that was the fix! Thanks so much.

closed 2021-02-07 by Killajoedotcom

NotExpectedYet, 2021-02-07:
> James, that was the fix! Thanks so much.

Most welcome man. Enjoy!