OctoFarm/OctoFarm #266: [DEP] Update NodeJS LTS version to V14. New LTS version of Node.

NotExpectedYet, 2021-02-07:
### To better help the developer understand your issue, please fill in the information below. Failing to do so will waste the developer and your own time in sorting the issue. Thanks!

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Versions (please complete the following information):**
 - NodeJS Version [e.g. 13]
 - OctoFarm Version [e.g. 1.5.5.7]
 - OctoPrint Version [e.g. 1.4.1]
 - OctoPrint Plugins [e.g. bed visualiser, octoklipper, printtimegenius]

**Additional context**
Add any other context about the problem here.

davidzwa, 2021-04-07:
@NotExpectedYet this is already on dev right now. Should it go to 1.1.13?

NotExpectedYet, 2021-04-07:
> @NotExpectedYet this is already on dev right now. Should it go to 1.1.13?
> Or maybe we have to introduce 1.2 and work from there?

Is already on Dev, I just need to make some on the fly changes to the current website instructions. 

Gonna add update instructions too. Will sort from Friday then is fine for 1.1.13 when we've done all that.

closed 2021-04-24 by NotExpectedYet