OctoFarm/OctoFarm #271: [BUG] tplink power plugin not working

mkevenaar, 2021-02-11:
**Describe the bug**
When setting up the tplink plugin like in the [docs](https://github.com/NotExpectedYet/OctoFarm/wiki/Setup-OctoFarms-Power-button#tplink-smart-plugin-doesnt-allow-toggle), the power isn't changed when trying to power on the printer

I have set the IP address to the address of the printer. Using curl from the octofarm machine works:

```bash
# curl -s -H "Content-Type: application/json" -H "X-Api-Key: <DELETED>" -X POST -d '{ "command":"checkStatus", "ip":"<DELETED>" }' http://octopi/api/plugin/tplinksmartplug
{"currentState":"off","emeter":null,"ip":"<DELETED>"}
# curl -s -H "Content-Type: application/json" -H "X-Api-Key: <DELETED>" -X POST -d '{ "command":"turnOn", "ip":"<DELETED>" }' http://octopi/api/plugin/tplinksmartplug
{"currentState":"on","emeter":null,"ip":"<DELETED>"}
# curl -s -H "Content-Type: application/json" -H "X-Api-Key: <DELETED>" -X POST -d '{ "command":"turnOff", "ip":"<DELETED>" }' http://octopi/api/plugin/tplinksmartplug
{"currentState":"off","emeter":null,"ip":"<DELETED>"}
```

**To Reproduce**
Steps to reproduce the behavior:
1. Setup the TP Link as described in the documentation
2. On the printer window, below actions click on the power button and click on "Power On Printer"
3. Printer isn't turned on

**Expected behavior**
The power to turn on or off depending on the status

**Screenshots**

**Versions (please complete the following information):**
 - NodeJS Version 12.20.2
 - OctoFarm Version 1.1.10
 - OctoPrint Version 1.5.3
 - OctoPrint Plugins TP-Link Smartplug (0.9.26) is the most important one.
 - OctoPi 0.18.0

**Additional context**
I haven't found anything in the log why it could have failed or not working

NotExpectedYet, 2021-02-11:
Hello, thanks for the report. 

I believe this is related to the issue I'm seeing with the enclosure plugin too. #263 

I gather it's a CORS error same as that but does anything output in the browser window when you press the power button? 

There's still legacy code in OctoFarm from when it was a single page application so those commands are fired client side until I get that all moved to the backend where it should be.

NotExpectedYet, 2021-02-11:
I'm doing a bug fix release over the weekend anyway so I'll add this on this list to look into with the enclosure plugin.

mkevenaar, 2021-02-11:
@NotExpectedYet If you need / want me to test something, let me know

Thanks for responding so quickly!

NotExpectedYet, 2021-02-11:
> @NotExpectedYet If you need / want me to test something, let me know
> 
> Thanks for responding so quickly!

Welcome :) 

Just the browser output if you could when you've fired the command. F12 and then the console tab will get you there. Just wanna confirm it's same issue. 

I'll give you a shout soon as I've got something to try. It will probably be on development for a bit and then master once confirmed.

mkevenaar, 2021-02-11:
@NotExpectedYet I think I have sort of fixed it. It seems that the API call is called from your browser, and not the octofarm server. I have been using VPN to connect to the server and it could not resolve the `octopi.local` address. I have changed it to the FQDN of the machine and now I am able to turn on / off the printer.

However I do get the following message:

[image]

NotExpectedYet, 2021-02-11:
> @NotExpectedYet I think I have sort of fixed it. It seems that the API call is called from your browser, and not the octofarm server. I have been using VPN to connect to the server and it could not [...]

Ahhhh makes sense. When I mentioned above about the legacy code? sent from the client... that's what I meant the browser. Yeah you'll sadly have to either use the ip:port combination which will work fine over VPN or use a local domain name not octopi.local as not every client will know how to route that.

NotExpectedYet, 2021-02-11:
I'll fix the error though! aha that's incorrectly firing. 

Could you show me the console output from that? if there is any.

mkevenaar, 2021-02-11:
I checked, there isn't any. It's empty.

NotExpectedYet, 2021-02-11:
Sounds about right for me, thanks for confirming! 

I'll update the title to reflect the actual issue. 

Regarding the stuff getting sent client side, that's going to be sorted for 1.2 as it's a hefty refactor of the code. I'm looking at a few months too before it's ready, Patreons will get a early access in a few weeks time all been well.

commit dc597f1f (2021-02-13): Fixed #271 - Incorrect error message when firing power settings.
closed 2021-02-27 by NotExpectedYet