OctoFarm/OctoFarm #277: [BUG] Final slash in printer URL not normalized in REST calls

davidzwa, 2021-02-12:
**Description**
Adding a slash after the printer URL causes the connection to fail (due to // ending up in the OctoPrint REST calls)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'printers'
2. Adjust printer URL to end with `/`
3. Save
4. See error connecting to printer Host 

**Expected behavior**
Final slash should be normalized before saved to database or when doing external API calls.

**Screenshots**
URL with slash failing
[image]
URL without slash working fine
[image]

**Version**
- Master branch 1.1.10 version
**Additional context**
Did also check development if any fix was underway, could not spot it.

davidzwa, 2021-02-12:
Can be closed on the spot if this is already fixed.

NotExpectedYet, 2021-02-12:
[image]

This was fixed ages ago, how are you producing that? Editing there? that may skip the above as it wouldn't re-setup the defaults.

davidzwa, 2021-02-12:
Yes continuously editing the URL as I experienced mDNS issues regularly at home (hostnames exist, but not resolving: yay!). 
Also the default came with slash.

I'll check right now on dev, if it persists, just to be sure.

NotExpectedYet, 2021-02-12:
It's not always DNS aha

The fix is in when you initially save the printer, as I said it may get skipped currently with updating. 

What default came with a slash?

davidzwa, 2021-02-12:
Double checked on dev.

So, what I filled in as URL initially was `http://prusa/`, and that failed to resolve (btw I have the same problems within PrusaSlicer, it tests fine but won't send prints to OctoFarm afterwards... my knowledge with FQDN is not 100% expert, I admit that. Just had a lot of instability issues with it last year during an IoT internship.)

I guess, don't really recall, changing `http://prusa` which connected to host but failed on websocket.
[image]

Then I changed to IP directly (without ending slash), and that worked. Changed to the one with slash, and as you state, we know what goes wrong there.

closed 2021-02-27 by NotExpectedYet