OctoFarm/OctoFarm #279: [BUG] API connection checked too early after powering up

Chrismettal, 2021-02-13:
As discussed externally. 

Powering up a printer might be detected during ping/ponging when the API is not ready yet, so the API check might fail. Printer 225 powered on in the screenshot. Not checked anymore afterwards, resync was required. Not happening every time, might be timing related.

[image]

NotExpectedYet, 2021-02-13:
Ahh new error code I didn't know about... the 502. Should catch that with the next push and follow through with the increased retry time now :)

davidzwa, 2021-02-13:
I must add to this issue that offline printers slow down and sometimes even prevent my OctoFarm from starting. Maybe the timeout is longer than my patience xD

NotExpectedYet, 2021-02-13:
> I must add to this issue that offline printers slow down and sometimes even prevent my OctoFarm from starting. Maybe the timeout is longer than my patience xD

There isn't much I can do about offline printers, I have to check and I have to use the available timeouts. For development I'd use a VM and my script to spin up virtual octoprint instances and printers. There's a script on my git that should help set that up. 

What are the errors your getting? I capture a HOST-because if host is unreachable it doesn't repol, so boot time would only be 1000ms * available printers. If that's already the case then, It needs to happen in the background and not have it effect the loading of the UI but I won't be tackling that with the current mess aha. For dev though, deffo the virtual printer route.

davidzwa, 2021-02-13:
```
Sat, 13 Feb 2021 12:59:56 GMT | ERROR | OctoFarm-State.log | Could not Establish API Connection: undefinedhttp://192.168.1.177 | data:"Couldn't grab initial connection for Printer: http://192.168.1.177" |
```
So host reachable, prusa is just off (as I did on purpose, I dont want my baby running all day).

I agree that is not our focus atm, given that it 'could be' a painful thing for users who have big farms.

NotExpectedYet, 2021-02-13:
> ```
> Sat, 13 Feb 2021 12:59:56 GMT | ERROR | OctoFarm-State.log | Could not Establish API Connection: undefinedhttp://192.168.1.177 | data:"Couldn't grab initial connection for Printer: [...]

Aye it is, I made it way faster than it was. I'm confused to why it actually stops yours booting though, or actually slows it down. 

I just restarted with all these and interface was live instantly watching them check through? 

[image]

NotExpectedYet, 2021-02-13:
That did really really slow down the web interface though. I can gather that's the ticker.

davidzwa, 2021-02-13:
Yes indeed, it's the UI halting slightly. My problem was RAM (WSL/Docker). So I take back what I said.
Both printer and host off work fine as far as I can see.

NotExpectedYet, 2021-02-13:
> Yes indeed, it's the UI halting slightly. My problem was RAM (WSL/Docker). So I take back what I said.
> Both printer and host off work fine as far as I can see.

Certainly the ticker I think, I was sending 10000 lines to the client which it was chugging through, even worse when you load your server up to find no database my browser crashed XD poor firefox dev.  

Just pushed a reduction to that so "should" be better now.

NotExpectedYet, 2021-02-13:
That logic does need to be better, 50+ printers and you probably wouldn't see much as the ping/pongs flow through. Might take them out of that, but I actually like seeing it and it's been handy with a recent issue.

davidzwa, 2021-02-13:
I have a lot of ... food for thoughts on logging. Gonna spare you that for now ;) Something something Grafana ... cough InfluxDb ... cough Prometheus.... cough cough

NotExpectedYet, 2021-02-13:
> I have a lot of ... food for thoughts on logging. Gonna spare you that for now ;) Something something Grafana ... cough InfluxDb ... cough Prometheus.... cough cough

bahaha Please refer to my influxDB export and my previous comments on dependencies cough :P 

No I totally agree, I would like to streamline that better and if you have any suggestions shove it in a discussion as my current plan was to use the state management hooks to create a log and then stream the last parts down to the client good and propa!

commit 60827959 (2021-02-13): Fixed #279 - A booting Pi would never try and reconnect
closed 2021-02-27 by NotExpectedYet